zigbeefordomoticz/Domoticz-Zigbee #782: AM43 Blind Drive - TS601-_TZE200_rddyvrci

deennoo, 2021-01-18:
* Product Name = AM43 Blind Drive
* Brand = MOES
* URL on a web site = https://www.moeshouse.com/collections/zigbee-chain-roller-blinds-motor/products/zigbee-smart-diy-motorized-roller-blinds-shades-drive-motor-hub-required-tuya-smart-life-app-control?variant=32739216228433
* What do you expect = Open / Close / %opening or closing / ability to reverse motor rotation
* The logs from the pairing process

[export (3).zip](https://github.com/pipiche38/Domoticz-Zigate/files/5832657/export.3.zip)

Domoticz create reconnize it as a eTRV, with 5 widgets (thermo mode, door, temperature, thermosetpoint and %)

We just need a Blind Percentage Widget

[Zigate_03_2021-01-18.log](https://github.com/pipiche38/Domoticz-Zigate/files/5832666/Zigate_03_2021-01-18.log)

Join : some log playing with device button (open, close, open stop open)

Some source : 
https://github.com/Koenkk/zigbee-herdsman-converters/issues/2001
https://zigbee.blakadder.com/Moes_AM43-0_45_40-ES-EB.html

Have a good trip

pipiche38, 2021-01-19:
d2af3d7f

Please update to the latest version of branch SireneTuyaTS0601

restart the plugin

remove the widget created from last pairing

redo the pairing, a blind percentage widget should be created

Don't do anything with Domoticz 

enable tuyaDebug

and play with the motor directly and collect the logs for each command done on the motor
I'm interested on logs like that one:
" [ZiGateForwarder_17] tuyaReadRawAPS - Nwkid: fc08/01 fcf: 09 sqn: 06 cmd: 02 status: 00 transid: 02 dp: 69 datatype: 02 fn: 00 data: 000000e3"

please drop the logs here

For the Domoticz to motor side, we will do in a 2nd step

deennoo, 2021-01-19:
[Zigate_03_2021-01-19.log](https://github.com/pipiche38/Domoticz-Zigate/files/5838763/Zigate_03_2021-01-19.log)

1ere manipulation : Store ouvert, j'appui sur le bouton fermer, et le laisse ce fermer seul
2 eme manip : store fermé j'appuie su ouvrir et le laisse s'ouvrir au max
3 eme manip : store ouvert au max, j'appuie sur fermer et le stop a 66%
4 manip store a 66% j'appuie sur ouvrir et le stop a 33%

RezaRose26, 2021-06-12:
I have the same issue. When connecting the device i have 3 widgets in Domoticz. 
-Zigate USB TLL - TS0601_ThermoMode_2-5c0272fffecbff21-01
-Zigate USB TLL - TS0601_ThermoSetpoint-5c0272fffecbff21-01
-Zigate USB TLL - TS0601_Temp-5c0272fffecbff21-01

RezaRose26, 2021-06-12:
[logexport.txt](https://github.com/pipiche38/Domoticz-Zigate/files/6642101/logexport.txt)

pipiche38, 2021-06-14:
@RezaRose26 could you kindly drop the export.json from the PluginDeviceList (RAW) ?