zigbeefordomoticz/Domoticz-Zigbee #788: Volets roulants interrupteur LoraTap SC400ZB-EU (TS130F)

Miloune, 2021-02-04:
Bonjour,

J'ai acheté un interrupteur Zigbee pour mes volets roulants Somfy, j'ai fais le branchement, mais il faudrait maintenant calibrer l'interrupteur. Ceci peut être fait via l'application Tuya mais il faut disposer de la Gateway que je n'ai pas.

J'ai donc appareillé l'interrupteur avec la Zigate, lorsque je l'active j'entends bien l'interrupteur faire un petit clic mais il ne bouge pas mes volets. C'est pour cela que je pense que la calibration est nécessaire.

* Product Name: Interrupteur connecté pour volets
* Brand: LoraTap SC400ZB-EU (TS130F)
* URL on a web site: https://fr.aliexpress.com/item/4001190275372.html
* What do you expect: Je peux calibrer l'interrupteur depuis domoticz
* The logs from the pairing process
* export.json file as described here above: [export.zip](https://github.com/pipiche38/Domoticz-Zigate/files/5927864/export.zip)

pipiche38, 2021-02-05:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3134#issuecomment-687631699

pipiche38, 2021-02-05:
Widgets for this device:

- Window Covering
- Switch to Enable/Disable Calibration
- Switch Selector for Motor Reverse

pipiche38, 2021-02-05:
@miloune I'll try to work on it during the week-end. But you will need to switch to the beta branch.

I'll provide a switch widget to enable/disable Calibration in addition of the Window covering widget

pipiche38, 2021-02-06:
@Miloune can you switch to the plugin beta branch ?

```
git checkout beta
git pull

```

1/ Then you restart the plugin
2/ you remove the Widgets which have been created during the first pairing
3/ you redo the pairing

4/ enable the following debug parameters ( Web Admin -> Tools -> Debug (Advanced )
enable debugTuya
enable debugInput
enable debugCluster
enable debuginRawAPS
enable debugzigateCmd

Then you can start playing.
- Enable Calibration and then do the calibration

I will need the full log (from the pairing up to all tests) in order to debug if there is any issues.

pipiche38, 2021-02-19:
@Miloune , any news ?

Miloune, 2021-03-02:
Sorry didn't have to much time actually, I will try the beta branch this week-end

Miloune, 2021-03-08:
Hello,

I switched to beta branch. Delete / redo the pairing of the device. Got two devices in Domoticz. One to control my shutter and one for the calibration.

I tried the calibration. It seems to be working because of the tic sound is now much longer, but this product seems to not control my shutter. I contacted the Aliexpress seller and I will send back the product. My shutter seem to use RTS protocol...

Sorry for the delay and for the time you "loose" to develop this since I cannot confirm at 100% it's working. In my point of view, it's ok for me, had no error.

closed 2021-03-08 by Miloune