zigbeefordomoticz/Domoticz-Zigbee #789: Error Code: TRANS-onMESS-01

Miloune, 2021-02-04:
**Describe the bug**
I have a lot of errors like this in domoticz logs, what should I do? Where this message come from?
```
2021-02-04 19:38:21.999 Error: (Zigate) on_message Error Code: TRANS-onMESS-01
2021-02-04 19:38:22.058 Error: (Zigate) on_message Error Code: TRANS-onMESS-03
2021-02-04 19:38:22.110 Error: (Zigate) on_message Frame error Crc/len bytearray(b'\x01\x08\t\xc2\x80\x130$ \x00\x13\x0bS\xa1\x11 \x06\x00\x19\x001\x80\x00S\x01(\x00\x13 \x00\x00\xb9B \x00\x97\xbb\t\x13\x06\x19\x00\xb7\xfe\x01\x80\x00\x00\x07\x93\x00\x00\x00\x14\x00\x00\x00\x03')
```
[errors.zip](https://github.com/pipiche38/Domoticz-Zigate/files/5927888/errors.zip)

**Plugin (please complete the following information):**
 - Plugin version  4.11.016

Miloune, 2021-02-04:
I had power problem on my RaspberryPi, since I switched to another power outlet and restarted the PI, I don't have these errors anymore for now.

pipiche38, 2021-02-06:
Please use the Forum for such type of issues in  order to
get support from the community
qualify if that is a plugin bug

Thanks

closed 2021-02-06 by pipiche38