zigbeefordomoticz/Domoticz-Zigbee #790: Standalone version of this plugin!

perhusaas, 2021-02-04:
This wonderful plugin could have even more users if you made it accessible to non-domoticz users? 

From the outside it seems to be only a minor change to make this run indepenently, as it has its own GUI? 

I understand if you'd like to keep it in the domoticz community, just thought i'd ask. (which should be acceptable in the Open Source World!) :)

pipiche38, 2021-02-06:
Thanks for the feedback. The plugin in its recent version is more and more decoupled from Domoticz. However, Domoticz is the User Interface used to report the status of devices and to take commands/actions . That part is not in the plugin.

Any suggestions more than welcome

perhusaas, 2021-02-06:
I think theres only 4 things that's needed:

- Run the plugin as a service
- web address to json data (like from domo: http://<domo ip:port>/json.htm?type=devices&filter=all&used=true&order=Name) for device creation
- place to input variables IP and Port for sending device changes by UDP (.conf file or preferably in web GUI)
- Some way to send change commands to the plugin. (domo: http://<domo ip:port>/json.htm?type=command&param=...)

I would think there is some equivalent functions for the latter 3 in the code already mabye? 
The json data would offcourse be simpler from that of domoticz, but some sort of device type data next to the sensor data would be nessecary.

I have no idea about the amount of changes needed on the plugin side, but thanks for entertaining the idea. :)

perhusaas, 2021-02-19:
I've been playing a lot with docker and MQTT lately, (and wass thrilled to find a dockerfile in your manual!) and it seems MQTT would be an even better universal interface than HTTP commands w. json?

Smanar, 2021-02-20:
> MQTT would be an even better universal interface than HTTP commands w. json

I can't say for command, but not for protocol, http is native, you can use it even with a 20 years old computer out of the box, not for MQTT.

perhusaas, 2021-02-20:
Fair point, but its the instant bi-directional updates that requires more overhead with HTTP, as i have understood it? 
Its either done with polling at intervals (lots of unnecessary traffic) or setting both sides up with some communication code. Thats a lot more bridge specific code than MQTT (or web sockets?) subscriptions..

Smanar, 2021-02-20:
>Its either done with polling at intervals (lots of unnecessary traffic)

Yeah realy bad idea

> or setting both sides up with some communication code. Thats a lot more bridge specific code than MQTT (or web sockets?) subscriptions

I can't be sure, websocket is realy something basic too, and don't need additionnal server.

Another pist some devs are looking on my side is Server Sent Events (SSE) https://kevinchoppin.dev/blog/server-sent-events-in-php (unidirectionnal protocol)

closed 2021-06-09 by stale[bot]