zigbeefordomoticz/Domoticz-Zigbee #791: Aqara Smart plug shows 10 times the value in watts

perhusaas, 2021-02-08:
My Aqara Smart plug reports that my car heater is using 16500 watts! When i used a Z-wave plug it was steady at about 1550 watts. The unit may be calibrated differently, but the *10 looks like a variable handling issue?

pipiche38, 2021-02-08:
please provides additional informations.

I need export.json of Plugin Devices RAW (Json)

On which version of the plugin are you ? on which branch of the plugin are you ?

Last:

in the Web Admin Tools -> Debug (Advnaced)

enable debugCluster 

please provides Logs when the debugCluster parameter is enabled and the Aqara plug is sending Data and get updated on the widget

perhusaas, 2021-02-08:
Will do - but it may be a couple of days, lots of other stuff going on..

pipiche38, 2021-02-19:
Any news, otherwise I'll close the issue and you will open a new one when ready

perhusaas, 2021-02-19:
Sorry, i'll have a look at this later, if you don't hear back by tomorrow, feel free to close it.

perhusaas, 2021-02-19:
correction: version 4.11. (didnt find it anywhere in the GUI?)

pipiche38, 2021-02-20:
fix in 9e26afae  beta-4.12.150

pipiche38, 2021-02-20:
please switch to beta branch, remove the existing widget from domoticz in order to do a full pairing from scratch

perhusaas, 2021-02-20:
First time i've tried to update, and when i type git pull, i get: 
`fatal: not a git repository (or any of the parent directories): .git`

pipiche38, 2021-02-20:
How did you install the plugin at first ?

From your description either you are not on the right folder or you didn't install the plugin via a git clone command

perhusaas, 2021-02-20:
Don't remember, but I established git connection now, and am on 4.12.151. I selected "accept new hardware", and reset one of the plugs. this now only has the 
Zigate - lumi.plug.maeu01_Power-04cf8cdf3c79d303-01
Zigate - lumi.plug.maeu01_Meter-04cf8cdf3c79d303-01
Zigate - lumi.plug.maeu01_Ampere-04cf8cdf3c79d303-01

the actuator is missing?

perhusaas, 2021-02-20:
Removed it comletely again, and reincluded. Now its in, showing correct wattage. Thanks! :)

closed 2021-02-20 by pipiche38