zigbeefordomoticz/Domoticz-Zigbee #795: Plus aucun log

grrosminet, 2021-03-11:
Bonjour,

Il n'y a pas si longtemps que ça, je pouvais surveiller les messages reçus par la ZiGate en surveillant mon fichier de log

Exemple de message :
 Mar 11 08:10:42.718464Decod8000 Received [  ] PacketType:  0530 TypeSqn: 00 sqn_app: 00/0 sqn_aps: 00/0 Status: [a6] npdu: None apdu: None

L'idée était de surveiller les messages Decod8011 d'un objet qui ne sera pleinement pris en charge que par le firmware 31e et de déclencher une action dans domoticz par l'API JSON

Aujourd'hui, je n'ai plus aucun log de généré, ni dans le fichier spécifique du plugin, ni dans celui de domoticz
J'ai essayé de tout cocher dans la page de debug, mais rien n'y fait

plugin 4.12.201, branche beta (pour cause de matériel non supporté en stable)

pipiche38, 2021-03-11:
Please do use the forum which is for this type of question and not the github which is about qualified issues

you have to enable debugzigateCmd in order to get this message

grrosminet, 2021-03-11:
Please consider this as a real and qualified issue.

When I say that everything is checked it's really everything : [https://cloud.charles.tf/index.php/apps/files_sharing/publicpreview/JLdfnqCQcRo3zSi?x=2592&y=858&a=true&file=zigate_plugin_debug_all_checked.png&scalingup=0](url)

And the log file remains empty (it is created at startup) : 0 byte

Please tell me on which active forum I can contact you (easydomoticz is down)

pipiche38, 2021-03-11:
[image]