zigbeefordomoticz/Domoticz-Zigbee #805: Domoticz limitation to 250 Widgets

pipiche38, 2021-04-29:
The ZiGate plugin is based on the Python Framework embedded into the Domoticz code.
Unfortunately the limitation of 255 widgets is given by Domoticz and the underlying framework.

Each widget created inside the plugin are named "device" and you get a Unit number which is a unique number to identify a domoticz widget which belongs to the plugin.

Until Domoticz Python Framework move the limitation of 255 (1 byte ) to 65K ( 2bytes) we will have such limit.

While Zigate V1 can go to 70 nodes (objects) which can generate from 1 to 6 widgets, we see the limit coming soon especially with the recent annouce ZiGate V2 which should be able to go beyond the 100 nodes!

pipiche38, 2021-04-29:
https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/325
domoticz/domoticz#4743
https://www.domoticz.com/forum/viewtopic.php?f=65&t=35452

pitchoun38, 2021-05-22:
This is becoming weird. any update on domoticz/domoticz#4743 , I have in my case 150 domoticz Devices for 42 real/physical devices. While I expect to grow now the number of devices due to the new ZiGate+ what is left to me ?
Is Domoticz finally going to remove such stupid limit in 2021 !

pipiche38, 2021-08-05:
Wont' fix

closed 2021-08-05 by pipiche38