zigbeefordomoticz/Domoticz-Zigbee #817: Seems that firmware detection and firmware available is not accurate

cmoicyril, 2021-05-30:
Hello
Thanks for your great work on the plug in

I have Zigate+ on USB, working well.

In the interface, corner bottom-right, there's "firmware" : "Firmware: 031e
Plugin: 5.1.002" I see it's in red, i click, it proposes me to go to GitHub.

https://github.com/fairecasoimeme/ZiGate/releases 

But there's no more recent version than 3.1e on GitHub.

My configuration : 

Domoticz ver: 2021.1 (build 13277) / Docker sous Synology DSM 6.2.4
dzVents Version: 3.18
Python Version: 3.7.3 [GCC 8.3.0]
Z-Wave via USB Key SA 413. Zigate+ -USB- 

[image]

pipiche38, 2021-05-30:
As recommended in the bug template it is better to use the forum in order  to qualify there is an issue.

Could you please provide the output of Tools->Zigate Info (json)

cmoicyril, 2021-05-30:
here it is : 

ZiGate infos
Les informations de la ZiGate

Channel : 11
Extended PANID : d6e08f1e35e7987b
Firmware Version : Branch: 01 Major: 05 Version: 031e
IEEE : 00158d000500734a
PANID : 290c
Short Address : 0000

pipiche38, 2021-05-30:
Are you with a Zigate V1 or V2 ? It looks a V2, and indeed for V2, there is a new firmware

https://github.com/fairecasoimeme/ZiGatev2/releases/tag/v3.1f

closed 2021-05-31 by pipiche38