zigbeefordomoticz/Domoticz-Zigbee #821: Basically we are not receiving what we expect from Zigate

Miloune, 2021-06-20:
Hello,

Got some errors /  problems with Zigate. I had problem with my raspberry pi and Wifi connection and cut power, didn't know if some data are corrups or not. 

Here is logs :

```
2021-06-20 15:34:15.366 Status: Zigate: (Zigate) ==> Transport Mode: auto
2021-06-20 15:34:15.368 Status: Zigate: (Zigate) Connection Name: Zigate, Transport: Serial, Address: /dev/serial0
2021-06-20 15:34:15.368 Status: Zigate: (Zigate) Connection open: Name: 'ZiGate', Transport: 'Serial', Protocol: 'None', Address: '/dev/serial0', Port: '', Baud: 115200, Timeout: 0, Connected: False, Parent: 'None'
2021-06-20 15:34:18.231 Zigate: (Zigate) BinMsg: bytearray(b'\x01\x13$ \x01\x80\x00\x00\x07\x97\x00\x00\x00\x10\x00\x00\x00\x03') ExpectedLen: 18 ComputedLen: 8200
2021-06-20 15:34:18.232 Status: Zigate: (Zigate) ZiGate Classic
2021-06-20 15:34:18.232 Status: Zigate: (Zigate) Installer Version Number: 031d
2021-06-20 15:34:18.182 Error: Zigate: (Zigate) on_message Error Code: TRANS-onMESS-01
2021-06-20 15:34:18.211 Error: Zigate: (Zigate) on_message Error Code: TRANS-onMESS-03
2021-06-20 15:34:18.231 Error: Zigate: (Zigate) on_message Frame error Crc/len bytearray(b'\x01\x13$ \x01\x80\x00\x00\x07\x97\x00\x00\x00\x10\x00\x00\x00\x03')
2021-06-20 15:34:21.189 Status: Zigate: (Zigate) ==> Transport Mode switch to: zigate31d
2021-06-20 15:34:21.190 Status: Zigate: (Zigate) ZiGate Classic
2021-06-20 15:34:21.190 Status: Zigate: (Zigate) Installer Version Number: 031d
2021-06-20 15:34:22.089 Xiaomi AirPurifier: (Xiaomi AirPurifier) Skip heartbeat handler because stage not enough 1/5.
2021-06-20 15:34:24.214 Zigate: (Zigate) setZigateIEEE - Set Zigate IEEE: 00158d0003af5f9f
2021-06-20 15:34:24.214 Status: Zigate: (Zigate) Zigate addresses ieee: 00158d0003af5f9f , short addr: 0000
2021-06-20 15:34:24.214 Status: Zigate: (Zigate) Network state UP, PANID: d679 extPANID: 0x9d53eef82e4fe6d0 Channel: 12
2021-06-20 15:34:24.097 Error: Zigate: (Zigate) We are having difficulties to start Zigate. Basically we are not receiving what we expect from Zigate
2021-06-20 15:34:24.097 Error: Zigate: (Zigate) Plugin is not started .
```

```
2021-06-20 15:36:33.415 Zigate: (Zigate) BinMsg: bytearray(b'\x013 \x00\x15\x003\x06\x80\x00S(b\t3 \x13#\x8a\x80\x00s b\t3 \x13\x00\x13{\xfe\x01\x81\x02\x00\x0e)\xc9=)\x01\x00\x06\x00\x00\x00\x10\x00\x01\x00o\x03') ExpectedLen: 53 ComputedLen: 28
2021-06-20 15:36:33.351 Error: Zigate: (Zigate) on_message Error Code: TRANS-onMESS-01
2021-06-20 15:36:33.382 Error: Zigate: (Zigate) on_message Error Code: TRANS-onMESS-03
2021-06-20 15:36:33.415 Error: Zigate: (Zigate) on_message Frame error Crc/len bytearray(b'\x013 \x00\x15\x003\x06\x80\x00S(b\t3 \x13#\x8a\x80\x00s b\t3 \x13\x00\x13{\xfe\x01\x81\x02\x00\x0e)\xc9=)\x01\x00\x06\x00\x00\x00\x10\x00\x01\x00o\x03')
```

[Json History from Zigate Web Interface](https://github.com/pipiche38/Domoticz-Zigate/files/6682448/errors.txt)

Versions:
Plugin: 4.11.017
Domoticz Version: 2021.1
Python Version: 3.7.3 (default, Jan 22 2021, 20:04:44) [GCC 8.3.0]

I can do some test if you want.
Regards

Miloune, 2021-06-20:
Didn't noticed there is a new branch stable. Switch to it => Plugin: 5.1.004

Got less errors, but the delay is pretty huge when I send a command to switch on / off the light (13 sec)

[errors (2).txt](https://github.com/pipiche38/Domoticz-Zigate/files/6682469/errors.2.txt)

pipiche38, 2021-06-20:
please kindly use one of the forum ( French, or English ), from what I see is you have communication issues with the ZiGate

closed 2021-06-20 by pipiche38