zigbeefordomoticz/Domoticz-Zigbee #826: Zigate Wifi : Error: Zigate: (Zigate) pingZigate - no Heartbeat with Zigate, try to reConnect

xbeaudouin, 2021-07-12:
**Describe the bug**

I have a Domoticz vers 2021.1 (build 13360 and older), using 2 interfaces one specific for usage (eg end user), and IOT (MQTT / Wifi modules etc...).

I currently runs FreeBSD 13.0-RELEASE, with my own port or Domoticz on FreeBSD, and python 3.7.

All plugins seems to works perfectly (some are from me), but, since the begining I have issue with domoticz and zigate wifi.

Since lots of time I got about one time per day the following log : 
` 2021-07-12 15:46:09.070 Error: Zigate: (Zigate) pingZigate - no Heartbeat with Zigate, try to reConnect `

It was "almost" ok when I am doing some nasty things on the wifi, BUT, now everyday I have to restart the plugin on domoticz (enable / disable) to get is working again.

I was suspecting the wifi to have not the good strength, but, now the Zigate Wifi is a 2,5m of an UAP-PRO, and this the only device on domoticz that such nasty behavior.

Since it "trys" but it does not connect to the zigate, which is not the expectect behavior.

**To Reproduce**

Quite simple : 
- install freebsd 13.0
- use my port https://framagit.org/kiwi/freebsd-ports/-/tree/master/www/domoticz-devel
- install zigate wifi
- add some device
- use the plugin and wait ...

**Expected behavior**

If the plugin lost its connectivity it should try again to connect. Maybe it does not do what is should do : try to connect again.

**Plugin (please complete the following information):**
 - OS: FreeBSD 13.0-RELEASE / AMD64
 - Plugin version : 4.11.017
 - Plugin branch : master

**Additional context**
Add any other context about the problem here.

pipiche38, 2021-07-12:
Please do use the forum ( French or English)

pipiche38, 2021-07-12:
I beleive you are refering to "stable" branch of the Zigate plugin? as you are on 4.11

Which Wifi Firmware are you using 1.x or 2.x ?
Are you sure that the Wifi doesn't get disconnected from the AP ?
Could you also check the PingZigate is enabled in the Plugin Web UI -> Settings ? 

[image]

xbeaudouin, 2021-08-12:
Sorry for the long reply... Holidays... :D 
I currently use Wifi Firmware v2.00 (with esp-link v3.2.50-gfe4f565-dirty)
The wifi does sometimes disconnect and roam (have unifi in the house) when the AP get a reboot or upgrades itself...

And the ping is enabled... Seems there is something special under FreeBSD ... Does ping on the pluging spanw the binary ping from OS?

Kind regards

pipiche38, 2021-08-13:
No the "ping" is home made in the plugin (not OS related), the ping is done inside the communication beteen the plugin and the Zigate module

But if your wifi network is not stable and then the Wifi module get disconnected , this will cut the session and the connection need to be restarted.

xbeaudouin, 2021-08-15:
Thanks. I have upgraded to version 5... Let's see if the issue is still there...

pipiche38, 2021-08-22:
Found a potention issue as regards to Wifi and TCP connection. The problem should occur randomly. Fix in progress

pipiche38, 2021-08-22:
00d91892ad92f9972868f7a4ac134e70cb2b0807

pipiche38, 2021-08-24:
should be fixed with beta 5.2.080

xbeaudouin, 2021-08-28:
Well : 
```
[root@domoticz /usr/local/domoticz/plugins/Domoticz-Zigate]# git pull
remote: Enumerating objects: 497, done.
remote: Counting objects: 100% (414/414), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 320 (delta 223), reused 231 (delta 143), pack-reused 0
Receiving objects: 100% (320/320), 792.45 KiB | 281.00 KiB/s, done.
Resolving deltas: 100% (223/223), completed with 62 local objects.
From https://github.com/pipiche38/Domoticz-Zigate
 * [new branch]        ReInitOnExistingDb -> origin/ReInitOnExistingDb
   cad8e16c..074a0676  beta               -> origin/beta
 * [new branch]        pipiche-debuging   -> origin/pipiche-debuging
   e6514926..5ee5378b  stable5            -> origin/stable5
   f97bd850..544644ff  stable5-fix        -> origin/stable5-fix
 * [new tag]           beta-5.2.078       -> beta-5.2.078
 * [new tag]           beta-5.2.079       -> beta-5.2.079
 * [new tag]           beta-5.2.080       -> beta-5.2.080
 * [new tag]           beta-5.2.081       -> beta-5.2.081
 * [new tag]           beta-5.2.082       -> beta-5.2.082
 * [new tag]           beta-5.2.083       -> beta-5.2.083
 * [new tag]           stable-5.1.007     -> stable-5.1.007
Already up to date.

```

Which log I can provide...? Do I have to set plugin to logging verbose?

pipiche38, 2021-08-28:
what about 

`git status`

To start with, I need the plugin logs (default one) at plugin start, then when the problem occurs .

xbeaudouin, 2021-08-28:
```
[root@domoticz /usr/local/domoticz/plugins/Domoticz-Zigate]# git status
On branch stable
Your branch is up to date with 'origin/stable'.

nothing to commit, working tree clean
```

Here is the startup log : 

```
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Started.
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Initialized version 4.11, author 'zaraki673  pipiche38'
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Zigate plugin stable-4.11.017 started
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Debug: 0
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) load PluginConf
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) port set to 80
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) channel set to 11
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) filename set to /usr/local/domoticz/plugins/Domoticz-Zigate/Conf/PluginConf-38.json
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) pluginHome set to /usr/local/domoticz/plugins/Domoticz-Zigate/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) homedirectory set to /usr/local/domoticz/plugins/Domoticz-Zigate/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) pluginData set to /usr/local/domoticz/plugins/Domoticz-Zigate/Data/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) pluginConfig set to /usr/local/domoticz/plugins/Domoticz-Zigate/Conf/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) pluginOTAFirmware set to /usr/local/domoticz/plugins/Domoticz-Zigate/OTAFirmware/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) pluginReports set to /usr/local/domoticz/plugins/Domoticz-Zigate/Reports/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) pluginWWW set to /usr/local/domoticz/plugins/Domoticz-Zigate/www/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) pluginLogs set to /usr/local/domoticz/plugins/Domoticz-Zigate/Logs/
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Python Version - 3.7.11 (default, Jul 22 2021, 22:18:10)  [Clang 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) DomoticzVersion: 2021.1 (build 13416)
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) DomoticzHash: c6d422f36
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) DomoticzBuildTime: 2021-08-15 08:37:02
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Opening DomoticzDB in raw
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Plugin Database: DeviceList-38.txt
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) DeviceConf loaded
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) load ListOfDevice
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) 12 Entries loaded from /usr/local/domoticz/plugins/Domoticz-Zigate/Data/DeviceList-38
.txt
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Transport mode: Wifi
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) ==> Transport Mode: auto
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Connection Name: Zigate, Transport: TCP/IP, Address: None:9999
Aug 28 13:28:57 domoticz domoticz[3847]: Zigate: (Zigate) Connection open: Name: 'Zigate', Transport: 'TCP/IP', Protocol: 'None ', Address: '19
2.168.20.105', Port: '9999', Baud: -1, Timeout: 0, Connected: False, Parent: 'None'
Aug 28 13:28:58 domoticz domoticz[3847]: Zigate: (Zigate) ==> Transport Mode switch to: zigate31d
Aug 28 13:28:58 domoticz domoticz[3847]: Zigate: (Zigate) ZiGate Classic
Aug 28 13:28:58 domoticz domoticz[3847]: Zigate: (Zigate) Installer Version Number: 031d
Aug 28 13:29:08 domoticz domoticz[3847]: Zigate: (Zigate) Zigate addresses ieee: 00158d00021e10e7 , short addr: 0000
Aug 28 13:29:08 domoticz domoticz[3847]: Zigate: (Zigate) Network state UP, PANID: 0227 extPANID: 0x6b844234127899dc Channel: 11
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) Request Disabling Accepting new Hardware
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) Accepting new Hardware: Disable
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) Number of devices recently active in Zigate = 9
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) [00] DevID = 00 Network addr = 7d0b IEEE = 00158d0004460622 LQI = 110 Power = 00 Mode
l = lumi.weather ( WC_T )
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) [01] DevID = 02 Network addr = 846f IEEE = 00158d0003585515 LQI = 103 Power = 00 Mode
l = lumi.sensor_switch ( Volets_SdB_Btn )
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) [02] DevID = 03 Network addr = e0c1 IEEE = 00158d00034d6f0d LQI = 118 Power = 00 Mode
l = lumi.sensor_motion ( Mvt_Entree )
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) [03] DevID = 04 Network addr = 8001 IEEE = 00158d00044cec74 LQI = 089 Power = 00 Mode
l = lumi.sensor_motion.aq2 ( Mvt_Degagement_Chambre )
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) [04] DevID = 05 Network addr = 4953 IEEE = 00158d00044b130c LQI = 104 Power = 00 Mode
l = lumi.sensor_motion.aq2 ( Mvt_Cuisine )
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) [05] DevID = 07 Network addr = d433 IEEE = 00158d0003cb4fd3 LQI = 132 Power = 00 Mode
l = lumi.sensor_magnet.aq2 ( Porte_Fenetre_Cuisine )
Aug 28 13:29:13 domoticz domoticz[3847]: Zigate: (Zigate) [06] DevID = 08 Network addr = 3bf5 IEEE = 00158d000413f238 LQI = 142 Power = 00 Mode
[... cut ...]
Aug 28 13:29:19 domoticz domoticz[3847]: Zigate: (Zigate) TxPower Attenuation : 0 dBm
Aug 28 13:29:24 domoticz domoticz[3847]: Zigate: (Zigate) There is a newer Zigate Firmware version available
Aug 28 13:29:28 domoticz domoticz[3847]: Zigate: (Zigate) No transport, write directive to '192.168.0.50:55493' ignored.
Aug 28 13:29:28 domoticz domoticz[3847]: Zigate: (Zigate) No transport, write directive to '192.168.0.50:55495' ignored.

```Seems I have version 4.x instead of 5.x, do I need to change of branch?

pipiche38, 2021-08-28:
Yes, seems that you are not on the right branch.

so either stable5 or beta would address that issue

git pull
git checkout stable5    (or beta )
git pull

then restart the plugin

xbeaudouin, 2021-08-28:
Done. Let's see if this is fixing the issue...

pipiche38, 2021-09-02:
@xbeaudouin Xavier any news ?

xbeaudouin, 2021-09-14:
Well seems very stable so far :)
So we can say this is successfully fixed !
Thanks to your help and quality of the code !

closed 2021-09-14 by xbeaudouin