zigbeefordomoticz/Domoticz-Zigbee #829: Intégration tuya TS0203

grrosminet, 2021-07-16:
When I try to pair my door/window sensor, I get this message : "_Unable to collect all informations for enrollment of this devices. See Logs._"

the error.json :
`"8":{"0":{"Time":"15/07/2021 12:30:41","Module":"Input","nwkid":"841b","PluginHealth":"Ready","message":"Something wrong on Device 841b 00124b001477fe0a pairing process. (aborting)","PermitToJoin":{"Duration":255,"Starttime":1626344829},"Thread":"ZiGateForwarder_2"},"LastLog":0,"StartTime":"15/07/2021 05:20:20","FirmwareVersion":"031e","FirmwareMajorVersion":"03","PluginVersion":["beta-5.2.050"]},"LastLog":8}
`

grrosminet, 2021-07-16:
Updated to beta 5.2.051
I can pair but sensor remains closed
[export.json.txt](https://github.com/pipiche38/Domoticz-Zigate/files/6832369/export.json.txt)

grrosminet, 2021-07-17:
> peux-tu activer le debuginRAWAPS et debugTuya ? et tu es bien sur une version du firmware 31e ?

Activated theses options, restarted the plugin, re-appairing with another identical device (in case the first one was broken), tryed to send some messages (open, close, open, close ...)

[domoticz.txt](https://github.com/pipiche38/Domoticz-Zigate/files/6834310/domoticz.txt)
[export.json.txt](https://github.com/pipiche38/Domoticz-Zigate/files/6834311/export.json.txt)

pipiche38, 2021-07-19:
Can you enable debugIAS and redo the full pairing and prodivides the Log ?

Do you have any Zigbee sniffer ?

grrosminet, 2021-07-19:
here is the debugIAS
[domoticz-dbg-ias.txt](https://github.com/pipiche38/Domoticz-Zigate/files/6843183/domoticz-dbg-ias.txt)

pipiche38, 2021-07-21:
3bf5f121

IAS Zone status change are based on Data Type 0x21

closed 2021-07-21 by pipiche38