zigbeefordomoticz/Domoticz-Zigbee #831: Osram Ledvance not pairing (previously worked fine) - 'ConsistencyCheck': 'Bad Pairing'

gorstj, 2021-08-15:
* Product Name 
Osram/Ledvance Smart+ Garden Pole RGBW

* Brand
Osram/Ledvance

* URL on a web site
*https://www.ledvance.com/consumer/products/smart-home/smart-home-luminaires/outdoor-luminaires/outdoor-led-luminaire-with-zigbee-technology--light-color-and-color-temperature-adjustable-c6296

* What do you expect
Device to pair and show up in domoticz.
This previously worked fine when I was trying the device within the house. Due to poor signal to a garden extension I changed the channel (from 11 to 19) and had to remove/reset my devices.
I now have good signal to some Zigbee devices within the extension and this device is just outside it (there are some switches acting as routers along the way)

* The logs from the pairing process
See below

* export.json file as described here above
*
[export.log](https://github.com/pipiche38/Domoticz-Zigate/files/6988832/export.log)

Domoticz log:
2021-08-15 21:03:14.651 Status: New sensors allowed for 5 minutes...
2021-08-15 21:03:26.000 Zigate: (Zigate) --> Adding device 6a50 in self.DevicesInPairingMode
2021-08-15 21:03:26.000 Zigate: (Zigate) --> ['6a50']
2021-08-15 21:03:29.706 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:03:32.415 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 0005 Status: a7
2021-08-15 21:03:32.982 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:03:36.077 Status: Zigate: (Zigate) [0] NEW OBJECT: 6a50 TimeOut in 0045 restarting at 0x004d
2021-08-15 21:03:36.116 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:03:39.616 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:03:42.495 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 0005 Status: a7
2021-08-15 21:03:43.608 RFLink: Temp + Humidity (Alecto V4)
2021-08-15 21:03:43.429 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:03:48.896 Zigate: (Zigate) UpdateDevice - (Zigate - lumi.ctrl_ln2.aq1_Meter-00158d000396135d-03) 0:55.01;114834.43
2021-08-15 21:03:51.086 Status: Zigate: (Zigate) [1] NEW OBJECT: 6a50 TimeOut in 0045 restarting at 0x004d
2021-08-15 21:03:55.866 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 8006 Status: a7
2021-08-15 21:03:57.509 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 0005 Status: a7
2021-08-15 21:04:06.058 Status: Zigate: (Zigate) [2] NEW OBJECT: 6a50 TimeOut in 0045 restarting at 0x004d
2021-08-15 21:04:12.481 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 0005 Status: a7
2021-08-15 21:04:16.179 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:04:18.607 RFLink: Temp + Humidity (Alecto V4)
2021-08-15 21:04:21.076 Status: Zigate: (Zigate) [3] NEW OBJECT: 6a50 TimeOut in 0045 restarting at 0x004d
2021-08-15 21:04:22.471 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:04:27.498 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 0005 Status: a7
2021-08-15 21:04:31.314 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:04:36.093 Status: Zigate: (Zigate) [4] NEW OBJECT: 6a50 TimeOut in 0045 restarting at 0x004d
2021-08-15 21:04:39.281 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:04:41.084 Error: Zigate: (Zigate) [5] NEW OBJECT: 6a50 Not able to get all needed attributes on time
2021-08-15 21:04:41.084 Error: Zigate: (Zigate) processNotinDB - not able to find response from 6a50 stop process at 0045
2021-08-15 21:04:41.084 Error: Zigate: (Zigate) processNotinDB - RIA: 5 waitForDomoDeviceCreation: False, capturePairingInfos: 1 Model:
2021-08-15 21:04:41.084 Error: Zigate: (Zigate) processNotinDB - Collected Infos are : {'Version': '3', 'ZDeviceName': '', 'Status': 'UNKNOW', 'SQN': '', 'Ep': {}, 'Heartbeat': '1', 'RIA': '5', 'LQI': {}, 'Battery': {}, 'Model': '', 'ForceAckCommands': [], 'MacCapa': '8e', 'IEEE': '7cb03eaa0a078267', 'Type': {}, 'ProfileID': {}, 'ZDeviceID': {}, 'App Version': '', 'Attributes List': {}, 'DeviceType': 'FFD', 'HW Version': '', 'Last Cmds': [], 'LogicalType': 'Router', 'Manufacturer': '', 'Manufacturer Name': '', 'NbEp': '', 'PowerSource': 'Main', 'ReadAttributes': {}, 'ReceiveOnIdle': '', 'Stack Version': '', 'Stamp': {'time': 1629057879.2810555, 'Time': '2021-08-15 21:04:39', 'MsgType': ' 4d', 'LastSeen': 1629057879}, 'ZCL Version': '', 'Health': 'Live', 'Capability': ['Full-Function Device', 'Main Powered', 'Receiver during Idle', 'Standard security', 'NwkAddr should be allocated'], 'PairingInProgress': True, 'ConsistencyCheck': 'Bad Pairing'}
2021-08-15 21:04:42.517 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 0005 Status: a7
2021-08-15 21:04:46.776 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:04:53.607 RFLink: Temp + Humidity (Alecto V4)
2021-08-15 21:04:57.299 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:04:59.150 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 8006 Status: a7
2021-08-15 21:05:09.663 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 8006 Status: a7
2021-08-15 21:05:28.610 RFLink: Temp + Humidity (Alecto V4)
2021-08-15 21:05:31.872 Zigate: (Zigate) UpdateDevice - ( Hutch) 0:18.1;88.4;3
2021-08-15 21:06:04.588 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:06:17.004 Zigate: (Zigate) Receive NACK from 6a50 clusterId: 8006 Status: a7
2021-08-15 21:06:19.250 Status: Zigate: (Zigate) Request Disabling Accepting new Hardware
2021-08-15 21:06:19.467 Status: Zigate: (Zigate) Accepting new Hardware: Disable (Off)
2021-08-15 21:06:29.155 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:06:36.647 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:06:38.608 RFLink: Temp + Humidity (Alecto V4)
2021-08-15 21:06:44.112 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:06:50.621 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:06:56.980 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e
2021-08-15 21:07:03.841 Status: Zigate: (Zigate) Device Annoucement: NwkId: 6a50 Ieee: 7cb03eaa0a078267 MacCap: 8e

pipiche38, 2021-08-16:
pleaqse the French or English forum

closed 2021-08-16 by pipiche38

gorstj, 2021-08-16:
Apologies - reposted in forum.

Is it worth changing the instructions on the Wiki if a device is not recognised not to post on GitHub?

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Problem_Dealing-with-none-certified-device.md

> Please create an issue here on GitHub: https://github.com/pipiche38/Domoticz-Zigate/issues/new?assignees=&labels=&template=Add_New_Hardware.md&title= with the here above informations

pipiche38, 2021-08-16:
@gorstj as you mentioned it was working before, this is not an issue of a new device not working with the plugin.
In your case I suspect more a firmware issue. but let see how it will move on the forum