zigbeefordomoticz/Domoticz-Zigbee #879: Aqara Door Sesnor MCCGQ14LM status not updated

rcrocus, 2021-11-14:
Hello,
I have an Aqara door sensor (MCCGQ11LM) what function very well.

I bought 2 news Aqara door sensor and I've received a new model : MCCQ14LM.
The pairing is OK in Domioticz, but after that, the status of the door sensor (open or close) is never updated in Domoticz even if i put the button on the sensor (in this case, the time of the last contact is well update but note the status.

Please find the log during the pairing :
 ```
2021-11-09 21:08:34.876 Status: Zigate: Request Accepting new Hardware for 240 seconds
2021-11-09 21:08:35.272 Status: Zigate: Accepting new Hardware: Enable (On)
2021-11-09 21:08:44.917 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:08:44.917 Zigate: --> Empty queue
2021-11-09 21:08:54.160 Zigate: --> Adding device c719 in self.DevicesInPairingMode
2021-11-09 21:08:54.164 Zigate: --> ['fa7c', 'c719']
2021-11-09 21:08:54.165 Status: Zigate: [None] NEW OBJECT: c719 None
2021-11-09 21:08:54.640 Status: Zigate: [-] NEW OBJECT: c719 Active Endpoint Response Ep: 01 LQI: 147
2021-11-09 21:08:54.643 Status: Zigate: [-] NEW OBJECT: c719 Request Simple Descriptor for Ep: 01
2021-11-09 21:08:55.058 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:08:55.058 Zigate: --> Empty queue
2021-11-09 21:08:55.162 Status: Zigate: [-] NEW OBJECT: c719 Simple Descriptor Response EP: 0x01 LQI: 138
2021-11-09 21:08:55.163 Status: Zigate: [-] NEW OBJECT: c719 ProfileID 0104
2021-11-09 21:08:55.163 Status: Zigate: [-] NEW OBJECT: c719 ZDeviceID 0402
2021-11-09 21:08:55.163 Status: Zigate: [-] NEW OBJECT: c719 Application Version 0000
2021-11-09 21:08:55.164 Status: Zigate: [-] NEW OBJECT: c719 Cluster IN Count: 05
2021-11-09 21:08:55.164 Status: Zigate: [-] NEW OBJECT: c719 Cluster In 1: 0000 (Basic)
2021-11-09 21:08:55.164 Status: Zigate: [-] NEW OBJECT: c719 Cluster In 2: 0001 (Power Configuration)
2021-11-09 21:08:55.164 Status: Zigate: [-] NEW OBJECT: c719 Cluster In 3: 0003 (Identify)
2021-11-09 21:08:55.165 Status: Zigate: [-] NEW OBJECT: c719 Cluster In 4: 0500 (IAS Zone)
2021-11-09 21:08:55.165 Status: Zigate: [-] NEW OBJECT: c719 Cluster In 5: fcc0
2021-11-09 21:08:55.165 Status: Zigate: [-] NEW OBJECT: c719 Cluster OUT Count: 02
2021-11-09 21:08:55.165 Status: Zigate: [-] NEW OBJECT: c719 Cluster Out 1: 0003 (Identify)
2021-11-09 21:08:55.166 Status: Zigate: [-] NEW OBJECT: c719 Cluster Out 2: 0019 (Over-the-Air Upgrade)
2021-11-09 21:08:56.081 Zigate: Request Basic Manufacturer via Read Attribute request: 0004
2021-11-09 21:08:56.083 Zigate: EP: {'01': {'0000': {}, '0001': {}, '0003': {}, '0500': {}, 'fcc0': {}, '0019': {}}}
2021-11-09 21:08:56.084 Status: Zigate: [None] NEW OBJECT: c719 Request Node Descriptor
2021-11-09 21:09:01.066 Zigate: EP: {'01': {'0000': {'0004': 'LUMI', '0001': '4', '0005': 'lumi.magnet.acn001'}, '0001': {}, '0003': {}, '0500': {}, 'fcc0': {}, '0019': {}}}
2021-11-09 21:09:01.064 Status: Zigate: [1] NEW OBJECT: c719 Model Name: lumi.magnet.acn001
2021-11-09 21:09:04.920 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:09:04.920 Zigate: --> Empty queue
2021-11-09 21:09:06.097 Zigate: EP: {'01': {'0000': {'0004': 'LUMI', '0001': '4', '0005': 'lumi.magnet.acn001'}, '0001': {}, '0003': {}, '0500': {}, 'fcc0': {}, '0019': {}}}
2021-11-09 21:09:06.207 Zigate: WARNING - send_zigatecmd : [300] 0100 02c7190101000100000000010020 c719 ZigateQueue: 16
2021-11-09 21:09:06.210 Zigate: WARNING - send_zigatecmd : [301] 0100 02c7190101000100000000010021 c719 ZigateQueue: 17
2021-11-09 21:09:06.213 Zigate: WARNING - send_zigatecmd : [302] 0100 02c7190101000100000000010033 c719 ZigateQueue: 18
2021-11-09 21:09:06.216 Zigate: WARNING - send_zigatecmd : [303] 0100 02c7190101000100000000010035 c719 ZigateQueue: 19
2021-11-09 21:09:06.217 Zigate: ReadAttributeRequest_0019 - Key: c719
2021-11-09 21:09:06.218 Zigate: Request 0x0019 info via Read Attribute request: c719 EPout = 01
2021-11-09 21:09:06.221 Zigate: WARNING - send_zigatecmd : [304] 0100 02c7190101001900000000010001 c719 ZigateQueue: 20
2021-11-09 21:09:06.223 Zigate: WARNING - send_zigatecmd : [305] 0100 02c7190101001900000000010002 c719 ZigateQueue: 21
2021-11-09 21:09:06.226 Zigate: WARNING - send_zigatecmd : [306] 0100 02c7190101001900000000010003 c719 ZigateQueue: 22
2021-11-09 21:09:06.229 Zigate: WARNING - send_zigatecmd : [307] 0100 02c7190101001900000000010004 c719 ZigateQueue: 23
2021-11-09 21:09:06.232 Zigate: WARNING - send_zigatecmd : [308] 0100 02c7190101001900000000010005 c719 ZigateQueue: 24
2021-11-09 21:09:06.234 Zigate: WARNING - send_zigatecmd : [309] 0100 02c7190101001900000000010006 c719 ZigateQueue: 25
2021-11-09 21:09:06.236 Zigate: WARNING - send_zigatecmd : [310] 0100 02c7190101001900000000010007 c719 ZigateQueue: 26
2021-11-09 21:09:06.238 Zigate: WARNING - send_zigatecmd : [311] 0100 02c7190101001900000000010008 c719 ZigateQueue: 27
2021-11-09 21:09:06.240 Zigate: WARNING - send_zigatecmd : [312] 0100 02c7190101001900000000010009 c719 ZigateQueue: 28
2021-11-09 21:09:06.241 Zigate: WARNING - send_zigatecmd : [313] 0100 02c719010100190000000001000a c719 ZigateQueue: 29
2021-11-09 21:09:06.245 Zigate: WARNING - send_zigatecmd : [314] 0100 02c7190101000000000000010004 c719 ZigateQueue: 30
2021-11-09 21:09:06.247 Zigate: WARNING - send_zigatecmd : [315] 0100 02c7190101000000000000010005 c719 ZigateQueue: 31
2021-11-09 21:09:06.248 Zigate: WARNING - send_zigatecmd : [316] 0100 02c7190101000000000000010000 c719 ZigateQueue: 32
2021-11-09 21:09:06.250 Zigate: WARNING - send_zigatecmd : [317] 0100 02c7190101000000000000010001 c719 ZigateQueue: 33
2021-11-09 21:09:06.252 Zigate: WARNING - send_zigatecmd : [318] 0100 02c7190101000000000000010002 c719 ZigateQueue: 34
2021-11-09 21:09:06.254 Zigate: WARNING - send_zigatecmd : [319] 0100 02c7190101000000000000010003 c719 ZigateQueue: 35
2021-11-09 21:09:06.255 Zigate: WARNING - send_zigatecmd : [320] 0100 02c7190101000000000000010006 c719 ZigateQueue: 36
2021-11-09 21:09:06.256 Zigate: WARNING - send_zigatecmd : [321] 0100 02c7190101000000000000010007 c719 ZigateQueue: 37
2021-11-09 21:09:06.256 Zigate: WARNING - send_zigatecmd : [322] 0100 02c719010100000000000001000a c719 ZigateQueue: 38
2021-11-09 21:09:06.257 Zigate: WARNING - send_zigatecmd : [323] 0100 02c719010100000000000001000f c719 ZigateQueue: 39
2021-11-09 21:09:06.258 Zigate: WARNING - send_zigatecmd : [324] 0100 02c7190101000000000000010010 c719 ZigateQueue: 40
2021-11-09 21:09:06.258 Zigate: WARNING - send_zigatecmd : [325] 0100 02c7190101000000000000010015 c719 ZigateQueue: 41
2021-11-09 21:09:06.259 Zigate: WARNING - send_zigatecmd : [326] 0100 02c7190101000000000000014000 c719 ZigateQueue: 42
2021-11-09 21:09:06.259 Zigate: WARNING - send_zigatecmd : [327] 0100 02c719010100000000000001f000 c719 ZigateQueue: 43
[... 17 lines cut ...]
2021-11-09 21:09:23.547 Zigate: ReadCluster 0019 - c719/01 Attribute: 0002 Type: 23 Size: 0004 Data: 00000004
2021-11-09 21:09:24.549 Zigate: ReadCluster 0019 - c719/01 Attribute: 0003 Type: 21 Size: 0002 Data: 0002
2021-11-09 21:09:24.952 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:09:24.952 Zigate: --> Empty queue
2021-11-09 21:09:25.551 Zigate: ReadCluster 0019 - c719/01 Attribute: 0004 Type: 23 Size: 0004 Data: ffffffff
2021-11-09 21:09:26.554 Zigate: ReadCluster 0019 - c719/01 Attribute: 0005 Type: 21 Size: 0002 Data: ffff
2021-11-09 21:09:27.555 Zigate: ReadCluster 0019 - c719/01 Attribute: 0006 Type: 30 Size: 0001 Data: 00
2021-11-09 21:09:28.550 Zigate: ReadCluster 0019 - c719/01 Attribute: 0007 Type: 21 Size: 0002 Data: 115f
2021-11-09 21:09:29.545 Zigate: ReadCluster 0019 - c719/01 Attribute: 0008 Type: 21 Size: 0002 Data: ffff
2021-11-09 21:09:30.546 Zigate: ReadCluster 0019 - c719/01 Attribute: 0009 Type: 21 Size: 0002 Data: 0000
 2021-11-09 21:09:34.927 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:09:34.927 Zigate: --> Empty queue
2021-11-09 21:09:44.942 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:09:44.943 Zigate: --> Empty queue
2021-11-09 21:09:54.970 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:09:54.970 Zigate: --> Empty queue
2021-11-09 21:09:58.682 Zigate: UpdateDevice - (Zigate - lumi.magnet.acn001_Door-54ef44100025b510-01) 1:Open
2021-11-09 21:10:04.975 Zigate: Assisted Pairing: Polling: []
2021-11-09 21:10:04.975 Zigate: --> Empty queue
```

I've updated the firmware of my Zigate USB (v1) to the last version : 3.20 => I always have the same problem.

Have you ideas ??
Thanks a lot.

pipiche38, 2021-11-14:
As requested on the Issue template, it is important to provide more information like the export.json of 'Plugin Device List RAW'
What is the plugin version and branch you are using ?

In between make sure to be on the latest beta version and check that it doesn't work.

rcrocus, 2021-11-14:
I use the plugin version 5.1.015 and the stable5 branch.
Please find below the export.json and the screen after the pairing from the plugin :
[export.json.txt](https://github.com/pipiche38/Domoticz-Zigate/files/7534634/export.json.txt)

[image]

pipiche38, 2021-11-15:
As requested please test with beta latest

rcrocus, 2021-11-15:
I'll test with beta ASAP.
After that can I easly return to stable5 branch ?

rcrocus, 2021-11-26:
Hello,

I test with beta latest (5.2.303) and I have the same problem.

pipiche38, 2021-11-26:
can you please retreive beta-5.2.305 and redo the test:
1/ git pull (to get the latest beta)
2/ remove the Voltage and Door widget created in Domoticz
3/ restart the plugin
4/ redo the pairing

rcrocus, 2021-11-26:
I have always the same problem.

pipiche38, 2021-11-26:
Could you please, enable some debug parameters via the WebUI > Tools > Debug

debugInput
debugCluster

The log will be visible on the Domoticz Logs

then try to interact with the door sensor. 

then drop the logs here

rcrocus, 2021-11-26:
Here the logs when I putsh the button :

 _2021-11-26 17:17:51.118 Zigate: ZigateRead - MsgType: 8102, MsgLength: 001f, MsgCRC: d4, Data: 48920b0100000005004200126c756d692e6d61676e65742e61636e303031, LQI: 153
2021-11-26 17:17:51.118 Zigate: Decode8102 - Attribute Reports: [920b:01] MsgSQN: 48 ClusterID: 0000 AttributeID: 0005 Status: 00 Type: 42 Size: 0012 ClusterData: >6c756d692e6d61676e65742e61636e303031<
2021-11-26 17:17:51.119 Zigate: scan_attribute_reponse - 8102 idx: 60 Read Attribute Response: [920b:01] ClusterID: 0000 MsgSQN: 48, i_sqn: None, AttributeID: 0005 Status: 00 Type: 42 Size: 0012 ClusterData: >6c756d692e6d61676e65742e61636e303031<
2021-11-26 17:17:51.119 Zigate: Decode8102 - LQI: 153 Received Cluster:0000 Attribute: 0005 Value: 6c756d692e6d61676e65742e61636e303031 from (920b/01)
2021-11-26 17:17:51.124 Zigate: ReadCluster - 0000 - 920b/01 AttrId: 0005 AttrType: 42 Attsize: 0012 Status: 00 AttrValue: 6c756d692e6d61676e65742e61636e303031
2021-11-26 17:17:51.124 Zigate: ReadCluster - 0000 / 0005 - Recepion Model: >lumi.magnet.acn001<
2021-11-26 17:17:51.125 Zigate: ReadCluster - 0000 / 0005 - 920b lumi.magnet.acn001 is already provisioned in Domoticz_

pipiche38, 2021-11-26:
when you push the button that is ok, but what about when you use the magnet and you simulate opening/closing a door

rcrocus, 2021-11-26:
When I push the button, I have these logs but nothing the status is not updated in Domoticz (only the time of the last contact was updated). When I simulate opening/closing a door, I have nothing in logs and no update in Domoticz.

pipiche38, 2021-11-26:
But I'm not talking about pushing any button. I'm taking about using the magnet

rcrocus, 2021-11-26:
Yes, when I using the magnet (opening or closing) nothing happens in log.

pipiche38, 2021-11-26:
Ok. so I'll purchase one in order to test and integrate

pipiche38, 2021-11-26:
can you give the link where you purchase it, so i'm sure to order the same ?

rcrocus, 2021-11-26:
I bought this one : https://fr.aliexpress.com/item/4001027772242.html?spm=a2g0o.productlist.0.0.76f7683f1aEGRg&algo_pvid=9a63c556-2f13-47c7-91f8-361eb7a0cf06&algo_exp_id=9a63c556-2f13-47c7-91f8-361eb7a0cf06-0&pdp_ext_f=%7B%22sku_id%22%3A%2210000013571930493%22%7D

pipiche38, 2021-11-26:
thanks to @koenkk https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5403#issuecomment-980436576

pipiche38, 2021-11-26:
@rcrocus the good news is that we know what has to be done, the issue is that I don't own the firmware of the ZiGate and that is where the changes has to be performed.
So when receiving the device I'll prepare a use case and submit the request for change

rcrocus, 2021-11-27:
Thanks a lot for your time and your work !

pipiche38, 2021-12-03:
If you have a Zigate V2 you can upgrade it to firmware 321 and it should eork

rcrocus, 2021-12-04:
Hello,
Unfortunately, I only have a Zigate V1. Is the firmware 321 is only for V2 ?

pipiche38, 2021-12-04:
Yes it is only for V2. Up to the zigate manufacturer to do it in V1.

You might want to ask for it here

https://github.com/fairecasoimeme/ZiGate/issues/381

Which I think make sense as you won’t be the only one in that situation

rcrocus, 2021-12-05:
I ask it yesterday...

SylvainPer, 2021-12-10:
@rcrocus available:
https://github.com/fairecasoimeme/ZiGate/releases/tag/v3.21-opdm
or 
https://github.com/fairecasoimeme/ZiGate/releases/tag/v3.21-legacy

rcrocus, 2021-12-10:
Hello,
After the upgrade of my Zigate v1 to v3.21, it works perfect.
Thanks a lot to everyone.

closed 2022-01-22 by pipiche38