zigbeefordomoticz/Domoticz-Zigbee #882: Multi click not working anymore on Mija buttons

mce35, 2021-11-20:
Hello,

After updating to the latest version of the beta branch, multi click is not working anymore on Mija buttons, only single click is working.
Branch stable5 is working.

Log for stable5 branch when double click on a button:
```
Nov 20 11:22:29 rasp3 domoticz: Zigate: (Zigate) ReadCluster - 0006 - 3d09/01 AttrId: 8000 AttrType: 20 Attsize: 0001 Status: 00 AttrValue: 02
Nov 20 11:22:29 rasp3 domoticz: Zigate: (Zigate) UpdateDevice - (ZG - btn cuisine) 1:10
Nov 20 11:22:30 rasp3 domoticz: Zigate: (Zigate) ReadCluster - ClusterId=0006 - reception General: On/Off: 02
```

Log on beta branch with 4-clicks:
```
Nov 20 11:12:23 rasp3 domoticz: Zigate: (Zigate) ReadCluster - 0006 - 3d09/01 AttrId: 8000 AttrType: 20 Attsize: 0001 Status: 00 AttrValue: 04
Nov 20 11:12:23 rasp3 domoticz: Zigate: (Zigate) ReadCluster - ClusterId=0006 - NwkId: 3d09 Ep: 01 Attr: 8000 Value: 04 (if something doesn't work anymore, please contact @pipiche
```

Thank you.

pipiche38, 2021-11-20:
Thank. Can you kindly provide me also an export.json ( Plugin Device list (RAW)  )

I’ll fix it by Monday

mce35, 2021-11-20:
Here it is.

Thank you.

[export.json](https://github.com/pipiche38/Domoticz-Zigate/files/7574854/export.json.txt)

pipiche38, 2021-11-21:
beta-5.2.291 should be ok. Thanks for reporting

closed 2021-11-22 by pipiche38

mce35, 2021-11-23:
> beta-5.2.291 should be ok. Thanks for reporting

Just updated to the latest beta branch (5.2.294 by now), the fix is working, thank you!