zigbeefordomoticz/Domoticz-Zigbee #892: [zigpy-integration] - what is the purpose of the sequence parameter

pipiche38, 2021-12-14:
In request()/mrequest() sequence is an expected parameter which is not used in the raw_aps_data_request(), for ZNP it seems that the sequence called TSN is expected, does that mean that we have to have a dedicated counter/sqn ?

Is that a simple counter, do we get it back in the call_handler ?

badzz, 2021-12-14:
interesting subject. The NXP function does not allow the manipulation of the ZPS sequence number.
It is the NXP stack that assigns them and keep the counter.  check ZPS_eAplAfApsdeDataReq for reference in NXP doc
Hence when using zigate, no sequence number needed to be send. 
However to keep track of ACK, the NXP stack and zigate firmware sends back the SQN assigned

pipiche38, 2021-12-15:
get_sequence() is provided by zigpy.application so everything good.

closed 2021-12-15 by pipiche38