zigbeefordomoticz/Domoticz-Zigbee #897: [zigpy-integration] - ZCL SQN management

pipiche38, 2021-12-15:
With zigpy we don't have to managed APS sqn , but via the raw 0x0530 (the sqn is embeeded into the RAW_APS payload). These is the ZCL/ZDP sqn we will get back when receiving the answer from the device.

pipiche38, 2021-12-19:
https://github.com/badzz/zigpy-zigate/commit/6f22a49ad17635b1c1fec36826bf62a6afd5fbca

badzz, 2021-12-21:
delivered .. there might some perf issues but let's make a separate bug for it ?

pipiche38, 2021-12-21:
@badz that one is still open, tI need to fix the use of the ZCL SQN instead of the i_sqn . like for configure reporting and bind responses.

pipiche38, 2021-12-26:
Issue remains on Zigpy-ZNP as the SQN seems to be overwritten .
https://github.com/pipiche38/Domoticz-Zigate/issues/932

badzz, 2021-12-26:
rewording the issue to APS SQN would be better ?

pipiche38, 2021-12-26:
I close that one, as the pending issue is under #932

closed 2021-12-26 by pipiche38