zigbeefordomoticz/Domoticz-Zigbee #905: [zigpy-integration] - zigate issue: zigpy-zigate doesn't handle 0x9999

pipiche38, 2021-12-16:
zigpy-zigate doesn't handle 0x9999

Dec 16 09:40:09 rasp domoticz[1295]: 2021-12-16:09:40:09,175 WARNING  [api.py:175] Received unhandled response 0x9999

pipiche38, 2021-12-16:
and last the message should be forwared to a call handler

badzz, 2021-12-22:
so , I can catch it but I am not sure how to sent it back with request return  . eg: if we send back the extended code instead of normal code, we might not know which layer is sending it back , eg 0x83 error code exists in ZDP error codes and extended codes.
If you do not need the details, I could just dump it in the logs.

pipiche38, 2021-12-22:
Cannot we have a dedicated call handler for it ?

badzz, 2021-12-22:
That would be specific to zigate ? This is low level hardware specific info.
What actions would higher layer take based on this error ?
What other hardware would send back ?

pipiche38, 2021-12-22:
You right, I suggest that we simply log a WARNING at the zigpy_zigate level

closed 2022-01-03 by pipiche38