zigbeefordomoticz/Domoticz-Zigbee #914: [zigpy-integration] - error when pairing Opple device - Cluster 51 has not been implemented !

pipiche38, 2021-12-19:
```
Dec 19 12:22:28 rasp domoticz[31682]: 2021-12-19:12:22:28,686 DEBUG    [application.py:1238] Intercepted a ZDO request: dst_addr=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x003D), dst_ep=0, src_ep=0, cluster=51, sequence=14, options=TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST, radius=30, data=b'\x01\x00'
Dec 19 12:22:28 rasp domoticz[31682]: 2021-12-19:12:22:28,688 ERROR    [application.py:1262] ZDO converter for cluster 51 has not been implemented! Please open a GitHub issue and attach a debug log: https://github.com/zigpy/zigpy-znp/issues/new
Dec 19 12:22:28 rasp domoticz[31682]: 2021-12-19 12:22:28.689  Error: Zigpy Plugin: 'worker_loop' failed 'No ZDO converter'
```

pipiche38, 2021-12-19:
https://github.com/zigpy/zigpy-znp/issues/112

pipiche38, 2021-12-19:
This is mainly link to the fact that the Domoticz plugin at the completion of the discovery/interview request the binding table of the device

zigbeefordomoticz, 2022-01-01:
https://github.com/zigpy/zigpy-znp/pull/110

closed 2022-01-01 by zigbeefordomoticz