zigbeefordomoticz/Domoticz-Zigbee #919: [zigpy-integration] - znp

pipiche38, 2021-12-20:
if you pair a device it works very well, then if you restart the plugin , it looks like ZNP layer lost its memory
and the device seems not recognized and do not handle the message
```
Dec 20 16:18:46 rasp domoticz[23941]: Received an AF message from an unknown device: 0xCC84
Dec 20 16:18:46 rasp domoticz[23941]: Received an AF message from an unknown device: 0xCC84
Dec 20 16:18:47 rasp domoticz[23941]: 2021-12-20 16:18:47.733  Zigpy Plugin: get_device nwk 0xCC84 ieee None
Dec 20 16:18:52 rasp domoticz[23941]: Received an AF message from an unknown device: 0xCC84
Dec 20 16:18:52 rasp domoticz[23941]: Received an AF message from an unknown device: 0xCC84
```

pipiche38, 2021-12-20:
@badzz I beleive this is link to get_device()

badzz, 2021-12-21:
https://github.com/pipiche38/Domoticz-Zigate/pull/920

closed 2021-12-23 by pipiche38