zigbeefordomoticz/Domoticz-Zigbee #930: [zigpy-integration] - stack trace during pairing of aqara switch

pipiche38, 2021-12-25:
```
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.461  Error: Zigpy: 'worker_loop' failed ''
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.481  Error: Zigpy: ----> Line 132 in '/var/lib/domoticz/plugins/Domoticz-zigpy/Classes/ZigpyTransport/zigpyThread.py', function worker_loop
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.501  Error: Zigpy: ----> Line 209 in '/var/lib/domoticz/plugins/Domoticz-zigpy/Classes/ZigpyTransport/zigpyThread.py', function process_raw_command
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.521  Error: Zigpy: ----> Line 455 in '/var/lib/domoticz/plugins/Domoticz-zigpy/zigpy_znp/zigbee/application.py', function request
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.542  Error: Zigpy: ----> Line 1485 in '/var/lib/domoticz/plugins/Domoticz-zigpy/zigpy_znp/zigbee/application.py', function _send_request
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.561  Error: Zigpy: ----> Line 1328 in '/var/lib/domoticz/plugins/Domoticz-zigpy/zigpy_znp/zigbee/application.py', function _send_request_raw
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.581  Error: Zigpy: ----> Line 1289 in '/var/lib/domoticz/plugins/Domoticz-zigpy/zigpy_znp/zigbee/application.py', function _send_zdo_request
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.601  Error: Zigpy: ----> Line 55 in '/usr/lib/python3.9/site-packages/async_timeout/__init__.py', function __aexit__
Dec 25 15:46:45 rasp domoticz[27346]: 2021-12-25 15:46:45.620  Error: Zigpy: ----> Line 92 in '/usr/lib/python3.9/site-packages/async_timeout/__init__.py', function _do_exit
Dec 25 15:46:46 rasp kernel: i2c-bcm2835 3f805000.i2c: i2c transfer timed out
```

pipiche38, 2021-12-25:
https://github.com/zigpy/zigpy/discussions/865#discussioncomment-1869839

zigbeefordomoticz, 2022-01-01:
close it for now.

closed 2022-01-01 by zigbeefordomoticz