zigbeefordomoticz/Domoticz-Zigbee #933: [zigpy-integration] - unable to retreive Active Ep and Simple descriptor of the Controler

pipiche38, 2021-12-26:
Unable to retrieve the Active Ep and simple descriptor of the Controller!

pipiche38, 2021-12-26:
When receiving 0x8009 ( Network Status) and we know that the controller is well initialized we start retreiving information in order to get the list of Endpoint and supported cluster

This doesn't work with zigpy layers either with Zigate where we get a message like:
```
2021-12-26 09:32:16,100 DEBUG   : [      ZigpyCom_18] got command {'cmd': 'RAW-COMMAND', 'datas': {'Profile': 0, 'Cluster': 5, 'TargetNwk': 0, 'TargetEp': 0, 'SrcEp': 0, 'Sqn': '00', 'payload': '000000', 'AddressMode': 2}, 'NwkId': None, 'TimeStamp': 1640507535.8846037, 'ACKIsDisable': False, 'Sqn': None}
2021-12-26 09:32:16,103 DEBUG   : [      ZigpyCom_18] ZigyTransport: process_raw_command ready to request NwkId: 0000 Cluster: 0005 Seq: 01 Payload: b'\x00\x00\x00' AddrMode: 02 EnableAck: True, Sqn: None
2021-12-26 09:32:16,142 DEBUG   :get_device
2021-12-26 09:32:16,145 DEBUG   :get_device
2021-12-26 09:32:16,146 ERROR   : [      ZigpyCom_18] handle_message Sender unkown device : <Device model=None manuf=None nwk=0x0000 ieee=None is_initialized=False> Profile: 0000 Cluster: 8005 sEP: 0 dEp: 0 message: b'\x00\x00\x00\x00\x07\x01\x0b\nn\x15\x08\x03'
2021-12-26 09:32:16,159 ERROR   : [      ZigpyCom_18] handle_message Sender unkown device : <Device model=None manuf=None nwk=0x0000 ieee=None is_initialized=False> Profile: 0000 Cluster: 8005 sEP: 0 dEp: 0 message: b'\x18\x00\x0b\x00\x00'
2021-12-26 09:32:23,160 INFO    : [      ZigpyCom_18] process_raw_command (zigpyThread) spend more than 1s (7060 ms) frame: {'Profile': 0, 'Cluster': 5, 'TargetNwk': 0, 'TargetEp': 0, 'SrcEp': 0, 'Sqn': '00', 'payload': '000000', 'AddressMode': 2} with Ack: False
```

Or with ZNP here I think we do not retreive the Controller messages !

Question is how to get those information at the plugin level ?

pipiche38, 2021-12-26:
@badzz

badzz, 2022-01-01:
The endpoints are already in the 0000 device that are populated in znp
for ep_id, endpoint in (self.app.get_device(nwk = t.NWK(0x0000)).endpoints.items)
    endpoint.out_clusters
we can do the same thing for zigate

pipiche38, 2022-01-01:
Ok. It works, I'll do an encoding to send it in the proper format to plugin

pipiche38, 2022-01-01:
Done! I feel that the list of clusters provided are not as they should be.

[image]

I would expect all supported clusters like 0006, 0008, 0102, 0201, 0004 ....

badzz, 2022-01-01:
this is the list of cluster for the controller itself, not the clusters it supports on devices.

pipiche38, 2022-01-01:
You are right, but comparing to Zigate it is very light!

closed 2022-01-01 by zigbeefordomoticz