zigbeefordomoticz/Domoticz-Zigbee #939: [zigpy-integration] - get_device() for device, but what for group ?

pipiche38, 2021-12-28:
get_device() allows to transform a NwkId or IEEE into a zigpy device. But what about group ? How can we transform a plugin groupId into a zigpy group ?

pipiche38, 2021-12-28:
destination = t.AddrModeAddress(mode=t.AddrMode.Group, address=NwkId)

pipiche38, 2021-12-28:
AddrModeAddress seems to be defined at ZNP but not zigpy level!

badzz, 2022-01-01:
why would you need to call get_device directly ?
you can call mrequest with a groupid for instance ?

zigbeefordomoticz, 2022-01-02:
here is the logs when issuing a mrequest, which I beleive is linked to https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/939#issuecomment-1002061225

```
2022-01-02 09:50:11,002 DEBUG   : [      ZigpyCom_18] ZigyTransport: process_raw_command ready to request NwkId: 573c Cluster: 0006 Seq: 2c Payload: 110001 AddrMode: 01 EnableAck: True, Sqn: 0
2022-01-02 09:50:11,005 ERROR   : [      ZigpyCom_18] Error while receiving a Plugin command: >module 'zigpy.types' has no attribute 'AddrModeAddress'<
2022-01-02 09:50:11,048 ERROR   : [      ZigpyCom_18] Issue in request, dumping stack
2022-01-02 09:50:11,104 ERROR   : [      ZigpyCom_18] ==>  {'cmd': 'RAW-COMMAND', 'datas': {'Profile': 260, 'Cluster': 6, 'TargetNwk': 22332, 'TargetEp': 1, 'SrcEp': 1, 'Sqn': 0, 'payload': '110001', 'AddressMode': 1}, 'NwkId': 22332, 'TimeStamp': 1641113410.903925, 'ACKIsDisable': False, 'Sqn': 0}
2022-01-02 09:50:11,164 ERROR   : [      ZigpyCom_18] Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-zigpy/Classes/ZigpyTransport/zigpyThread.py", line 134, in worker_loop
    await dispatch_command( self, data)
  File "/var/lib/domoticz/plugins/Domoticz-zigpy/Classes/ZigpyTransport/zigpyThread.py", line 225, in dispatch_command
    await process_raw_command(self, data["datas"], AckIsDisable=data["ACKIsDisable"], Sqn=data["Sqn"])
  File "/var/lib/domoticz/plugins/Domoticz-zigpy/Classes/ZigpyTransport/zigpyThread.py", line 267, in process_raw_command
    destination = t.AddrModeAddress(mode=t.AddrMode.Group, address=NwkId)
AttributeError: module 'zigpy.types' has no attribute 'AddrModeAddress'
```

badzz, 2022-01-03:
https://github.com/zigbeefordomoticz/Domoticz-Zigbee/pull/974

closed 2022-01-03 by pipiche38