zigbeefordomoticz/Domoticz-Zigbee #940: [zigpy-integration– ZNP Endpoints

pipiche38, 2021-12-28:
in order to align ZNP controller on the Zigate, we must make sure that ZNP is listening on the same Endpoint as Zigate

EndPoint 0x00 (0) --> ZDP
EndPoint 0x01  (1) --> Standard
EndPoint 0x0b (11) --> Wiser Schneider
EndPoint 0x0a  (10)
EndPoint 0x6E (110)   --> 
EndPoint 0x15 (21)   --->
EndPoint 0x08 (8) -->
EndPoint 0x03 (3) --> New Wiser Schneider

https://github.com/zigpy/zigpy/discussions/865#discussioncomment-1879846

pipiche38, 2021-12-30:
For Wiser Thermostat, we will have register the endPoint ( https://github.com/zigpy/zigpy/discussions/865#discussioncomment-1879846 )

Same issue as on ZiGate EndPoint receiving command needs to be registered in order to capture the event!

Need to see how to do that in the integration side

pipiche38, 2021-12-30:
@badzz to be check if we can use _register_endpoints() from AppZNP during startup and hving the possibility to add additional Endpoint on top of the 0x00, 0x01 and 0x02

badzz, 2021-12-30:
Yep seen that..  I will dig into it . Thx

badzz, 2022-01-01:
https://github.com/zigbeefordomoticz/Domoticz-Zigbee/pull/958

closed 2022-01-03 by pipiche38