zigbeefordomoticz/Domoticz-Zigbee #965: Enhacement of debuging facility

zigbeefordomoticz, 2022-01-02:
It would be nice to be able to specify the Thread we want to get logs. In such we could check all logs for a particular thread.

zigbeefordomoticz, 2022-01-02:
@SylvainPer

SylvainPer, 2022-01-02:
Just one thread or multiple choice?

pipiche38, 2022-01-02:
I think the idea might be to have 
debugThreadForwarder
debugThreadDomoticz
debugThreadZigpy
debugThreadWriter (native Zigate)
debugThreadSerial (native Zigate)

and in that case, this will overwrite the other debug flags

SylvainPer, 2022-01-04:
#980

closed 2022-01-05 by pipiche38

Hedda, 2022-01-19:
FYI, ZHA also features the option to separately specify debug logging per radio library for zigpy and other dependencies:

https://www.home-assistant.io/integrations/zha#debug-logging

### Debug logging

To enable debug logging for ZHA component and radio libraries, add the following [logger](https://www.home-assistant.io/integrations/logger/) configuration to `configuration.yaml`:

```yaml
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```

pipiche38, 2022-01-19:
@hedda thanks for the information, please do not that Domoticz is not ZHA, but we are pretty aware on how to set the debug.