zigbeefordomoticz/Domoticz-Zigbee #1008: [zigpy-integration] - zigpy will require Python version 3.7 and soon maybe even require at least 3.8

Hedda, 2022-01-12:
FYI, see zigpy devopers discussion about dropping support for python-version 3.6 and 3.7 (3.7.7):

https://github.com/zigpy/zigpy/pull/861

https://github.com/zigpy/zigpy/pull/861/commits/37671f8c857c987d42ad50387fbe9a5ca44d61d0

As I understand their strategy is drop python-version 3.6 support now and also plan to drop 3.7 soon.

https://github.com/zigpy/zigpy/blob/dev/.github/workflows/ci.yml

Noted that plugin currently only requires Python version 3.5 according to zigbeefordomoticz wiki:

https://github.com/zigbeefordomoticz/wiki/blob/zigpy/en-eng/Plugin_Installation.md

https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/Plugin_Installation.md

> In every installation methodes, the plugin requires Domoticz version 3.87xx or greater and Python version 3.5 or higher.
Make sure that you have __python3-dev package__ installed. More informations on [Domoticz Python page](https://www.domoticz.com/wiki/Using_Python_plugins).

pipiche38, 2022-01-12:
thanks for the information. 
From a domoticz standpoint and the plugin we don't have any requirements on that matters. We just try to be conservative and keep on friends running on old platform happy.

pipiche38, 2022-01-15:
Noted and documented in the Readme section.

closed 2022-01-15 by pipiche38