zigbeefordomoticz/Domoticz-Zigbee #1016: [zigpy-integration] ValueError at plugin startup

SylvainPer, 2022-01-15:
At plugin startup:
`
2022-01-15 09:56:56,109 DEBUG   : [       ZigpyCom_2] build_plugin_8043_frame_list_node_descriptor 1 <Endpoint id=1 in=[basic:0x0000, ota:0x0019] out=[ias_zone:0x0500, ias_wd:0x0502] status=<Status.ZDO_INIT: 1>>
2022-01-15 09:56:56,109 DEBUG   : [       ZigpyCom_2] AppZnp - get_device ieee:None nwk:0x0000 
2022-01-15 09:56:56,110 DEBUG   : [       ZigpyCom_2] AppZnp - get_device found device: <ZNPCoordinator model='CC1352/CC2652, Z-Stack 3.30+ (build 20211217)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:23:8d:ab:c9 is_initialized=True>
2022-01-15 09:56:56,110 DEBUG   : [       ZigpyCom_2] Controller Model CC1352/CC2652, Z-Stack 3.30+ (build 20211217)
2022-01-15 09:56:56,110 DEBUG   : [       ZigpyCom_2] AppZnp - get_device ieee:None nwk:0x0000 
2022-01-15 09:56:56,110 DEBUG   : [       ZigpyCom_2] AppZnp - get_device found device: <ZNPCoordinator model='CC1352/CC2652, Z-Stack 3.30+ (build 20211217)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:23:8d:ab:c9 is_initialized=True>
2022-01-15 09:56:56,110 DEBUG   : [       ZigpyCom_2] Controller Manufacturer Texas Instruments
2022-01-15 09:56:56,111 DEBUG   : [       ZigpyCom_2] worker_loop - ZigyTransport: worker_loop start.
2022-01-15 09:56:56,111 DEBUG   : [       ZigpyCom_2] got a command RAW-COMMAND
2022-01-15 09:56:56,111 DEBUG   : [       ZigpyCom_2] RAW-COMMAND: {'Profile' : 0000,'Cluster' : 8002,'TargetNwk' : 0000,'TargetEp' : 00,'SrcEp' : 00,'Sqn' : 01,'payload' : 01890000,'AddressMode' : 02,}
2022-01-15 09:56:56,101 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.NO_DESCRIPTOR: 137>, 0x0000, None]
2022-01-15 09:56:56,112 DEBUG   :[0x0000:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_rsp([<Status.NO_DESCRIPTOR: 137>, 0x0000, None])
2022-01-15 09:56:56,111 DEBUG   : [       ZigpyCom_2] ZigyTransport: process_raw_command ready to request NwkId: 0000 Cluster: 8002 Seq: 01 Payload: 01890000 AddrMode: 02 EnableAck: True, Sqn: 1
2022-01-15 09:56:56,112 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x0000), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=32770, TSN=1, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x89\x00\x00')
2022-01-15 09:56:56,113 DEBUG   : [       ZigpyCom_2] AppZnp - get_device ieee:None nwk:0x0000 
2022-01-15 09:56:56,115 DEBUG   : [       ZigpyCom_2] AppZnp - get_device found device: <ZNPCoordinator model='CC1352/CC2652, Z-Stack 3.30+ (build 20211217)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:23:8d:ab:c9 is_initialized=True>
2022-01-15 09:56:56,115 DEBUG   : [       ZigpyCom_2] process_raw_command  call request destination: <ZNPCoordinator model='CC1352/CC2652, Z-Stack 3.30+ (build 20211217)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:23:8d:ab:c9 is_initialized=True> Profile: 0 Cluster: 32770 sEp: 0 dEp: 0 Seq: 1 Payload: b'\x01\x89\x00\x00'
2022-01-15 09:56:56,115 DEBUG   : [ ZigpyForwarder_2] ZigateRead - MsgType: 8015,  Data: , LQI: 0
2022-01-15 09:56:56,116 INFO    : [ ZigpyForwarder_2] Number of devices recently active in Zigate = 0
2022-01-15 09:56:56,116 DEBUG   : [ ZigpyForwarder_2] Decode8015 - IEEE2NWK      : {'00124b00238dabc9': '0000', '00124b0002214bae': 'd037', '00124b0002ce39cc': '6e12'}
2022-01-15 09:56:56,117 DEBUG   : [ ZigpyForwarder_2] ZigateRead - MsgType: 8010,  Data: 20211217, LQI: 0
2022-01-15 09:56:56,117 DEBUG   : [ ZigpyForwarder_2] Decode8010 - Reception Version list:20211217 Branch: 20 Major: 21 Version: 1217
2022-01-15 09:56:56,117 INFO    : [ ZigpyForwarder_2] Texas Instrument ZNP CC1352/CC2652, Z-Stack 3.30+
2022-01-15 09:56:56,118 INFO    : [ ZigpyForwarder_2] Installer Version Number: 1217
2022-01-15 09:56:56,119 INFO    : [ ZigpyForwarder_2] Branch Version: ==> zigpy-znp 3.x <==
2022-01-15 09:56:56,120 DEBUG   : [ ZigpyForwarder_2] ZigateRead - MsgType: 8009,  Data: 000000124b00238dabc951091234484504015e100f, LQI: 0
2022-01-15 09:56:56,120 DEBUG   : [ ZigpyForwarder_2] Decode8009: Network state - Address:0000 extaddr:00124b00238dabc9 PanID: 5109 Channel: 15
2022-01-15 09:56:56,121 DEBUG   : [       ZigpyCom_2] AppZnp - get_device ieee:None nwk:0x0000 
2022-01-15 09:56:56,121 DEBUG   : [       ZigpyCom_2] AppZnp - get_device found device: <ZNPCoordinator model='CC1352/CC2652, Z-Stack 3.30+ (build 20211217)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:23:8d:ab:c9 is_initialized=True>
2022-01-15 09:56:56,121 DEBUG   : [       ZigpyCom_2] handle_message from Controller Sender: 0x0000 Profile: 0000 Cluster: 8002 srcEp: 00 dstEp: 00 message: 01890000
2022-01-15 09:56:56,122 INFO    : [ ZigpyForwarder_2] Zigate addresses ieee: 00124b00238dabc9 , short addr: 0000
2022-01-15 09:56:56,123 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-01-15 09:56:56,123 INFO    : [ ZigpyForwarder_2] Network state UP, PANID: 5109 extPANID: 0x1234484504015e10 Channel: 15
2022-01-15 09:56:56,124 ERROR   :Received an exception while passing frame to API: TransportFrame(payload=GeneralFrame(header=CommandHeader(id=0x82, subsystem=Subsystem.ZDO, type=CommandType.AREQ), data=b'\x00\x00\x89\x00\x00'))
Traceback (most recent call last):
  File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/zigpy_znp/types/commands.py", line 414, in from_frame
    params[param.name], data = param.type.deserialize(data)
  File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/zigpy_znp/commands/zdo.py", line 129, in deserialize
    return super().deserialize(data)
  File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/zigpy/types/struct.py", line 228, in deserialize
    raise ValueError(f"Data is too short to contain {bitfields}")
ValueError: Data is too short to contain [StructField(name='logical_type', type=<enum 'LogicalType'>, requires=None), StructField(name='complex_descriptor_available', type=<class 'zigpy.types.basic.uint1_t'>, requires=None), StructField(name='user_descriptor_available', type=<class 'zigpy.types.basic.uint1_t'>, requires=None), StructField(name='reserved', type=<class 'zigpy.types.basic.uint3_t'>, requires=None)]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/zigpy_znp/uart.py", line 79, in data_received
    self._api.frame_received(frame.payload)
  File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/zigpy_znp/api.py", line 677, in frame_received
    command = command_cls.from_frame(frame, align=self.nvram.align_structs)
  File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/zigpy_znp/types/commands.py", line 422, in from_frame
    f"Frame data is truncated (parsed {params}),"
ValueError: Frame data is truncated (parsed {'Src': 0x0000, 'Status': <Status.NO_DESCRIPTOR: 137>, 'NWK': 0x0000}), required parameter remains: Param(name='NodeDescriptor', type=<class 'zigpy_znp.commands.zdo.NullableNodeDescriptor'>, description='Node descriptor', optional=False)
2022-01-15 09:56:56,126 DEBUG   : [ ZigpyForwarder_2] ZigateRead - MsgType: 8045,  Data: 0000000009000308156e0a0b0201, LQI: 0
2022-01-15 09:56:56,130 DEBUG   : [ ZigpyForwarder_2] ZigateRead - MsgType: 8043,  Data: 000000000103010408200001000000, LQI: 0
2022-01-15 09:56:56,130 DEBUG   : [ ZigpyForwarder_2] ZigateRead - MsgType: 8043,  Data: 000000000108010408200001000000, LQI: 0
`

pipiche38, 2022-01-15:
https://github.com/zigpy/zigpy/discussions/865#discussioncomment-1974380

pipiche38, 2022-01-15:
git pull --recurse-submodules
to get the fix: https://github.com/zigpy/zigpy-znp/commit/dd61b11d43c3f997afb92cc7e439353372946421

closed 2022-01-15 by pipiche38