zigbeefordomoticz/Domoticz-Zigbee #1018: [zigpy-integration] - Topology network missing some end devices

pipiche38, 2022-01-15:
There is an issue in the bitfield deccoding in buildframe_management_lqi_response() and sometime End devices are seen as Routers.

The issue is that the original frame received bitfield1 and bitfield2, but the plugin expect only 1 bitfield (due to NXP stack)

jp-keros, 2022-01-15:
The lor error : 

{"0":{"0":{"Time":"15/01/2022 16:32:28","Module":"NetworkMap","nwkid":"None","PluginHealth":"Ready","message":"Found 5f41 in a Neighbour table tag as a router, but is not","PermitToJoin":{"Duration":0,"Starttime":1642260600},"Thread":"ZigpyForwarder_2"},"1":{"Time":"15/01/2022 16:32:30","Module":"NetworkMap","nwkid":"None","PluginHealth":"Ready","message":"Found f63b in a Neighbour table tag as a router, but is not","PermitToJoin":{"Duration":0,"Starttime":1642260600},"Thread":"ZigpyForwarder_2"},"LastLog":1,"StartTime":"15/01/2022 16:29:18","FirmwareVersion":"1217","FirmwareMajorVersion":"21","PluginVersion":"zigpy-6.0.002"},"LastLog":0}

My device export :

[{"Status":"","RIA":"10","PowerSource":"Main","ZDeviceName":"Zigbee Coordinator","Version":"3","Ep":{"15":{"0000":{}},"00":{},"03":{"0000":{}},"08":{"0000":{}},"6e":{"0000":{}},"0a":{"0000":{}},"0b":{"0000":{}},"02":{"0000":{}},"01":{"0000":{},"0019":{},"0500":{},"0502":{}}},"IEEE":"00124b00238da09f","LogicalType":"Coordinator","Health":"","ConsistencyCheck":"","LQI":{},"Max Buffer Size":"50","Max Rx":"00a0","Max Tx":"00a0","macapa":"8f","bitfield":"4000","server_mask":"2c01","descriptor_capability":"00","Manufacturer":"0000","DeviceType":"FFD","ReceiveOnIdle":"On","Stamp":{"LasteSeen":{},"Time":"2022-01-15 16:32:30","MsgType":"8002","time":1642260750.144307},"SQN":"00","ZDPSQN":"04","NeighbourTableSize":3,"Neighbours":[{"Time":"2022-01-15 16:30:55","Devices":[{"611b":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":63,"_rxonwhenidl":"Rx-Off","_IEEE":"842e14fffe3666e4","_permitjnt":"??"}}]},{"Time":"2022-01-15 16:32:15","Devices":[{"611b":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":61,"_rxonwhenidl":"Rx-Off","_IEEE":"842e14fffe3666e4","_permitjnt":"??"}},{"5f41":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":163,"_rxonwhenidl":"Rx-Off","_IEEE":"0000000000000000","_permitjnt":"??"}},{"f63b":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":127,"_rxonwhenidl":"Rx-Off","_IEEE":"0000000000000000","_permitjnt":"??"}}]},{"Time":"2022-01-15 16:32:40","Devices":[{"611b":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":62,"_rxonwhenidl":"Rx-Off","_IEEE":"842e14fffe3666e4","_permitjnt":"??"}},{"5f41":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":163,"_rxonwhenidl":"Rx-Off","_IEEE":"0000000000000000","_permitjnt":"??"}},{"f63b":{"_relationshp":"Sibling","_devicetype":"Router","_depth":255,"_lnkqty":127,"_rxonwhenidl":"Rx-Off","_IEEE":"0000000000000000","_permitjnt":"??"}}]}],"MapLQI":{"611b":82},"NwkID":"0000"},{"Version":"3","ZDeviceName":"Sonde Aqara","Status":"inDB","SQN":"36","Ep":{"01":{"0000":{"0005":"lumi.weather","0001":"5","ff01":"0121b30b0421a81305210700062400000000006429ea076521970f662ba38901000a210000"},"0003":{},"0402":{"0000":2044},"0403":{"0000":1007,"0014":"ff","0010":1007.7},"0405":{"0000":39.6},"ffff":{},"Type":"Temp/Humi/Baro","ClusterType":{"3":"Temp+Hum+Baro","4":"Temp+Hum","5":"Temp","6":"Humi","7":"Baro"},"0001":{"0000":2995}}},"Heartbeat":"3802","RIA":"0","LQI":116,"Battery":61,"Model":"lumi.weather","ForceAckCommands":[],"MacCapa":"80","IEEE":"00158d00056e64b3","Type":"Temp/Humi/Baro","ProfileID":"0104","ZDeviceID":"0302","App Version":"5","Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"","Manufacturer Name":"","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":"15/01/2022 16:31:50","iSQN":{"0004":"00","0005":"01","ff01":"02","ff02":"03"},"Attributes":{"0001":"00","ff01":"00","0005":"00"},"ZigateRequest":{}},"0405":{"TimeStamp":"15/01/2022 16:33:31","iSQN":{},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0402":{"TimeStamp":"15/01/2022 16:33:31","iSQN":{},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0403":{"TimeStamp":"15/01/2022 16:33:31","iSQN":{},"Attributes":{"0000":"00","0014":"00","0010":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1642260811.8798354,"Time":"2022-01-15 16:33:31","MsgType":"8102","LastSeen":"15/01/2022 16:33:31"},"ZCL Version":"","Health":"Live","Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"ffff":{},"0402":{},"0403":{},"0405":{}},"ClusterOut":{"0000":{},"0004":{},"ffff":{}},"ProfileID":"0104","ZDeviceID":"0302"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0402":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0403":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0405":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"ffff":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ConfigSource":"DeviceConf","LUMI":{"RSSI dB":-249,"LQI":"0000","BatteryVoltage":2995},"BindingTable":{"Devices":[],"SQN":1,"TimeStamp":"15/01/2022 11:17:00"},"ConsistencyCheck":"ok","Param":{},"CheckParam":false,"RollingLQI":[127,127,120,120,116],"BatteryUpdateTime":"15/01/2022 16:31:50","MapLQI":{"0000":127},"NwkID":"f63b"},{"Version":"3","ZDeviceName":"Bouton Aqara","Status":"inDB","SQN":"10","Ep":{"01":{"0000":{"0005":"lumi.sensor_switch.aq3","0001":"5","ff01":"0121bd0b0328190421a8130521190006240100000 [... cut ...] sted","Status":"","i_sqn":"04"}}},"Tuya":{"TuyaTransactionId":5,"MaxAlarmHumi":"00000050","Humidity":"00000024","Temperature":"000000cf","TUYA_MCU_VERSION_RSP":"40","TemperatureUnit":"01","SirenDuration":"00000005","SirenVolume":"02","SirenMelody":"01","Alarm":"00","PowerMode":"04"},"BindingTable":{"Devices":[],"SQN":1,"TimeStamp":"15/01/2022 11:18:35"},"CheckParam":false,"ConsistencyCheck":"ok","RollingLQI":[65,61,61,65,65,61,61,65,65,61,61],"SQN_000a":"5b","MapLQI":{"611b":50},"NwkID":"fda8"}]

jp-keros, 2022-01-16:
I still have the issue : 

The DomoticZ Logs :

2022-01-16 13:54:49.105 Status: ZigBee: (ZigBee) Network Topology report
2022-01-16 13:54:49.111 Status: ZigBee: (ZigBee) ------------------------------------------------------------------------------------------
2022-01-16 13:54:49.116 Status: ZigBee: (ZigBee)
2022-01-16 13:54:49.118 Status: ZigBee: (ZigBee) Node Node Relation Type Deepth LQI Rx-Idle PmtJoin
2022-01-16 13:54:49.119 Status: ZigBee: (ZigBee) 0000 611b Sibling Router 255 173 ?? ??
2022-01-16 13:54:49.120 Status: ZigBee: (ZigBee) 0000 f63b Sibling Router 255 140 ?? ??
2022-01-16 13:54:49.121 Status: ZigBee: (ZigBee) 611b 0000 Parent Coordinator 0 200 Rx-On ??
2022-01-16 13:54:49.121 Status: ZigBee: (ZigBee) 611b fda8 Child End Device 2 158 Rx-Off Off
2022-01-16 13:54:49.123 Status: ZigBee: (ZigBee) --

jp-keros, 2022-01-16:
The Logs with DebugNetworkMap activated :

2022-01-16 16:37:00,869 DEBUG   :Sending request: SYS.Ping.Req()
2022-01-16 16:37:00,881 DEBUG   :Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-01-16 16:37:20,840 DEBUG   :Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x611B, Relays=[])
2022-01-16 16:37:20,845 DEBUG   : [       ZigpyCom_2] AppZnp - get_device ieee:None nwk:0x611B 
2022-01-16 16:37:20,846 DEBUG   : [       ZigpyCom_2] AppZnp - get_device found device: <Device model=None manuf=None nwk=0x611B ieee=84:2e:14:ff:fe:36:66:e4 is_initialized=False>
2022-01-16 16:37:20,891 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1794, SrcAddr=0x611B, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=174, SecurityUse=<Bool.false: 0>, TimeStamp=12896260, TSN=0, Data=b'\x08\x55\x0A\x00\x00\x25\xB2\x08\x00\x00\x00\x00', MacSrcAddr=0x611B, MsgResultRadius=29)
2022-01-16 16:37:20,899 DEBUG   : [       ZigpyCom_2] AppZnp - get_device ieee:None nwk:0x611B 
2022-01-16 16:37:20,905 DEBUG   : [       ZigpyCom_2] AppZnp - get_device found device: <Device model=None manuf=None nwk=0x611B ieee=84:2e:14:ff:fe:36:66:e4 is_initialized=False>
2022-01-16 16:37:20,908 DEBUG   : [       ZigpyCom_2] handle_message device 1: <Device model=None manuf=None nwk=0x611B ieee=84:2e:14:ff:fe:36:66:e4 is_initialized=False> Profile: 0104 Cluster: 0702 sEP: 1 dEp: 1 message: 08550a000025b20800000000 lqi: 174
2022-01-16 16:37:20,914 DEBUG   : [       ZigpyCom_2] handle_message device 2: 611b Profile: 0104 Cluster: 0702 sEP: 1 dEp: 1 message: 08550a000025b20800000000 lqi: 174
2022-01-16 16:37:20,918 DEBUG   : [       ZigpyCom_2] build_plugin_8002_frame_content 611b 260 1794 1 1 b'\x08\x55\x0A\x00\x00\x25\xB2\x08\x00\x00\x00\x00' 174 0 2 2
2022-01-16 16:37:20,920 DEBUG   : [       ZigpyCom_2] ==> build_plugin_8002_frame_content - SourceAddr: 611b message: b'\x08\x55\x0A\x00\x00\x25\xB2\x08\x00\x00\x00\x00'
2022-01-16 16:37:20,921 DEBUG   : [       ZigpyCom_2] handle_message Sender: 611b frame for plugin: 0180020032ff0001040702010102611b02000008550a000025b20800000000ae03
2022-01-16 16:37:20,923 DEBUG   : [ ZigpyForwarder_2] ZigateRead - MsgType: 8002,  Data: 0001040702010102611b02000008550a000025b20800000000, LQI: 174
2022-01-16 16:37:20,924 DEBUG   : [ ZigpyForwarder_2] decode8002_and_process ProfileId: 0104 611b 0180020032ff0001040702010102611b02000008550a000025b20800000000ae03
2022-01-16 16:37:20,925 DEBUG   : [ ZigpyForwarder_2] decode8002_and_process ProfileID: 0104 NwkId: 611b Ep: 01 Cluster: 0702 Payload: 08550a000025b20800000000
2022-01-16 16:37:20,926 DEBUG   : [ ZigpyForwarder_2] zcl_decoders GlobalCommand: True Sqn: 55 ManufCode: None Command: 0a Data: 000025b20800000000 Payload: 08550a000025b20800000000
2022-01-16 16:37:20,927 DEBUG   : [ ZigpyForwarder_2] zcl_decoders Sqn: 85/55 ManufCode: None Command: 0a Data: 000025b20800000000 
2022-01-16 16:37:20,928 DEBUG   : [ ZigpyForwarder_2] buildframe_report_attribute_response - 611b 01 0702 Data: 000025b20800000000
2022-01-16 16:37:20,929 DEBUG   : [ ZigpyForwarder_2] decode8002_and_process return ZCL frame: 0181020024ff55611b010702000000250006000008b20000ae03
2022-01-16 16:37:20,929 DEBUG   : [ ZigpyForwarder_2] Decode8102 - Attribute Reports: [611b:01] MsgSQN: 55 ClusterID: 0702 AttributeID: 0000 Status: 00 Type: 25 Size: 0006 ClusterData: >000008b20000<
2022-01-16 16:37:20,930 DEBUG   : [ ZigpyForwarder_2] scan_attribute_reponse - 8102 idx: 36 Read Attribute Response: [611b:01] ClusterID: 0702 MsgSQN: 55, i_sqn: 55, AttributeID: 0000 Status: 00 Type: 25 Size: 0006 ClusterData: >000008b20000<
2022-01-16 16:37:20,931 DEBUG   : [ ZigpyForwarder_2] Decode8102 - LQI: 174 Received Cluster:0702 Attribute: 0000 Value: 000008b20000 from (611b/01)
2022-01-16 16:37:20,932 DEBUG   : [ ZigpyForwarder_2] ReadCluster - 0702 - 611b/01 AttrId: 0000 AttrType: 25 Attsize: 0006 Status: 00 AttrValue: 000008b20000
2022-01-16 16:37:20,933 DEBUG   : [ ZigpyForwarder_2] decodeAttribute( 25, 000008b20000) 
2022-01-16 16:37:20,934 DEBUG   : [ ZigpyForwarder_2] Cluster0702 - MsgAttrID: 0000 MsgAttType: 25 DataLen: 0006 Data: 000008b20000 decodedValue: 145883136
2022-01-16 16:37:20,935 DEBUG   : [ ZigpyForwarder_2] Cluster0702 - 0x0000 CURRENT_SUMMATION_DELIVERED Value: 000008b20000 Conso: 1458831360 
2022-01-16 16:37:21,041 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1794, SrcAddr=0x611B, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=174, SecurityUse=<Bool.false: 0>, TimeStamp=12905673, TSN=0, Data=b'\x08\x55\x0A\x00\x00\x25\xB2\x08\x00\x00\x00\x00', MacSrcAddr=0x611B, MsgResultRadius=29)
2022-01-16 16:37:21,046 DEBUG   : [       ZigpyCom_2] AppZnp - get_device ieee:None nwk:0x611B 
2022-01-16 16:37:21,050 DEBUG   : [       ZigpyCom_2] AppZnp - get_device found device: <Device model=None manuf=None nwk=0x611B ieee=84:2e:14:ff:fe:36:66:e4 is_initialized=False>
2022-01-16 16:37:21,052 DEBUG   : [       ZigpyCom_2] handle_message device 1: <Device model=None manuf=None nwk=0x611B ieee=84:2e:14:ff:fe:36:66:e4 is_initialized=False> Profile: 0104 Cluster: 0702 sEP: 1 dEp: 1 message: 08550a000025b20800000000 lqi: 174
2022-01-16 16:37:21,054 DEBUG   : [       ZigpyCom_2] handle_message device 2: 611b Profile: 0104 Cluster: 0702 sEP: 1 dEp: 1 message: 08550a000025b20800000000 lqi: 174
[... cut ...]
2022-01-16 16:38:42,711 DEBUG   : [ ZigpyForwarder_2] ReadCluster - 0000 - 611b/01 AttrId: ffdf AttrType: 42 Attsize: 0014 Status: 00 AttrValue: 0000000012df6975291200000000050000000005
2022-01-16 16:38:42,712 DEBUG   : [ ZigpyForwarder_2] ReadCluster - 0000 611b/01 attribute Tuya/Zemismat - ffdf: 0x0000000012df6975291200000000050000000005
2022-01-16 16:38:42,713 DEBUG   : [ ZigpyForwarder_2] decodeAttribute( 42, 0000000012df6975291200000000050000000005) 
2022-01-16 16:38:42,713 DEBUG   : [ ZigpyForwarder_2] decodeAttribute - seems errors, returning with errors ignore From: 0000000012df6975291200000000050000000005 to >iu)<

pipiche38, 2022-01-16:
Fixed.
Problem is related to the fact that the device is considered as a router in a neighbourg table.

closed 2022-01-16 by pipiche38