zigbeefordomoticz/Domoticz-Zigbee #1024: Get Zigbee data out of domoticz

usky73, 2022-01-19:
Hi all
First of all just conect my zigate to domoticz, and I want to thanks all the people that works on this plugin and doc. It was easy to install and the interface is really professionnal.

I have only one thing I would like to adress and I did not find how to do.
I want to maage my energy data outside of domoticz. I have made my own system based on a Mysql database.
Is there a way to record my Zigbee sensor to this database using this plugin ?

Thanks 

Laurent

pipiche38, 2022-01-19:
This is more a Domoticz question than a Zigbee for Domoticz plugin one. I suggest that you use the https://www.domoticz.com/forum/ for that

pipiche38, 2022-01-22:
Here is an example of pushing data to outside
https://github.com/tvwerkhoven/domoticz2influxdb/blob/master/smartmeterpusher.lua
Mainly the idea is to use a script (lua, dzVents would be better) to push the incoming data to whatever you want and in your case to the mysql DB.
I'm sure that you can find support on Domoticz forum as well.

Closing the Issue

closed 2022-01-22 by pipiche38

usky73, 2022-01-27:
thx for your help, I will have a look on this.