zigbeefordomoticz/Domoticz-Zigbee #1026: [zigpy-integration] - how to use the zigpy OTA services instead of the plugin OTA

pipiche38, 2022-01-22:
The idea is to investigate if there is a way to make the plugin using directly the zigpy OTA service instead of going through the plugin one

Hedda, 2022-01-25:
FYI, for back-story and reference, you might be interested in reading the previous related discussion in https://github.com/zigpy/zigpy/pull/849

pipiche38, 2022-08-22:
Might start from here
https://github.com/zigpy/zigpy-cli/tree/dev/zigpy_cli

pipiche38, 2022-08-22:
I think this is not really possible, as for now the zigpy integration is at the low level and if we want to have OTA, we will need to integrate ZCL level in order to manage the OTA cluster, which is not the current strategy.

Closing for now

closed 2022-08-22 by pipiche38