zigbeefordomoticz/Domoticz-Zigbee #1028: [zigpy-integration] - Zigate is responding to IAS Zone Request

pipiche38, 2022-01-23:
Very suspicious , during a Zone Enrollement, it looks like the Zigate is responding to the device with an Zone Enroll Response ( Unknown response code) and a not corresponding SQN.

Basically the issue is he following

paquet: 435 Enroll Request with sqn 1
paquet: 454 Enroll Response with sqn 100
paquet: 480 Enroll Response with sqn 1 (provided by the plugin)

In the lof file, we can see 0x8012 and 0x8011, but no 0x8000 !
```
2022-01-23 16:56:01,608 DEBUG   :data_received : data confirm received 0x8012 sqn:227
2022-01-23 16:56:01,655 DEBUG   :data_received : ack received 0x8011 sqn:227
```

Question why is the Zigate doing that ?

[image]
[PIR323-zigate-zigpy-pairing-IAS.pcapng.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/7920845/PIR323-zigate-zigpy-pairing-IAS.pcapng.zip)
[pp6.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/7920849/pp6.zip)

pipiche38, 2022-01-23:
@fairecasoimeme FYI ..

pipiche38, 2022-01-23:
I did a full paring on a Native Zigate and the behaviour looks the same when the device do the Enroll Zone request
Zigate respond with:
Zone Enroll Response - Unknown
Zone Enroll Response - Success

Why is Zigate doing that ? And of course why is the 2nd response on a different Sequence Number

[image]

[PIR323-zigate-zigpy-pairing-IAS-natif.pcapng.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/7920866/PIR323-zigate-zigpy-pairing-IAS-natif.pcapng.zip)
r

pipiche38, 2022-01-24:
@badz should be  inside , as there is the full association

badzz, 2022-01-24:
I suspect this FALSE should be a TRUE
https://github.com/fairecasoimeme/ZiGate/blob/e54cabc22a49f33121e6e9c7cbecf779203b8ec7/ModuleRadio/Firmware/src/sdk/JN-SW-4170/Components/ZCL/Clusters/SecurityAndSafety/Source/IASZoneCommands.c#L340
not sure why we go there in raw mode however .. checking

pipiche38, 2022-02-03:
not a priority

closed 2022-02-03 by pipiche38