zigbeefordomoticz/Domoticz-Zigbee #1030: [Zigpy-integration] - Leave during pairing process

pipiche38, 2022-01-26:
This is happening every time with the PIR323 Multi-sensor.
The Pairing process start and then during the process the device send a Leave.

From the sniffer, I have observed that there is a Request Key happening after the process has started and the leave is happening just after the Verify Key

Attached is 2 Sniffs, and the plugin log of the latest sniff
[Archive.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/7944161/Archive.zip)
[Pairing-Leave.txt.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/7944167/Pairing-Leave.txt.zip)

pipiche38, 2022-01-26:
https://github.com/zigpy/zigpy/discussions/865#discussioncomment-2053334

pipiche38, 2022-02-03:
@puddly for now I gave up on that issue, because I didn't see any pattern. Basically only few time I have been to go through a stable pairing, meanings that the discovery was done, the IAS enrollment completed and the sensor behaving correctly. But in most of the time, it was just failing with a leave message.

So right now, I don't really now how to tackle it. If you have any idea, this would be very welcome

pipiche38, 2022-02-05:
Here are 2 captures on a zzh key.

One with a full successfull pairing/discovery
One with a leave during the pairing/discovery

[Wireshark capture.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/8008557/Wireshark.capture.zip)

puddly, 2022-02-05:
Do you mean that it doesn't leave 100% of the time when using the ZZH?

pipiche38, 2022-02-05:
Yes, exactly.

So Zigate native 100% of cases correctly paired and discovered fully functional
Zigate with zigpy 100% of cases correctly paired and discovered fully functional
ZNP (zzh or SonOff) most of the time leave during the pairing process, I would Once over 10 times success

pipiche38, 2022-04-04:
for now PIR323 is not supported on zigpy-znp

closed 2022-04-04 by pipiche38