zigbeefordomoticz/Domoticz-Zigbee #1045: [Zigpy-Integration] - Set TX Power

pipiche38, 2022-02-04:
On Zigate V1, we have the possibility to lower the TX power. On the CC2652R and CC2652P data sheet I see that there are several Power level.

Do we have on zigpy-znp API to control such power ? If so we should them implementing using the existing plugin setup.

CC: @puddly

puddly, 2022-02-07:
https://github.com/zigpy/zigpy-znp/blob/f069b4580e9b39eb0c2bda09a04ae9c236447a31/zigpy_znp/zigbee/application.py#L320-L323

Can also be passed in the config object to `zigpy_znp.zigbee.application.ControllerApplication`: https://github.com/zigpy/zigpy-znp#configuration

pipiche38, 2022-02-10:
e67389f7

pipiche38, 2022-02-10:
Zigate was only able to reduce the Tx Power, not increase.
Let's work on the Web UI to allow also the possibility to increase the power

puddly, 2022-02-10:
Anything greater than +22 will be limited by firmware to +22, but the same doesn't seem to apply when you go below the lower limit.  The incorrect parsing of the error code is a bug, the fix will be merged shortly.

pipiche38, 2022-02-10:
fixed with PR #1048 (integrate also fix on zigpy-znp https://github.com/zigpy/zigpy-znp/pull/136)

Hedda, 2022-02-11:
> Anything greater than +22 will be limited by firmware to +22, but the same doesn't seem to apply when you go below the lower limit. The incorrect parsing of the error code is a bug, the fix will be merged shortly.

So firmware developers config a default in Texas Instruments Z-Stack and then TX power can either be increased or lowered via ZNP and long as the maximum or minimum is exceeded as then the firmware will override and set maximum or minimum respectively?

> It's different between different firmware versions. The response is checked by ZNP so if it doesn't throw an error, it worked.

FYI, here is a reference of Texas Instruments chips hardware capabilities with maximum limits (unknown minimum) for reference:

- CC1352P ("P" for embedded Power-Amplifier) has a programmable output power up to +20dBm (and down to -?dBm).
- CC1352R has a maximum TX power configuration of +5dBm (and down to -?dBm unknown)
- CC2652P ("P" for embedded Power-Amplifier) has a programmable output power up to +20dBm (and down to -?dBm)
- CC2652R has a programmable output power up to +5dBm  (and down to -?dBm unknown)
- CC2652RB has a programmable output power up to +5dBm (and down to -?dBm unknown)
- CC2538 without a power amplifier chip has a programmable output power up to +7dBm (and down to -?dBm)
- CC2538 + CC2591 or CC2592 power amplifier chip has a programmable output power up to +19dBm (and down to -?dBm)
- CC2531 without has a programmable output power up to +4.5dBm (and down to -?dBm unknown)
- CC2531 with a CC2591 or CC2592 power amplifier chip has a programmable output power up to +19dBm (and down to -?dBm)
- CC2530 without has a programmable output power up to +4.5dBm (and down to -?dBm)
- CC2530 with a CC2591 or CC2592 power amplifier chip has a programmable output power up to +19dBm (and down to -?dBm)

I think bellows should be able to do the same for Silicon Labs EZSP radios but look to be hardcoded now? -> https://github.com/zigpy/bellows/issues/446

Silabs EFR32MG21 (EFR32 Mighty Gecko Series 2) and EFR32MG13 (EFR32 Mighty Gecko Series 1) chips and modules reference:

- EFR32MG24A420xxxxxxxx has a programmable output power up to +19.5dBm and down to -19.5dBm ?
- EFR32MG24A410xxxxxxxx has a programmable output power up to +10dBm and down to -19.5dBm ?
- MGM210P ("P" for embedded Power-Amplifier) has a programmable output power up to +20dBm and down to -30dBm
- MGM210L has a programmable output power up to +12.5dBm and down to -30dBm
- EFR32MG21A020xxxxxxxxx has a programmable output power up to +20dBm and down to -20dBm
- EFR32MG21A010xxxxxxxxx has a programmable output power up to +10dBm and down to -20dBm
- EFR32MG21B020xxxxxxxxx has a programmable output power up to +20dBm and down to -20dBm
- EFR32MG21B010xxxxxxxxx has a programmable output power up to +10.5dBm and down to -20dBm
- MGM13P ("P" for embedded Power-Amplifier) has a programmable output power up to +19dBm and down to -30dBm
- MGM13S has a programmable output power up to +10dBm and down to -30dBm
- EFR32MG13P732xxxxxxxx has a programmable output power up to +10dBm and down to -30dBm
- EFR32MG13P632xxxxxxxx has a programmable output power up to +10dBm and down to -30dBm

pipiche38, 2022-02-11:
@Hedda thanks for such detailed description. 
@jp-keros might be good if there is not such comprehensive table like the informations shared by @Hedda to put it in our wiki in English and French ( so eventually HA could also refer to that page)

jp-keros, 2022-02-12:
It's added to the [ToDo List](https://github.com/zigbeefordomoticz/wiki/issues/183).

closed 2022-02-12 by pipiche38

Hedda, 2022-02-14:
On a side note, I understand that in most countries illegal to use higher than +10dBM on some Zigbee channels like Zigbee channel 25 and Zigbee channel 26 because they are on the edge of the allowed frequency range of the worldwide 2.4 GHz ISM radio band, while you are legally allowed to use higher power in the middle of the 2.4 GHz ISM radio band in most countries. Some countries have however made it simple and set one power limit for the whole 2.4 GHz ISM radio band in their country. 

As such I understand that most manufacturers of Zigbee Coordinator adapter that allow Zigbee channels 25 and 26 (i.e. all modern Zigbee Coordinators adapters based old relatively new chips) will still only set the default in firmware to +9dBm or +10dBM (even if the chip hardware support higher power) to allow it to pass FCC and CE certifications for international worldwide sale. Some manufacturers like Digi XBee play it even safer and hardcode SDK so can not exceed +10 dBm in ERP (Effective Radiated Power). Others put the responsibility on the Zigbee implementation if they want to allow higher power configuration on all or just some Zigbee channels/frequencies.

pipiche38, 2022-02-14:
@hedda that is good point, however from my side, I do not see any benefit to increase the power Tx, as at the end you will be able to reach the devices, but they won't be able reach you. So Power Tx as an interest from a decrease standpoint

Hedda, 2022-02-14:
> I do not see any benefit to increase the power Tx, as at the end you will be able to reach the devices, but they won't be able reach you. So Power Tx as an interest from a decrease standpoint

To loosly quote puddly; increasing TX-power only makes the Zigbee coordinator is only "shouting louder", not "listening better". 

While that is true you also have to remember that a CC2652P/CC1352P with an external antenna can also be a very good listener ;-)

So let us agree to disagree on this point, as while I agree that it might not always be a good idea to set +20dBm in TX-power if got CC2652P or CC1352P as a Zigbee Coordinator and many Zigbee end-devices without many Zigbee routers, it can still certainly be a good idea to set it to +10dBm in TX-power to find a balance if got a healthy Zigbee network mesh in an environment that is not noisy with a lot of interference on the same frequencies.