zigbeefordomoticz/Domoticz-Zigbee #1047: Backup and restore Zigbee network to/from file using the "Open ZigBee Coordinator Backup Format"

Hedda, 2022-02-09:
**Is your feature request related to a problem? Please describe.**

Suggest adding an option to backup and restore users Zigbee network to/from file using the "Open ZigBee Coordinator Backup Format":

https://github.com/zigpy/open-coordinator-backup

The main benefit of this new "Open ZigBee Coordinator Backup Format" is that it is designed to be an open standard that allows backup and restoration of users Zigbee network independent of which Zigbee Coordinators hardware manufacturer and Zigbee stack used.

Having this option implemented would that users could backup to file for example a Zigbee network running on a Texas Instruments based Zigbee Coordinator adapter and restore such file backups to a Silicon Labs based Zigbee Coordinator adapter, or vice versa.

**Describe the solution you'd like**

The "Open ZigBee Coordinator Backup Format" is already implemented into zigpy-znp and bellows for zigpy while its developers is working on implementing this into zigpy-cli using a new radio API for zigpy:

https://github.com/zigpy/zigpy-cli/pull/2

If using a non-ZiGate Zigbee Coordinator via Zigpy integration then will soon be able to do backup and restore to the "Open ZigBee Coordinator Backup Format" via zigpy-cli using the new zigpy radio API.

Don't know if backup and restore of ZiGate could be implemented into zigpy-zigate or if would have make your own implementation.

**Describe alternatives you've considered**

An alternative could be to come up with your own Zigbee network backup format instead of using this as a shared standard format.

**Additional context**

Today this "Open ZigBee Coordinator Backup Format" is not only used by Home Assistant's ZHA and the official Jeedom Zigbee Plugin via zigpy (zigpy-znp and bellows only so far),  and support for this format has as well also been implemented (without zigpy) by zigbee-herdsman which is used by Zigbee2MQTT (Z2M) and IoBroker. 

Supporting this open standard format could as such potentially also enable cross-implementation migrations, meaning that a user could do a backup of their Zigbee network in  Home Assistant's ZHA or Zigbee2MQTT and then restore that to the Zigbee Plugin for Domoticz, or vice versa, as an open standard format for Zigbee network backup shared among the most popular open-source home automation applications.

pipiche38, 2022-02-09:
This will be available is there is a cli command.
I do not plan to get integrated inside the plugin , to be launched via the Domoticz Web Interface

Hedda, 2022-02-09:
OK, but might want to consider adding scheduled automatic daily/nightly backups of Zigbee network to file enabled by default.

Many arguments for automatic daily/nightly backup of Zigbee network by default -> https://community.home-assistant.io/t/zha-integration-to-do-nightly-backup-of-both-zigbee-coordinator-adapter-dongle-stick-and-zigbee-database/357558/

pipiche38, 2022-02-09:
I do consider that a proper backup required a device stopped, otherwise you'll get the risk to backup with inconsistency.
If the user wants to make a scheduled task, fine, but I think this is the user decision

Hedda, 2022-02-10:
> I do consider that a proper backup required a device stopped, otherwise you'll get the risk to backup with inconsistency.

FYI, running Home Assistant's ZHA integration it is possible to run backup online using "zha-toolkit" service without stopping ZHA:

https://github.com/mdeweerd/zha-toolkit/

https://github.com/mdeweerd/zha-toolkit/#znp_backup-backup-znp-network-data

https://github.com/mdeweerd/zha-toolkit/#znp_nvram_backup-backup-znp-nvram-data

https://github.com/mdeweerd/zha-toolkit/#ezsp_backup-backup-ezspbellows-network-data

pipiche38, 2022-02-10:
please let us doing what we want . I have understood then possibility and the functionallities, now we do implement what we want and what we can.

Thanks

zigbeefordomoticz, 2022-03-09:
not in the priority list for now.

closed 2022-03-09 by zigbeefordomoticz