zigbeefordomoticz/Domoticz-Zigbee #1051: [zigpy-integration] - zigate version_int() missing MajorVersion

pipiche38, 2022-02-16:
When using version_int() or even version_str() we do not get the FirmwareMajorVersion which here after is the 0x0004
```
2022-02-16 11:44:09,326 DEBUG   :data_received : status received 0x8000 status:0x00 cmd:0x0010 sqn:None
2022-02-16 11:44:09,327 DEBUG   :zigate_callback_handler [0, 0, 16, 0, 0, b'\x00\x00']
2022-02-16 11:44:09,329 DEBUG   :Frame received: 80100005b30004032100
2022-02-16 11:44:09,330 DEBUG   :data received 0x8010 b'00040321' LQI:0
2022-02-16 11:44:09,331 DEBUG   :data_received : status received 0x8010
2022-02-16 11:44:09,332 DEBUG   :zigate_callback_handler [4, 801]
2022-02-16 11:44:09,334 DEBUG   :command : Got status for 0x0010 : sqn:None
2022-02-16 11:44:09,335 DEBUG   :command : Wait for response 0x8010
2022-02-16 11:44:09,337 DEBUG   :command : Got response 0x8010 : ([4, 801], 0)
2022-02-16 11:44:09,339 DEBUG   :command : end command cmd:0x0010 result:([4, 801], 0)

2022-02-16 11:44:09,673 DEBUG   : [      ZigpyCom_19] App - version_str:3.21
2022-02-16 11:44:09,684 DEBUG   : [      ZigpyCom_19] App - version_int:801
```

pipiche38, 2022-02-16:
6d60b1ea

zigbeefordomoticz, 2022-02-19:
fixed

closed 2022-02-19 by zigbeefordomoticz