zigbeefordomoticz/Domoticz-Zigbee #1079: [zigpy-integration] - bellows/cli/ncp fails

pipiche38, 2022-03-10:
```
 python3 -m bellows.cli.ncp.py
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/bellows/cli/__init__.py", line 2, in <module>
    from . import application, backup, dump, ncp, network
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/bellows/cli/dump.py", line 6, in <module>
    import pure_pcapy
  File "/usr/local/lib/python3.9/dist-packages/pure_pcapy/__init__.py", line 88
    except Exception, error:
                    ^
SyntaxError: invalid syntax
```
CC: @puddly

pipiche38, 2022-03-10:
this looks to be valid for all cli/program like networks.py ....

puddly, 2022-03-10:
This was fixed in *2010*: https://github.com/rcloran/pure-pcapy-3/commit/04d434f2234c36bdfb11778055292240df5d3446

Are you manually installing dependencies? There's no need to do this. Any of the following work correctly and properly resolve all dependencies:

 - `pip install bellows`
 - `pip install git+https://github.com/puddly/bellows.git@puddly/zigpy-radio-api`
 - `git clone your-repo && pip install -e ./your-repo`

pipiche38, 2022-03-10:
We are not installing bellows with pip.

```
 sudo pip3 install pure-pcapy
 /usr/lib/python3/dist-packages (from pure-pcapy) (52.0.0)
```

puddly, 2022-03-10:
Why not? Bellows has specific dependencies, you can't just `pip install` the module name and expect it to work.

pipiche38, 2022-03-14:
You're right, just confirming your comment that pip install the dependency is not enought!

```
domoticz@pi4:~/domoticz/plugins/Domoticz-Zigbee $ sudo python3 -m bellows.cli.main -d /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 dump
/usr/lib/python3.9/runpy.py:127: RuntimeWarning: 'bellows.cli.main' found in sys.modules after import of package 'bellows.cli', but prior to execution of 'bellows.cli.main'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
Usage: python -m bellows.cli.main [OPTIONS] COMMAND [ARGS]...
Try 'python -m bellows.cli.main --help' for help.

Error: No such command 'dump'.
```