zigbeefordomoticz/Domoticz-Zigbee #1081: [zigpy-integration] Request ConBee and RaspBee (deCONZ based Zigbee Coordinators) support via zigpy-deconz

Hedda, 2022-03-10:
Suggest your project next consider implementing ConBee and RaspBee (deCONZ based Zigbee Coordinators) support via zigpy-deconz:

https://github.com/zigpy/zigpy-deconz

Also posted a sponsoring suggestion to a dresden-elektronik deCONZ Serial Protocol and ConBee and RaspBee firmware developer:

https://github.com/dresden-elektronik/deconz-serial-protocol/issues/17

@manup (Manuel Pietschmann) from [dresden elektronik](https://github.com/dresden-elektronik) / [Phoscon](https://phoscon.de/en/conbee2) is maybe be willing to sponsor with ConBee and RaspBee hardware?

https://github.com/dresden-elektronik/deconz-serial-protocol

https://phoscon.de

Btw, nice to read about new initial beta support for Silicon Labs EZSP Zigbee Coordinators via zigpy in addition to Texas Instrument ZNP!

https://github.com/zigbeefordomoticz/Domoticz-Zigbee/tree/beta6#tested-hardware-zigbee-adaptersdonglesstickskeys

https://github.com/zigbeefordomoticz/Domoticz-Zigbee/commits/beta6

pipiche38, 2022-03-11:
@puddly do you plan to have the new radio api implemented as well for deConz ?

pipiche38, 2022-03-11:
For information, I had already a Conbee II which was kindly shipped 2 years back when doing the first zigpy investigation.

I would say, thanks to zigpy library as I did the first integration is almost 2 hours of work and was able to get ConBee II connected.

Unfortunately for now, the Permit to Join mode doesn't happen. need to investigate

@manup (Manuel Pietschmann) We will investigate in our side, and I'll come back to you via email. 

In between what is the way to get the ConBee II upgraded ? (I'm running right now 26580700 Firmware ) ? I do not have Windows system. hope that is possible

puddly, 2022-03-11:
> do you plan to have the new radio api implemented as well for deConz ?

It's already done: https://github.com/zigpy/zigpy-deconz/pull/172

> what is the way to get the ConBee II upgraded ?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually
https://github.com/dresden-elektronik/gcfflasher

pipiche38, 2022-03-13:
Looks that we need to wait for https://github.com/zigpy/zigpy-deconz/pull/172
And I'm also looking to get respons from @manup as there is an existing Domoticz plugin for deCONZ ( https://github.com/Smanar/Domoticz-deCONZ  )

Hedda, 2022-03-14:
> > what is the way to get the ConBee II upgraded ?
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually 
> 
> https://github.com/dresden-elektronik/gcfflasher

GCFFlasher update on Linux -> http://ronhks.hu/2021/04/22/conbee-2-firmware-upgrade-on-ubuntu/

GCFFlasher update in Docker container -> https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

GCFFlasher update on Windows -> https://flemmingss.com/how-to-update-conbee-conbee-ii-firmware-in-windows-10/

Also see changelog -> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog

While not technically required it is highly recommended that users often upgrade deconz firmware on ConBee or RaspBee.

PS: Dresten Elektronik also (strongly) recommend using a long USB extension cable connected to a USB 2.0 port due to EMF/EMI:

https://phoscon.de/en/conbee2/install#connection

manup, 2022-03-14:
As far as I understand @Smanar 's  Domoticz plugin is communicating with deCONZ REST-API. In comparison to HA this is like the HA deCONZ integration https://www.home-assistant.io/integrations/deconz in contrast to ZHA which can can talk to ConBee firmware via serial protocol.

Hedda, 2022-03-15:
> As far as I understand @Smanar 's Domoticz plugin is communicating with deCONZ REST-API. In comparison to HA this is like the HA deCONZ integration https://www.home-assistant.io/integrations/deconz in contrast to ZHA which can can talk to ConBee firmware via serial protocol.

Yes correct, just the same as the [deCONZ integration Home Assistant](https://www.home-assistant.io/integrations/deconz) the [Smanar Domoticz-deCONZ](https://github.com/Smanar/Domoticz-deCONZ) communicate via [deCONZ REST-API](https://github.com/dresden-elektronik/deconz-rest-doc), (and not via [deCONZ Serial Protocol](https://github.com/dresden-elektronik/deconz-serial-protocol) like [zigpy-deconz](https://github.com/zigpy/zigpy-deconz)). As such Smanar's deCONZ Plugin for Domoticz is dependent on the [deCONZ application (closed source propriatory software by Dresden Elektronik)](https://phoscon.de/en/conbee2/software#deconz) to control and set up the Zigbee network.

In contrast, both the [ZHA integration for Home Assistant](https://www.home-assistant.io/integrations/zha) and the [Jeedom Zigbee Plugin](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/), as well both as [Zigbee2MQTT](https://github.com/Koenkk/zigbee2mqtt) and [IoBroker](https://github.com/ioBroker/ioBroker.zigbee), instead only use the [deCONZ Serial Protocol](https://github.com/dresden-elektronik/deconz-serial-protocol) via either [zigpy-deconz](https://github.com/zigpy/zigpy-deconz) or [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman) respectivly as their dependency, and as such they use direct serial-communication to the deCONZ firmware running on ConBee/RaspBee adaptera and are not in any way dependent on the [deCONZ application software](https://phoscon.de/en/conbee2/software#deconz) at all.

@manup Other than the [deCONZ application software](https://phoscon.de/en/conbee2/software#deconz) not being open source I personally think the main downside from ConBee/RaspBee adapters end-users point-of-view is that the two different methods do not support the same backup format and as such it is not easy for users to migrate between different implementation without having to manually repair all their Zigbee devices. I however beleieve that problem could however be solved by Dresden Elektronik could add support for the new "[Open ZigBee Coordinator Backup Format](https://github.com/zigpy/open-coordinator-backup)" to [deCONZ application software](https://phoscon.de/en/conbee2/software#deconz) and/or [deCONZ REST-API plugin](https://github.com/dresden-elektronik/deconz-rest-plugin), as this new "[Open ZigBee Coordinator Backup Format](https://github.com/zigpy/open-coordinator-backup)" is now supported by those other mentioned implemenations, including [ZHA integration for Home Assistant](https://www.home-assistant.io/integrations/zha), [Jeedom Zigbee Plugin](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/), [Zigbee2MQTT](https://github.com/Koenkk/zigbee2mqtt), and [IoBroker](https://github.com/ioBroker/ioBroker.zigbee),. Check out -> https://github.com/zigpy/open-coordinator-backup

zigbeefordomoticz, 2022-03-23:
as of now and based on the integration work done with the nea-radio-API , I see the following major show stopper to push the deConz integration to production

* #1104 Serial line seems not stable. they are several reset of the serial line along a day. The zigpy-deconz layer is doing the reconnect, but this look suspicious.
`2022-03-23 00:49:36,904 ERROR   :Lost serial connection: device reports readiness to read but returned no data (device disconnected or multiple access on port?)`

* #1089 not able to form a new network

* #1090 not able to add EndPoint 

We are awaiting the sample , so the other team members will be able to test and improve

cc: @Hedda , @puddly , @manup

pipiche38, 2022-03-29:
short update

- #1104 @deufo who is also testing the ConBee II now, is not having serial line stability issues. May be mine is related to the old  Conbee (from 2 years back). 
- #1089 Forming a new network is now closed

open issues are

- #1090 Not able to add Coordinator EndPoints (lack zigpy-deconz api)
- #1120 Not able to change Network Channel via the Network Update Request ( Cluster 0x0038)
- https://github.com/zigpy/zigpy/pull/848 Availability of new Radio API

cc: @Hedda , @puddly , @manup

pipiche38, 2022-05-09:
zigpy-deconz maintenance and development seems quiet limited as zigp-zigate.

For now, not sure that deconz will go live on stable

pipiche38, 2022-06-29:
release on the beta6 branch for now, for futur launch to stable. But deConz considered ready with minor limitation

- no channel change possible
- no support of Wiser/Livolo ... until new firmware release

closed 2022-06-29 by pipiche38