zigbeefordomoticz/Domoticz-Zigbee #1089: [zigpy-integration] - how to force a new network with deCONZ

zigbeefordomoticz, 2022-03-11:
how to force a new newtork with deCONZ

zigpy-znp, zigpy-bellows have a way to force a new network, looks like there is no way for deconz

zigbeefordomoticz, 2022-03-11:
https://github.com/zigpy/zigpy-deconz/issues/186

pipiche38, 2022-03-14:
looks like we never come back from and hang in it, as shown by the logs

`await super().form_network()`

```
2022-03-14 21:13:54,034 DEBUG   :startup form new network
2022-03-14 21:13:54,036 DEBUG   :Command Command.write_parameter (5, <NetworkParameter.nwk_frame_counter: 39>, b'\x00\x00\x00\x00')
2022-03-14 21:13:54,037 DEBUG   :Send: 0x0b42000c0005002700000000
2022-03-14 21:13:54,040 DEBUG   :Frame received: 0x0b42000800010027
2022-03-14 21:13:54,041 DEBUG   :Write parameter nwk_frame_counter: SUCCESS
2022-03-14 21:13:54,042 DEBUG   :Command Command.write_parameter (2, <NetworkParameter.aps_designed_coordinator: 9>, b'\x01')
2022-03-14 21:13:54,043 DEBUG   :Send: 0x0b4300090002000901
2022-03-14 21:13:54,047 DEBUG   :Frame received: 0x0b43000800010009
2022-03-14 21:13:54,048 DEBUG   :Write parameter aps_designed_coordinator: SUCCESS
2022-03-14 21:13:54,049 DEBUG   :Command Command.write_parameter (3, <NetworkParameter.nwk_address: 7>, b'\x00\x00')
2022-03-14 21:13:54,050 DEBUG   :Send: 0x0b44000a000300070000
2022-03-14 21:13:54,053 DEBUG   :Frame received: 0x0b44000800010007
2022-03-14 21:13:54,054 DEBUG   :Write parameter nwk_address: SUCCESS
```

pipiche38, 2022-03-14:
Stack trace when doing an new network form

```
2022-03-14 21:23:04,297 DEBUG   :Request id: 0x10 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x0000 endpoint=0>, status: 0x00
2022-03-14 21:23:04,299 INFO    :[0x0000:80] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=80, profile=56832, device_type=1, device_version=1, input_clusters=[], output_clusters=[])
2022-03-14 21:23:04,299 INFO    :[0x0000] Already have model and manufacturer info
2022-03-14 21:23:04,300 INFO    :[0x0000] Discovered basic device information for <DeconzDevice model='ConBee II' manuf='dresden elektronik' nwk=0x0000 ieee=00:21:2e:ff:ff:05:fe:7f is_initialized=True>
2022-03-14 21:23:04,301 DEBUG   :Device is initialized <DeconzDevice model='ConBee II' manuf='dresden elektronik' nwk=0x0000 ieee=00:21:2e:ff:ff:05:fe:7f is_initialized=True>
2022-03-14 21:23:04,301 DEBUG   :Checking quirks for dresden elektronik ConBee II (00:21:2e:ff:ff:05:fe:7f)
2022-03-14 21:23:04,371 DEBUG   :Frame received: 0x0e40000700a600
2022-03-14 21:23:04,372 DEBUG   :Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-03-14 21:23:04,373 DEBUG   :Command Command.aps_data_confirm (0,)
2022-03-14 21:23:04,373 DEBUG   :Send: 0x04400007000000
2022-03-14 21:23:04,375 DEBUG   :Frame received: 0x04400013000c00221202fcff00000000000000
2022-03-14 21:23:04,376 DEBUG   :APS data confirm response for request with id 18: 00
2022-03-14 21:23:04,376 DEBUG   :Request id: 0x12 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xfffc endpoint=0>, status: 0x00
2022-03-14 21:23:04,377 DEBUG   :Command Command.write_parameter (2, <NetworkParameter.permit_join: 33>, b'\x00')
2022-03-14 21:23:04,378 DEBUG   :Send: 0x0b4100090002002100
2022-03-14 21:23:04,380 DEBUG   :Frame received: 0x0b41000800010021
2022-03-14 21:23:04,381 DEBUG   :Write parameter permit_join: SUCCESS
2022-03-14 21:23:04,382 DEBUG   :startup form new network
2022-03-14 21:23:04,383 DEBUG   :Command Command.write_parameter (5, <NetworkParameter.nwk_frame_counter: 39>, b'\x00\x00\x00\x00')
2022-03-14 21:23:04,383 DEBUG   :Send: 0x0b42000c0005002700000000
2022-03-14 21:23:04,394 DEBUG   :Frame received: 0x0b42000800010027
2022-03-14 21:23:04,394 DEBUG   :Write parameter nwk_frame_counter: SUCCESS
2022-03-14 21:23:04,395 DEBUG   :Command Command.write_parameter (2, <NetworkParameter.aps_designed_coordinator: 9>, b'\x01')
2022-03-14 21:23:04,396 DEBUG   :Send: 0x0b4300090002000901
2022-03-14 21:23:04,399 DEBUG   :Frame received: 0x0b43000800010009
2022-03-14 21:23:04,400 DEBUG   :Write parameter aps_designed_coordinator: SUCCESS
2022-03-14 21:23:04,401 DEBUG   :Command Command.write_parameter (3, <NetworkParameter.nwk_address: 7>, b'\x00\x00')
2022-03-14 21:23:04,402 DEBUG   :Send: 0x0b44000a000300070000
2022-03-14 21:23:04,405 DEBUG   :Frame received: 0x0b44000800010007
2022-03-14 21:23:04,406 DEBUG   :Write parameter nwk_address: SUCCESS
2022-03-14 21:23:04,412 ERROR   :Error Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppDeconz.py", line 63, in startup
    await super().form_network()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 167, in form_network
    await self.write_network_info(network_info=network_info, node_info=node_info)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 161, in write_network_info
    await self._api.write_parameter(NetworkParameter.mac_address, node_info.ieee)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 460, in write_parameter
    v = t.serialize(args, NETWORK_PARAMETER_SCHEMA[param])
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/types.py", line 15, in serialize
    return b"".join(t(v).serialize() for t, v in zip(schema, data))
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/types.py", line 15, in <genexpr>
    return b"".join(t(v).serialize() for t, v in zip(schema, data))
TypeError: 'NoneType' object is not iterable
```

cc: @puddly

pipiche38, 2022-03-14:
In addition it looks like the force_form doesn't really work, as after the devices which were already paired with it still continue to reach the network and the coordfinator

pipiche38, 2022-03-15:
Looks like the form new network doesn't work. Ether due to the stack trace before, or something else!

puddly, 2022-03-15:
`force_form` is internal to zigpy-znp and will probably be removed in a future version, I wouldn't rely on it existing.

As for the traceback, zigpy is passing `node_info.ieee = None` and causing formation to fail halfway through the process. I will have to think about how to fix it. The idea was to have network formation re-use the device's hardware IEEE address but I don't think the Conbee stores this information, nor can I find a fast way to do this with ZNP at the moment.

puddly, 2022-03-15:
You just call `form_network`. This is a bug with zigpy and/or zigpy-deconz.

pipiche38, 2022-03-15:
so we are good, thank @puddly

puddly, 2022-03-15:
> For bellows, just before we leave the network.

Not necessary, new radio API does this in `write_network_settings`.

In the code that you've posted I think it will fail if you pass `force_form=True` but a device has no formed network, since `startup(auto_form=False)` will fail with `NetworkNotFormed`.

While it will take a few seconds longer in some cases, I think it would be safer to just do something like this:

```Python
await self.startup(auto_form=True)

if force_form:
    await self.form_network()
```

pipiche38, 2022-03-18:
let see with the other testers how it works

pipiche38, 2022-03-22:
@puddly puddly for your information, I'm not able to create a new network, It creates issue

```

Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppDeconz.py", line 57, in startup
    await super().startup(auto_form=True)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 61, in startup
    await self.connect()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 72, in connect
    self.version = await api.version()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 473, in version
    (self._proto_ver,) = await self[NetworkParameter.protocol_version]
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 438, in read_parameter
    r = await self._command(Command.read_parameter, 1 + len(data), param, data)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 323, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

puddly, 2022-03-22:
Your serial connection is not functional if the Conbee does not respond to the `version` request sent immediately after connecting.

[Following the zigpy-cli installation instructions](https://github.com/zigpy/zigpy-cli/pull/2) and connecting to the radio works for me:

```console
$ zigpy -vv radio deconz /dev/serial/by-id/... info
...
```

puddly, 2022-03-22:
Are you plugging in the Conbee and then immediately connecting? Or is it still plugged in when you fail to start up?

pipiche38, 2022-03-22:
Basically the way to reproduce is:

- plugin start - it fails
- I restart - it works
- I restart - it fails
- I restart - it works

I do not touch the ConBee II at all and it stays connected. Should I have to wait in between the stop and start ?

What looks very strange from my end is every time I shutdown the plugin I get that in dmesg

```
Mar 22 21:44:53 pi4 kernel: usb 1-1.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar 22 21:44:53 pi4 kernel: usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 22 21:44:53 pi4 kernel: usb 1-1.1: Product: ConBee II
Mar 22 21:44:53 pi4 kernel: usb 1-1.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar 22 21:44:53 pi4 kernel: usb 1-1.1: SerialNumber: DE2227699
Mar 22 21:44:53 pi4 kernel: cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
```

pipiche38, 2022-03-22:
```
Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppDeconz.py", line 59, in startup
    await super().startup(auto_form=True)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 61, in startup
    await self.connect()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 71, in connect
    await api.connect()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 263, in connect
    self._uart = await zigpy_deconz.uart.connect(self._config, self)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/uart.py", line 149, in connect
    _, protocol = await serial_asyncio.create_serial_connection(
  File "/usr/local/lib/python3.9/dist-packages/serial_asyncio/__init__.py", line 448, in create_serial_connection
    serial_instance = serial.serial_for_url(*args, **kwargs)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/serial/serialposix.py", line 332, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00'
```

pipiche38, 2022-03-23:
Ok I sorted out the issue at plugin startup by adding an extra sleep(2) before calling startup().

However still when trying to create a NEW network, I'm getting this error, which happen only when forming a new network!

```
2022-03-23 08:22:23,553 ERROR   :Error Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppDeconz.py", line 63, in startup
    await super().form_network()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 163, in form_network
    await self.write_network_info(network_info=network_info, node_info=node_info)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 205, in write_network_info
    await self._api.write_parameter(
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 460, in write_parameter
    v = t.serialize(args, NETWORK_PARAMETER_SCHEMA[param])
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/types.py", line 15, in serialize
    return b"".join(t(v).serialize() for t, v in zip(schema, data))
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/types.py", line 15, in <genexpr>
    return b"".join(t(v).serialize() for t, v in zip(schema, data))
TypeError: 'NoneType' object is not iterable
```

pipiche38, 2022-03-28:
Here after a full log, which is happening in a systematic way when trying to form a new network
cc: @puddly 

```
2022-03-28 12:59:28,004 INFO    : [ ZigpyForwarder_6] ZigpyTransport: thread_processing_and_sending Thread start.
2022-03-28 12:59:28,009 INFO    : [       MainThread] Start Web Server connection
2022-03-28 12:59:28,011 DEBUG   : [       ZigpyCom_6] In radio_start deCONZ
2022-03-28 12:59:28,011 INFO    : [       ZigpyCom_6] Form a New Network with Channel: 0(0x00) ExtendedPanId: 0xdcba484504015e10
2022-03-28 12:59:28,012 INFO    : [       MainThread] Web backend for Web User Interface started on port: 9440
2022-03-28 12:59:29,997 DEBUG   :Connection made
2022-03-28 12:59:29,998 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.protocol_version: 34>, b'')
2022-03-28 12:59:29,999 DEBUG   :Send: 0x0a02000800010022
2022-03-28 12:59:30,002 DEBUG   :Frame received: 0x0a02000a000300220e01
2022-03-28 12:59:30,003 DEBUG   :Read parameter protocol_version response: [270]
2022-03-28 12:59:30,003 DEBUG   :Command Command.version (0,)
2022-03-28 12:59:30,004 DEBUG   :Send: 0x0d0300090000000000
2022-03-28 12:59:30,007 DEBUG   :Frame received: 0x0d0300090000077226
2022-03-28 12:59:30,007 DEBUG   :Version response: 26720700
2022-03-28 12:59:30,009 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.mac_address: 1>, b'')
2022-03-28 12:59:30,009 DEBUG   :Send: 0x0a04000800010001
2022-03-28 12:59:30,012 DEBUG   :Frame received: 0x0a040010000900017ffe05ffff2e2100
2022-03-28 12:59:30,013 DEBUG   :Read parameter mac_address response: [00:21:2e:ff:ff:05:fe:7f]
2022-03-28 12:59:30,014 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.aps_designed_coordinator: 9>, b'')
2022-03-28 12:59:30,014 DEBUG   :Send: 0x0a05000800010009
2022-03-28 12:59:30,017 DEBUG   :Frame received: 0x0a0500090002000901
2022-03-28 12:59:30,018 DEBUG   :Read parameter aps_designed_coordinator response: [1]
2022-03-28 12:59:30,018 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.nwk_address: 7>, b'')
2022-03-28 12:59:30,019 DEBUG   :Send: 0x0a06000800010007
2022-03-28 12:59:30,027 DEBUG   :Frame received: 0x0a06000a000300070000
2022-03-28 12:59:30,028 DEBUG   :Read parameter nwk_address response: [0x0000]
2022-03-28 12:59:30,028 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.nwk_panid: 5>, b'')
2022-03-28 12:59:30,029 DEBUG   :Send: 0x0a07000800010005
2022-03-28 12:59:30,032 DEBUG   :Frame received: 0x0a07000a000300056f38
2022-03-28 12:59:30,033 DEBUG   :Read parameter nwk_panid response: [0x386f]
2022-03-28 12:59:30,033 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.aps_extended_panid: 11>, b'')
2022-03-28 12:59:30,034 DEBUG   :Send: 0x0a0800080001000b
2022-03-28 12:59:30,037 DEBUG   :Frame received: 0x0a0800100009000b105e01044548cdab
2022-03-28 12:59:30,038 DEBUG   :Read parameter aps_extended_panid response: [ab:cd:48:45:04:01:5e:10]
2022-03-28 12:59:30,039 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.current_channel: 28>, b'')
2022-03-28 12:59:30,039 DEBUG   :Send: 0x0a0900080001001c
2022-03-28 12:59:30,041 DEBUG   :Frame received: 0x0e09000700a200
2022-03-28 12:59:30,042 DEBUG   :Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 162>, 0]
2022-03-28 12:59:30,043 DEBUG   :Network state transition: OFFLINE -> CONNECTED
2022-03-28 12:59:30,043 DEBUG   :Frame received: 0x1c0a000c0005000251ceffce
2022-03-28 12:59:30,046 DEBUG   :Frame received: 0x0a0900090002001c0f
2022-03-28 12:59:30,047 DEBUG   :Read parameter current_channel response: [15]
2022-03-28 12:59:30,047 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.channel_mask: 10>, b'')
2022-03-28 12:59:30,048 DEBUG   :Send: 0x0a0a00080001000a
2022-03-28 12:59:30,051 DEBUG   :Frame received: 0x0a0a000c0005000a00800000
2022-03-28 12:59:30,052 DEBUG   :Read parameter channel_mask response: [<Channels.CHANNEL_15: 32768>]
2022-03-28 12:59:30,052 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.nwk_update_id: 36>, b'')
2022-03-28 12:59:30,053 DEBUG   :Send: 0x0a0b000800010024
2022-03-28 12:59:30,056 DEBUG   :Frame received: 0x0a0b00090002002400
2022-03-28 12:59:30,057 DEBUG   :Read parameter nwk_update_id response: [0]
2022-03-28 12:59:30,057 DEBUG   :Command Command.read_parameter (2, <NetworkParameter.network_key: 24>, b'\x00')
2022-03-28 12:59:30,058 DEBUG   :Send: 0x0a0c00090002001800
2022-03-28 12:59:30,061 DEBUG   :Frame received: 0x0a0c00190012001800cc5a335bfa5b37ef6556c99f7ce94b93
2022-03-28 12:59:30,062 DEBUG   :Read parameter network_key response: [0, [204, 90, 51, 91, 250, 91, 55, 239, 101, 86, 201, 159, 124, 233, 75, 147]]
2022-03-28 12:59:30,063 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.nwk_frame_counter: 39>, b'')
2022-03-28 12:59:30,063 DEBUG   :Send: 0x0a0d000800010027
2022-03-28 12:59:30,066 DEBUG   :Frame received: 0x0a0d000c000500270e040100
2022-03-28 12:59:30,067 DEBUG   :Read parameter nwk_frame_counter response: [66574]
2022-03-28 12:59:30,067 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.trust_center_address: 14>, b'')
2022-03-28 12:59:30,068 DEBUG   :Send: 0x0a0e00080001000e
2022-03-28 12:59:30,075 DEBUG   :Frame received: 0x0a0e00100009000e7ffe05ffff2e2100
2022-03-28 12:59:30,076 DEBUG   :Read parameter trust_center_address response: [00:21:2e:ff:ff:05:fe:7f]
[... cut ...]
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 460, in write_parameter
    v = t.serialize(args, NETWORK_PARAMETER_SCHEMA[param])
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/types.py", line 15, in serialize
    return b"".join(t(v).serialize() for t, v in zip(schema, data))
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/types.py", line 15, in <genexpr>
    return b"".join(t(v).serialize() for t, v in zip(schema, data))
TypeError: 'NoneType' object is not iterable
```

puddly, 2022-03-28:
Please use the latest versions of both zigpy and zigpy-deconz when testing, I believe this issue was fixed almost a week ago.

pipiche38, 2022-03-28:
OK . Indeed works better but now new error

`2022-03-28 19:06:09,967 INFO    : [       MainThread] ZigpyTransport __init__
2022-03-28 19:06:09,968 DEBUG   :Using selector: EpollSelector
2022-03-28 19:06:09,969 INFO    : [ ZigpyForwarder_6] ZigpyTransport: thread_processing_and_sending Thread start.
2022-03-28 19:06:09,970 INFO    : [       MainThread] Start Web Server connection
2022-03-28 19:06:09,972 DEBUG   :startup in AppDeconz
2022-03-28 19:06:09,992 INFO    : [       MainThread] Web backend for Web User Interface started on port: 9440
2022-03-28 19:06:11,984 DEBUG   :Connection made
2022-03-28 19:06:11,985 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.protocol_version: 34>, b'')
2022-03-28 19:06:11,985 DEBUG   :Send: 0x0a02000800010022
2022-03-28 19:06:11,987 DEBUG   :Frame received: 0x0a02000a000300220e01
2022-03-28 19:06:11,987 DEBUG   :Read parameter protocol_version response: [270]
2022-03-28 19:06:11,987 DEBUG   :Command Command.version (0,)
2022-03-28 19:06:11,988 DEBUG   :Send: 0x0d0300090000000000
2022-03-28 19:06:11,997 DEBUG   :Frame received: 0x0d0300090000077226
2022-03-28 19:06:11,997 DEBUG   :Version response: 26720700
2022-03-28 19:06:11,997 DEBUG   :Frame received: 0x0e040007008100
2022-03-28 19:06:11,998 DEBUG   :Device state changed response: [<DeviceState.128|1: 129>, 0]
2022-03-28 19:06:11,998 DEBUG   :Network state transition: OFFLINE -> JOINING
2022-03-28 19:06:11,998 DEBUG   :Data request queue full.
2022-03-28 19:06:11,999 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.mac_address: 1>, b'')
2022-03-28 19:06:11,999 DEBUG   :Send: 0x0a04000800010001
2022-03-28 19:06:12,001 DEBUG   :Frame received: 0x0a040010000900017ffe05ffff2e2100
2022-03-28 19:06:12,001 DEBUG   :Read parameter mac_address response: [00:21:2e:ff:ff:05:fe:7f]
2022-03-28 19:06:12,001 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.aps_designed_coordinator: 9>, b'')
2022-03-28 19:06:12,002 DEBUG   :Send: 0x0a05000800010009
2022-03-28 19:06:12,010 DEBUG   :Frame received: 0x0a0500090002000901
2022-03-28 19:06:12,010 DEBUG   :Read parameter aps_designed_coordinator response: [1]
2022-03-28 19:06:12,010 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.nwk_address: 7>, b'')
2022-03-28 19:06:12,011 DEBUG   :Send: 0x0a06000800010007
2022-03-28 19:06:12,013 DEBUG   :Frame received: 0x0a06000a000300070000
2022-03-28 19:06:12,013 DEBUG   :Read parameter nwk_address response: [0x0000]
2022-03-28 19:06:12,013 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.nwk_panid: 5>, b'')
2022-03-28 19:06:12,014 DEBUG   :Send: 0x0a07000800010005
2022-03-28 19:06:12,022 DEBUG   :Frame received: 0x0a07000a00030005c715
2022-03-28 19:06:12,022 DEBUG   :Read parameter nwk_panid response: [0x15c7]
2022-03-28 19:06:12,022 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.aps_extended_panid: 11>, b'')
2022-03-28 19:06:12,023 DEBUG   :Send: 0x0a0800080001000b
2022-03-28 19:06:12,025 DEBUG   :Frame received: 0x0a0800100009000b105e01044548badc
2022-03-28 19:06:12,025 DEBUG   :Read parameter aps_extended_panid response: [dc:ba:48:45:04:01:5e:10]
2022-03-28 19:06:12,026 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.current_channel: 28>, b'')
2022-03-28 19:06:12,026 DEBUG   :Send: 0x0a0900080001001c
2022-03-28 19:06:12,034 DEBUG   :Frame received: 0x0a0900090002001c00
2022-03-28 19:06:12,034 DEBUG   :Read parameter current_channel response: [0]
2022-03-28 19:06:12,034 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.channel_mask: 10>, b'')
2022-03-28 19:06:12,035 DEBUG   :Send: 0x0a0a00080001000a
2022-03-28 19:06:12,037 DEBUG   :Frame received: 0x0a0a000c0005000a00800000
2022-03-28 19:06:12,037 DEBUG   :Read parameter channel_mask response: [<Channels.CHANNEL_15: 32768>]
2022-03-28 19:06:12,037 DEBUG   :Command Command.read_parameter (1, <NetworkParameter.nwk_update_id: 36>, b'')
2022-03-28 19:06:12,038 DEBUG   :Send: 0x0a0b000800010024
2022-03-28 19:06:12,046 DEBUG   :Frame received: 0x0a0b00090002002400
2022-03-28 19:06:12,046 DEBUG   :Read parameter nwk_update_id response: [0]
2022-03-28 19:06:12,047 DEBUG   :Command Command.write_parameter (5, <NetworkParameter.nwk_frame_counter: 39>, b'\x00\x00\x00\x00')
2022-03-28 19:06:12,047 DEBUG   :Send: 0x0b0c000c0005002700000000
2022-03-28 19:06:12,049 DEBUG   :Frame received: 0x0b0c000800010027
2022-03-28 19:06:12,049 DEBUG   :Write parameter nwk_frame_counter: SUCCESS
2022-03-28 19:06:12,049 DEBUG   :Command Command.write_parameter (2, <NetworkParameter.aps_designed_coordinator: 9>, b'\x01')
2022-03-28 19:06:12,050 DEBUG   :Send: 0x0b0d00090002000901
2022-03-28 19:06:12,058 DEBUG   :Frame received: 0x0b0d000800010009
2022-03-28 19:06:12,058 DEBUG   :Write parameter aps_designed_coordinator: SUCCESS
2022-03-28 19:06:12,058 DEBUG   :Command Command.write_parameter (3, <NetworkParameter.nwk_address: 7>, b'\x00\x00')
2022-03-28 19:06:12,059 DEBUG   :Send: 0x0b0e000a000300070000
2022-03-28 19:06:12,061 DEBUG   :Frame received: 0x0b0e000800010007
2022-03-28 19:06:12,061 DEBUG   :Write parameter nwk_address: SUCCESS
2022-03-28 19:06:12,061 DEBUG   :Command Command.write_parameter (5, <NetworkParameter.channel_mask: 10>, b'\x00\x80\x00\x00')
2022-03-28 19:06:12,062 DEBUG   :Send: 0x0b0f000c0005000a00800000
2022-03-28 19:06:12,070 DEBUG   :Frame received: 0x0b0f00080001000a
2022-03-28 19:06:12,070 DEBUG   :Write parameter channel_mask: SUCCESS
[... cut ...]
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 128, in _change_network_state
    await self._api.change_network_state(target_state)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 323, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 481, in wait_for
    return fut.result()
zigpy_deconz.exception.CommandError: Command.change_network_state, status: Status.ERROR

2022-03-28 19:06:12,136 DEBUG   :Closed serial connection`

puddly, 2022-03-28:
I've never seen `Command.change_network_state` result in an `ERROR` status. Are you using a USB extension cable?

I just re-tested zigpy-deconz and `form_network` works fine for me but see if the most recent push changes things.

pipiche38, 2022-03-29:
@puddly Looks good with the latest. so closing for now

closed 2022-03-29 by pipiche38