zigbeefordomoticz/Domoticz-Zigbee #1090: [zigpy-integration] - wiser thermostat with deCONZ do not behave correctly

zigbeefordomoticz, 2022-03-11:
Wiser thermostat required EndPoint Coordinator at Ep 0x0b

How to register additional Cooridnator Ep with deconz ?

When Wiser Thermostat is paired we get some messages like that
```

2022-03-11 16:11:15,467 DEBUG   :Frame received: 0x1cdf000c00050002bfaaffca
2022-03-11 16:11:16,600 DEBUG   :Frame received: 0x1ce0000c00050002bfaaffcf
2022-03-11 16:11:17,606 DEBUG   :Frame received: 0x1ce1000c00050002bfaaffd6
2022-03-11 16:11:18,611 DEBUG   :Frame received: 0x1ce2000c00050002bfaaffd5
2022-03-11 16:11:19,616 DEBUG   :Frame received: 0x1ce3000c00050002bfaaffd5
2022-03-11 16:11:21,663 DEBUG   :Frame received: 0x1ce4000c00050002bfaaffd6
2022-03-11 16:11:22,668 DEBUG   :Frame received: 0x1ce5000c00050002bfaaffd6
2022-03-11 16:11:23,675 DEBUG   :Frame received: 0x1ce6000c00050002bfaaffd6
2022-03-11 16:11:24,679 DEBUG   :Frame received: 0x1ce7000c00050002bfaaffd6
```

pipiche38, 2022-03-13:
@puddy Is there a possibility to register Additional Ep with zigpy-deconz ?

Hedda, 2022-03-23:
For reference, which exact model of Schneider Electric Wiser thermostat is it? 

@manup Have you tested any Schneider Electric Wiser ( legacy) devices like Wiser thermostats with ConBee 2 / RaspBee 2?

https://phoscon.de/en/conbee2/compatible

https://phoscon.de/en/support#compatibility-of-devices

Only one TVR model is listed as confirmed working with deCONZ / Phoscon in Blakadder's Zigbee compatibility repository:

https://zigbee.blakadder.com/Schneider_Electric_CCTFR6100.html

While several other models are not yet listed as confirmed working:

https://zigbee.blakadder.com/Schneider_Electric_EER53000.html

https://zigbee.blakadder.com/Schneider_Electric_EER51000.html

https://zigbee.blakadder.com/Schneider_Electric_CCTFR6400.html

pipiche38, 2022-03-23:
@hedda, the all Wiser Legacy and new works with our plugin with Zigate, ZZH/SonOff and all CCxxxx as well as the Silicon Labs eelabs coordinator.
For that we need to be able to add Endpoint to the Coordinator. For now there is no such feature available on zigpy-deconz. The limitation is not deconz itself it is the zigpy layer except if deconz is not able to add EndPoint (which is not my understanding).
So we need to extend the zigpy_deconz in order to address that feature as well as the possibility to update the Manufacturer Id which is used to respond to Node Descriptor request.

SylvainPer, 2022-03-28:
Just manage to pair wiser devices on the conbee ii.
The end points are well managed without particular need like bellows.

[image]

One thing weird, I didn't succeed to pair through a router, only directly to the key.

SylvainPer, 2022-04-02:
After some clarification with @pipiche38, my test wasn't relevant. The Bind Request is an outgoing request and it's well managed by the coordinator. The problem is when the device asked from and to an endpoint different from 01, in my case ep 0x11 to 0x11.
I've made some test on the different libraries with a wiser thermostat, the Report Attributes and Read Attributes requests are not transferred to the application level. The coordinator answer an ACK and nothing more:
[image]

@puddly as for the znp library, the register endpoint function is needed for those particular devices.

closed 2022-04-29 by pipiche38