zigbeefordomoticz/Domoticz-Zigbee #1091: SASWELL Watering

pbda, 2022-03-11:
SASWELL watering device is not correctly recognized by Zigate*. It is seen as a radiator thermostat. 3 devices are created: temperature, set point and a selector (off, auto, manual). It can't be used in Domoticz because there is no on/off device

* Water timer Model: SAS980SWT-7-Z01
* SASWELL
* https://www.saswell.com/saswell-smart-irrigation-best-smart-sprinkler-water-timer-2021-eu_n97

Here is the device in export.json. :+1: 
```
  {
    "Version": "3",
    "ZDeviceName": "Watering",
    "Status": "inDB",
    "SQN": "39",
    "Heartbeat": "528",
    "RIA": "1",
    "LQI": 156,
    "Battery": {},
    "Model": "TS0601",
    "ForceAckCommands": [],
    "MacCapa": "80",
    "IEEE": "2c1165fffeb3c71f",
    "Type": "Temp/ThermoSetpoint/ThermoMode_2",
    "ProfileID": "0104",
    "ZDeviceID": "0051",
    "App Version": "85",
    "Attributes List": {},
    "DeviceType": "RFD",
    "HW Version": "1",
    "Last Cmds": [
      [
        341,
        "02",
        "6119",
        "0100",
        "0101000000000000010002"
      ],
      [
        342,
        "02",
        "6119",
        "0100",
        "0101000000000000010003"
      ],
      [
        343,
        "02",
        "6119",
        "0100",
        "0101000000000000010004"
      ],
      [
        344,
        "02",
        "6119",
        "0100",
        "0101000000000000010005"
      ],
      [
        345,
        "02",
        "6119",
        "0100",
        "0101000000000000010007"
      ],
      [
        346,
        "02",
        "6119",
        "0110",
        "010100000000000001ffde2013"
      ],
      [
        355,
        "02",
        "6119",
        "0092",
        "010100"
      ],
      [
        356,
        "02",
        "6119",
        "0092",
        "010100"
      ],
      [
        375,
        "02",
        "6119",
        "0092",
        "010100"
      ],
      [
        378,
        "02",
        "6119",
        "0092",
        "010100"
      ]
    ],
    "LogicalType": "End Device",
    "Manufacturer": "1002",
    "Manufacturer Name": "_TZE200_akjefhj5",
    "NbEp": "1",
    "PowerSource": "Battery",
    "ReadAttributes": {
      "Ep": {
        "01": {
          "0000": {
            "TimeStamp": "11/03/2022 15:49:03",
            "iSQN": {
              "0000": 339,
              "0001": 340,
              "0002": 341,
              "0003": 342,
              "0004": 343,
              "0005": 344,
              "0007": 345
            },
            "Attributes": {
              "fffe": "00",
              "0000": "00",
              "0001": "00",
              "0002": "00",
              "0003": "00",
              "0004": "00",
              "0005": "00",
              "0007": "00"
            },
            "ZigateRequest": {}
          },
          "0019": {
            "TimeStamp": "11/03/2022 15:41:11",
            "iSQN": {
              "0001": 265,
              "0002": 266,
              "0003": 267,
              "0004": 268,
              "0005": 269,
              "0006": 270,
              "0007": 271,
              "0008": 272,
              "0009": 273,
              "000a": 274
            },
            "Attributes": {
              "0001": "86",
              "0002": "86",
              "0003": "86",
              "0004": "86",
              "0005": "86",
              "0006": "86",
              "0007": "86",
              "0008": "86",
              "0009": "86",
              "000a": "86"
            },
            "ZigateRequest": {}
          }
        }
      }
    },
    "ReceiveOnIdle": "Off",
    "Stack Version": "0",
    "Stamp": {
      "time": 1647015887.2660437,
      "Time": "2022-03-11 17:24:47",
      "MsgType": "8011",
      "LastSeen": "11/03/2022 17:24:47"
    },
    "ZCL Version": "3",
    "Health": "Live",
    "Capability": [
      "Reduced-Function Device",
      "Standard security",
      "NwkAddr should be allocated"
    ],
    "RollingLQI": [
      171,
      216,
      216,
      216,
      216,
      162,
      162,
      132,
      132,
      156,
      156
    ],
    "Epv2": {
      "01": {
        "ClusterIn": {
          "0000": {},
          "0004": {},
          "0005": {},
          "ef00": {}
        },
        "ClusterOut": {
          "0019": {},
          "000a": {}
        },
        "ProfileID": "0104",
        "ZDeviceID": "0051"
      }
    },
    "ZDeviceVersion": "0001",
    "ConfigSource": "DeviceConf",
    "PairingInProgress": false,
    "Ep": {
      "01": {
        "0000": {
          "0004": "_TZE200_akjefhj5",
          "0005": "TS0601",
          "0000": "3",
          "0001": "85",
          "0002": "0",
          "0003": "1",
          "0007": "3",
          "fffe": "0"
        },
        "0004": {},
        "ef00": {},
        "000a": {},
        "0019": {},
        "Type": "Temp/ThermoSetpoint/ThermoMode_2",
        "ClusterType": {
          "615": "Temp",
          "616": "ThermoSetpoint",
          "617": "ThermoMode_2"
        }
      }
    },
    "Max Buffer Size": "52",
    "Max Rx": "0052",
    "Max Tx": "0052",
    "SQN_000a": "10",
    "Tuya": {
      "Valve 0x01": "00",
      "TUYA_MCU_VERSION_RSP": "40",
      "TuyaTransactionId": 9,
      "Valve 0x0b": "00000258"
    },
    "Bind": {
      "01": {
        "0019": {
          "Target": "0000",
          "Stamp": "11/03/2022 15:41:09",
          "Phase": "binded",
          "Status": "00",
          "i_sqn": 263
        },
        "ef00": {
          "Target": "0000",
          "Stamp": "11/03/2022 15:41:09",
          "Phase": "binded",
          "Status": "00",
          "i_sqn": 264
        }
      }
    },
    "WriteAttributes": {
      "Ep": {
        "01": {
          "0000": {
            "TimeStamp": "11/03/2022 15:49:01",
            "iSQN": {
              "ffde": 346
            },
            "Attributes": {
              "ffde": "00"
            },
            "ZigateRequest": {
              "ffde": {
                "Status": "fullfilled",
                "DataType": "20",
                "EPin": "01",
                "EPout": "01",
                "manuf_id": "0000",
                "manuf_spec": "00",
                "data": "13",
                "ackIsDisabled": false
              }
            }
          },
          "0004": {
            "TimeStamp": 0,
            "iSQN": {},
            "Attributes": {},
            "ZigateRequest": {}
          },
          "ef00": {
            "TimeStamp": 0,
            "iSQN": {},
            "Attributes": {},
            "ZigateRequest": {}
          },
          "000a": {
            "TimeStamp": 0,
            "iSQN": {},
            "Attributes": {},
            "ZigateRequest": {}
          },
          "0019": {
            "TimeStamp": 0,
            "iSQN": {},
            "Attributes": {},
            "ZigateRequest": {}
          }
        }
      }
    },
    "BindingTable": {
      "Devices": [
        {
          "2c1165fffeb3c71f": {
            "sourceEp": "01",
            "Cluster": "0019",
            "targetIEEE": "00158d0001eded61",
            "targetNickName": "Zigate",
            "targetEp": "01"
          }
        },
        {
          "2c1165fffeb3c71f": {
            "sourceEp": "01",
            "Cluster": "ef00",
            "targetIEEE": "00158d0001eded61",
            "targetNickName": "Zigate",
            "targetEp": "01"
          }
        }
      ],
      "SQN": 1,
      "TimeStamp": "11/03/2022 15:41:13",
      "BindingTableTableSize": 2,
      "Status": "Success"
    },
    "Param": {},
    "CheckParam": false,
    "NwkID": "6119"
  }
```
The devices seen in Domoticz:

[image]
```

pipiche38, 2022-03-11:
That is the main issue with Tuya manufacturer, manly each device is different.

In case you are not on beta6 branch, please follow this step by step 
https://github.com/zigbeefordomoticz/wiki/blob/zigpy/en-eng/Plugin_Version-6.md

* then switch to branch pipiche38/device/_TZE200_akjefhj5 

```
git pull
git checkout pipiche38/device/_TZE200_akjefhj5
```

* then you need to remove the 3 Thermostat widgets created

* restart the plugin

* then  redo the pairing

I cannot guaranty that it will work. I'm just relying on an other Water Valve from the same manufacturer

Try to play with the Command, and then redo an export.json and drop the file here ( it is better than a copy paste).

Depending on the result I'm might ask you to do further tests

pbda, 2022-03-12:
Thank you for your help and your quick answer.
I followed all the procedures (I was in stable 5 branch) and the device works fine now. In Domoticz there is now only one switch device which control the water valve :-)

I join export.json in case you need it.
Thank you again.
[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/8237280/export.json.txt)

pipiche38, 2022-03-12:
Super. 
If I'm not mistaken when you switch on the Valve, I guess it stay open for 300s ( 5 minutes), correct ?

pbda, 2022-03-12:
I didn't notice but you are right. More annoying, the state of valve is not reported to Domoticz.

pipiche38, 2022-03-12:
What do you mean the state of the valve is not reported ?

if you can add debugTuya and debugInput via the WebUI Tools and then reproduce the issue and send me the logs which are in Log/PluginZigbee-xx.log ( xx is the hardwareId of the plugin)

pbda, 2022-03-12:
> What do you mean the state of the valve is not reported ?

When I open valve in Domoticz widget, the valve is shown as "Open". 5 minutes later, the internal timer close the valve but the Domoticz widget always show the valve as "Open". Also, if I change manually the state of the valve, Domoticz widget didn't change.
Here is the log. I opened the valve with Domoticz at 20:23.47 and the valve closed itself at 20:28.09
[PluginZigbee_11.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/8238291/PluginZigbee_11.log)

pbda, 2022-03-12:
Another log. I manually opened the valve at 20:47.01 and I closed it at  20:47.21

[PluginZigbee_11.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/8238310/PluginZigbee_11.log)

pipiche38, 2022-03-13:
Thanks for the quality of your comments and logs.
please simply do a 

git pull

then  restart the plugin, the valve state should be reported now.

Let me know how it goes, if not working please drop me the logs once more

pbda, 2022-03-13:
It works. The valve state is reported now.
Thank you for your quick and efficient help.

pipiche38, 2022-03-13:
Oki .. thanks for your support, will merge it to beta6 branch
Plugin beta6 version 6.0.122'

closed 2022-03-13 by pipiche38