zigbeefordomoticz/Domoticz-Zigbee #1102: [zigpy-integration] - stack trace when starting deconz

pipiche38, 2022-03-14:
```
2022-03-14 21:40:06,097 WARNING :No response to 'Command.read_parameter' command with seq id '0x02'
2022-03-14 21:40:06,106 ERROR   :Error Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppDeconz.py", line 60, in startup
    await super().startup(auto_form=auto_form)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 61, in startup
    await self.connect()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 72, in connect
    self.version = await api.version()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 473, in version
    (self._proto_ver,) = await self[NetworkParameter.protocol_version]
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 438, in read_parameter
    r = await self._command(Command.read_parameter, 1 + len(data), param, data)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 323, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

cc: @puddly

pipiche38, 2022-03-14:
Ok, this is what I'm doing, so we should consider as a "normal" issue

pipiche38, 2022-03-18:
this looks to #1104

zigbeefordomoticz, 2022-03-23:
fixed

closed 2022-03-23 by zigbeefordomoticz