zigbeefordomoticz/Domoticz-Zigbee #1104: [zigpy-integration] - deconz serial line not stable

pipiche38, 2022-03-15:
Like #1102 where you have crash at plugin start one start over two, the serial line got stopped during the plugin run !!

```
2022-03-15 09:06:39,256 DEBUG   :get_device raise KeyError ieee: None nwk: f044 !!
2022-03-15 09:06:39,257 DEBUG   : [       ZigpyCom_6] App - get_device ieee:None nwk:0xf044
2022-03-15 09:06:39,257 DEBUG   :Received frame from unknown device: 0xf044
2022-03-15 09:06:39,258 DEBUG   : [       ZigpyCom_6] App - get_device calling callBackGetDevice None (<class 'NoneType'>) f044 (<class 'str'>)
2022-03-15 09:06:39,259 DEBUG   :'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xf044>, ep: 39, profile: 0x0104, cluster_id: 0x0400, data: b'184d0a000021b84c'
2022-03-15 09:06:39,260 INFO    : [       ZigpyCom_6] zigpy_get_device( None(<class 'NoneType'>), f044(<class 'str'>)) NOT FOUND
2022-03-15 09:06:40,729 DEBUG   :Frame received: 0x1c4b000c0005000244f0ffe4
2022-03-15 09:06:40,734 DEBUG   :Frame received: 0x1c4c000c0005000244f0ffe4
2022-03-15 09:06:42,040 DEBUG   :Frame received: 0x0e4d000700aa00
2022-03-15 09:06:42,041 DEBUG   :Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-03-15 09:06:42,042 DEBUG   :Command Command.aps_data_indication (1, 1)
2022-03-15 09:06:42,043 DEBUG   :Send: 0x174b000800010001
2022-03-15 09:06:42,047 DEBUG   :Frame received: 0x174b002a0023002202000001024acb01040119000c0001ba010021100e000342390000afffba090000b2
2022-03-15 09:06:42,048 DEBUG   :APS data indication response: [35, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xcb4a>, 1, 260, 25, b'\x01\xba\x01\x00!\x10\x0e\x00\x03B9\x00', 0, 175, 255, 186, 9, 0, 0, -78]
2022-03-15 09:06:42,051 DEBUG   :get_device raise KeyError ieee: None nwk: cb4a !!
2022-03-15 09:06:42,051 DEBUG   : [       ZigpyCom_6] App - get_device ieee:None nwk:0xcb4a
2022-03-15 09:06:42,052 DEBUG   :Received frame from unknown device: 0xcb4a
2022-03-15 09:06:42,053 DEBUG   : [       ZigpyCom_6] App - get_device calling callBackGetDevice None (<class 'NoneType'>) cb4a (<class 'str'>)
2022-03-15 09:06:42,054 DEBUG   :'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xcb4a>, ep: 1, profile: 0x0104, cluster_id: 0x0019, data: b'01ba010021100e0003423900'
2022-03-15 09:06:42,055 INFO    : [       ZigpyCom_6] zigpy_get_device( None(<class 'NoneType'>), cb4a(<class 'str'>)) NOT FOUND
2022-03-15 09:06:43,190 ERROR   :Lost serial connection: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 09:06:43,195 WARNING :Serial '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' connection lost unexpectedly: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 09:06:43,196 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 09:06:43,198 DEBUG   :Couldn't re-open '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port, retrying in 2s: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00'
2022-03-15 09:06:45,200 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 09:06:45,208 DEBUG   :Connection made
2022-03-15 09:06:45,209 DEBUG   :Reconnected '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port after 2 attempts
2022-03-15 09:06:48,310 ERROR   :Lost serial connection: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 09:06:48,317 WARNING :Serial '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' connection lost unexpectedly: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 09:06:48,318 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 09:06:48,320 DEBUG   :Couldn't re-open '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port, retrying in 2s: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00'
2022-03-15 09:06:50,321 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 09:06:50,327 DEBUG   :Connection made
2022-03-15 09:06:50,329 DEBUG   :Reconnected '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port after 2 attempts
```

CC: @puddly, @hedda

pipiche38, 2022-03-15:
Happening at the same time on the console
```
Mar 15 12:57:32 pi4 kernel: usb 1-1.2: USB disconnect, device number 94
Mar 15 12:57:32 pi4 kernel: usb 1-1.2: new full-speed USB device number 95 using xhci_hcd
Mar 15 12:57:32 pi4 kernel: usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar 15 12:57:32 pi4 kernel: usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 15 12:57:32 pi4 kernel: usb 1-1.2: Product: ConBee II
Mar 15 12:57:32 pi4 kernel: usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar 15 12:57:32 pi4 kernel: usb 1-1.2: SerialNumber: DE2227699
Mar 15 12:57:32 pi4 kernel: cdc_acm 1-1.2:1.0: ttyACM1: USB ACM device
Mar 15 12:57:37 pi4 kernel: usb 1-1.2: USB disconnect, device number 95
Mar 15 12:57:37 pi4 kernel: usb 1-1.2: new full-speed USB device number 96 using xhci_hcd
Mar 15 12:57:37 pi4 kernel: usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Mar 15 12:57:37 pi4 kernel: usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 15 12:57:37 pi4 kernel: usb 1-1.2: Product: ConBee II
Mar 15 12:57:37 pi4 kernel: usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Mar 15 12:57:37 pi4 kernel: usb 1-1.2: SerialNumber: DE2227699
Mar 15 12:57:37 pi4 kernel: cdc_acm 1-1.2:1.0: ttyACM1: USB ACM device
```

pipiche38, 2022-03-15:
trying to use the short port name /dev/ttyACM1 instead of the serial/by-id path

puddly, 2022-03-15:
Is it running the latest firmware? Do you have it on a USB 2.0 extension cable, away from USB 3.0 ports and other sources of interference?

Normally the Conbee shouldn't disconnect.

pipiche38, 2022-03-15:
Yes, latest firmware. It was on USB 3.0, I just have switch it to USB2.0 and monitor

pipiche38, 2022-03-15:
Same problem on USB 2.0 after a while```

2022-03-15 19:46:38,814 ERROR   :Lost serial connection: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 19:46:38,815 WARNING :Serial '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' connection lost unexpectedly: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 19:46:38,815 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 19:46:38,816 DEBUG   :Couldn't re-open '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port, retrying in 2s: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00'
2022-03-15 19:46:40,817 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 19:46:40,825 DEBUG   :Connection made
2022-03-15 19:46:40,827 DEBUG   :Reconnected '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port after 2 attempts
2022-03-15 19:46:43,680 ERROR   :Lost serial connection: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 19:46:43,681 WARNING :Serial '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' connection lost unexpectedly: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-15 19:46:43,682 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 19:46:43,684 DEBUG   :Couldn't re-open '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port, retrying in 2s: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00'
2022-03-15 19:46:45,687 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-15 19:46:45,693 DEBUG   :Connection made
2022-03-15 19:46:45,695 DEBUG   :Reconnected '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port after 2 attempts
```

pipiche38, 2022-03-16:
```
022-03-16 09:47:44,088 DEBUG   :APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 96]
2022-03-16 09:47:45,637 ERROR   :Lost serial connection: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-16 09:47:45,638 WARNING :Serial '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' connection lost unexpectedly: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-16 09:47:45,639 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-16 09:47:45,640 DEBUG   :Couldn't re-open '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port, retrying in 2s: [Errno 19] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00: [Errno 19] No such device: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00'
2022-03-16 09:47:47,641 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-16 09:47:47,647 DEBUG   :Connection made
2022-03-16 09:47:47,648 DEBUG   :Reconnected '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port after 2 attempts
2022-03-16 09:47:50,503 ERROR   :Lost serial connection: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-16 09:47:50,504 WARNING :Serial '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' connection lost unexpectedly: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-03-16 09:47:50,506 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-16 09:47:50,508 DEBUG   :Couldn't re-open '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port, retrying in 2s: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00'
2022-03-16 09:47:52,509 DEBUG   :Reconnecting '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port
2022-03-16 09:47:52,515 DEBUG   :Connection made
2022-03-16 09:47:52,516 DEBUG   :Reconnected '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2227699-if00' serial port after 2 attempts
2022-03-16 09:48:44,091 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-7998' coro=<transport_request() done, defined at /home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py:486> exception=TimeoutError()>
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 501, in transport_request
    result, msg = await self.app.request( destination, Profile, Cluster, sEp, dEp, sequence, payload, expect_reply, use_ieee )
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 399, in request
    r = await asyncio.wait_for(req.result, SEND_CONFIRM_TIMEOUT)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

Hedda, 2022-03-16:
Glitchy or faulty USB extension cable? Maybe try a other USB extension cable to at least rule out that USB extension cable?

pipiche38, 2022-03-16:
I'll continue to monitor it, there is no cable for now.

pipiche38, 2022-03-18:
happening several time a day, even when changing the USB cable.
so with or without cable , serial line is un-stable.

let see when @deufo, @jp-keros , @badz will get their device , to see if the same will happen

pipiche38, 2022-06-22:
https://github.com/zigpy/zigpy-deconz/issues/197

closed 2022-06-29 by pipiche38