zigbeefordomoticz/Domoticz-Zigbee #1105: [zigpy-integration] - stack trace crash

pipiche38, 2022-03-15:
```
2022-03-15 10:42:22,561 DEBUG   : [       ZigpyCom_6] Controller Manufacturer None
2022-03-15 10:42:22,562 DEBUG   : [       ZigpyCom_6] worker_loop - ZigyTransport: worker_loop start.
2022-03-15 10:42:22,563 DEBUG   : [ ZigpyForwarder_6] ZigateRead - MsgType: 8009,  Data: fffefffffffffffffffffffeffffffffffffffff00, LQI: 0
2022-03-15 10:42:22,564 DEBUG   : [ ZigpyForwarder_6] Decode8009: Network state - Address:fffe extaddr:ffffffffffffffff PanID: fffe Channel: 0
2022-03-15 10:42:22,578 DEBUG   : [ ZigpyForwarder_6] ZigateRead - MsgType: 8045,  Data: 000000000100, LQI: 0
2022-03-15 10:42:22,578 DEBUG   : [ ZigpyForwarder_6] Decode8045 - Reception Active endpoint response: SQN: 00, Status [00] Success, short Addr 0000, List 01, Ep list 00
2022-03-15 10:42:22,580 DEBUG   : [ ZigpyForwarder_6] ZigateRead - MsgType: 0302,  Data: , LQI: 0
2022-03-15 10:42:22,580 DEBUG   : [ ZigpyForwarder_6] Decode0302 - PDM Load
2022-03-15 10:42:22,581 DEBUG   : [ ZigpyForwarder_6] zcl_write_attributeNoResponse ffff 01 01 0000 1021 01 f000 23 00000000
2022-03-15 10:42:22,581 DEBUG   : [ ZigpyForwarder_6] zcl_raw_write_attributeNoResponse ffff 01 01 0000 1021 01 f000 23 00000000
2022-03-15 10:42:22,582 DEBUG   : [ ZigpyForwarder_6] rawaps_write_attribute_req ==== payload: 142110000500f02300000000
2022-03-15 10:42:22,583 INFO    : [ ZigpyForwarder_6] sendData       - [0] RAW-COMMAND {'Function': 'zcl_raw_write_attributeNoResponse', 'Profile': 260, 'Cluster': 0, 'TargetNwk': 65535, 'TargetEp': 1, 'SrcEp': 1, 'Sqn': 0, 'payload': '142110000500f02300000000', 'timestamp': 1647340942.576252, 'AddressMode': 2} 65535 Queue Length: 0
2022-03-15 10:42:22,584 DEBUG   : [ ZigpyForwarder_6] ===> sendData - Cmd: RAW-COMMAND Datas: {'Function': 'zcl_raw_write_attributeNoResponse', 'Profile': 260, 'Cluster': 0, 'TargetNwk': 65535, 'TargetEp': 1, 'SrcEp': 1, 'Sqn': 0, 'payload': '142110000500f02300000000', 'timestamp': 1647340942.576252, 'AddressMode': 2}
2022-03-15 10:42:22,647 DEBUG   :Sending Zigbee broadcast with tsn 1 under 2 request id, data: b'142110000500f02300000000'
2022-03-15 10:42:22,648 DEBUG   : [       ZigpyCom_6] got a command RAW-COMMAND
2022-03-15 10:42:22,649 DEBUG   : [       ZigpyCom_6] RAW-COMMAND: {'Function' : zcl_raw_write_attributeNoResponse,'Profile' : 0104,'Cluster' : 0000,'TargetNwk' : ffff,'TargetEp' : 01,'SrcEp' : 01,'Sqn' : 00,'payload' : 142110000500f02300000000,'timestamp' : 1647340942.576252,'AddressMode' : 02,}
2022-03-15 10:42:22,650 DEBUG   : [       ZigpyCom_6] ZigyTransport: process_raw_command ready to request Function: zcl_raw_write_attributeNoResponse NwkId: ffff/1 Cluster: 0000 Seq: 01 Payload: 142110000500f02300000000 AddrMode: 02 EnableAck: True, Sqn: 0
2022-03-15 10:42:22,650 DEBUG   : [       ZigpyCom_6] process_raw_command  call broadcast destination: ffff
2022-03-15 10:42:22,651 ERROR   : [       ZigpyCom_6] Error while receiving a Plugin command: >'NoneType' object has no attribute 'aps_data_request'<
2022-03-15 10:42:22,653 DEBUG   :Issue in request {'cmd': 'RAW-COMMAND', 'datas': {'Function': 'zcl_raw_write_attributeNoResponse', 'Profile': 260, 'Cluster': 0, 'TargetNwk': 65535, 'TargetEp': 1, 'SrcEp': 1, 'Sqn': 0, 'payload': '142110000500f02300000000', 'timestamp': 1647340942.576252, 'AddressMode': 2}, 'NwkId': 65535, 'TimeStamp': 1647340942.5766456, 'ACKIsDisable': False, 'Sqn': 0}, dumping stack: Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 204, in worker_loop
    await dispatch_command(self, data)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 313, in dispatch_command
    await process_raw_command(self, data["datas"], AckIsDisable=data["ACKIsDisable"], Sqn=data["Sqn"])
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 358, in process_raw_command
    result, msg = await self.app.broadcast( Profile, Cluster, sEp, dEp, 0x0, 0x0, sequence, payload, )
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 439, in broadcast
    await self._api.aps_data_request(
AttributeError: 'NoneType' object has no attribute 'aps_data_request'
```

puddly, 2022-03-15:
Radio is not connected if `self._api` is `None`.

pipiche38, 2022-03-15:
Ok, but if I simply restart it works. so it failed with this error, then a restart again works !
And all is based on the new radio API, so I don't think that I'm doing anything with self._api

It looks like there is this error just before which could explain why self._api is not set !

2022-03-15 16:47:23,030 WARNING :No response to 'Command.read_parameter' command with seq id '0x02'
2022-03-15 16:47:23,039 ERROR   :Error Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:
```

Traceback (most recent call last):
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppDeconz.py", line 60, in startup
    await super().startup(auto_form=auto_form)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 61, in startup
    await self.connect()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/zigbee/application.py", line 72, in connect
    self.version = await api.version()
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 473, in version
    (self._proto_ver,) = await self[NetworkParameter.protocol_version]
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 438, in read_parameter
    r = await self._command(Command.read_parameter, 1 + len(data), param, data)
  File "/home/domoticz/domoticz/plugins/Domoticz-Zigbee/zigpy_deconz/api.py", line 323, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

closed 2022-03-23 by zigbeefordomoticz