zigbeefordomoticz/Domoticz-Zigbee #1106: [zigpy-integration] -  await self.load_network_info( load_devices=True ) load nothing

pipiche38, 2022-03-15:
based on new-radio-API

On a ConBee running with 15 devices ( 7 routers and the rest are End devices), laod_network_info ( load_devices = True )
do not load anything in self.state.network_info

`2022-03-15 10:57:33,741 DEBUG   :startup Network Info: NetworkInfo(extended_pan_id=ab:cd:48:45:04:01:5e:10, pan_id=0x1545, nwk_update_id=0, nwk_manager_id=0x0000, channel=15, channel_mask=<Channels.CHANNEL_15: 32768>, security_level=5, network_key=Key(key=[255, 161, 163, 45, 166, 54, 220, 21, 82, 199, 120, 178, 62, 239, 84, 54], tx_counter=76092, rx_counter=0, seq=0, partner_ieee=ff:ff:ff:ff:ff:ff:ff:ff), tc_link_key=Key(key=[90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57], tx_counter=0, rx_counter=0, seq=0, partner_ieee=00:21:2e:ff:ff:05:fe:7f), key_table=[], children=[], nwk_addresses={}, stack_specific={})`

CC: @puddly

puddly, 2022-03-15:
Conbee doesn't expose this information. That's why you have to reload the child table from the application database on startup: https://github.com/zigpy/zigpy-deconz/pull/126

pipiche38, 2022-03-15:
Ok. that is a shame as this make the things more challenging when a device change its ShortId when the plugin is disconnected.

Hedda, 2022-03-16:
> Conbee doesn't expose this information. That's why you have to reload the child table from the application database on startup: [zigpy/zigpy-deconz#126](https://github.com/zigpy/zigpy-deconz/pull/126)

@manup If that or other functions/features are missing from deCONZ Serial Protocol then should devs post feature request to https://github.com/dresden-elektronik/deconz-serial-protocol so it could be considered for future ConBee/RaspBee firmware?

pipiche38, 2022-03-18:
So as per design this is not loading any devices.

closed 2022-03-18 by pipiche38

manup, 2022-03-20:
> @manup If that or other functions/features are missing from deCONZ Serial Protocol then should devs post feature request to https://github.com/dresden-elektronik/deconz-serial-protocol so it could [...]

If I understand the issue right this is about the network topology?

The coordinator only knows it's own neighbor table, the application is expected to query the routers in the network (including the coordinator) to get the neighbor tables via standard *ZDP Mgmt_Lqi_req*. This has the advantage that the network size can be fairly large without having an impact on the coordinator RAM usage.