zigbeefordomoticz/Domoticz-Zigbee #1114: [Stable6] To Do List

jp-keros, 2022-03-23:
Liste des choses à faire pour la sortie de la stable6 :

- [ ] Mettre à jour les liens du webUI vers le wikiUI
     - [ ] Modifier les traductions > @jp-keros
     - [ ] Pousser la modif des traductions > @jp-keros
     - [ ] Mettre à jour le plugin > @pipiche38

- [ ] Mettre le WikiUI en ligne
     - [ ] Merge de la branche Github-page dans Zigpy > @jp-keros 
     - [ ] Merge de la branche ZigPy dans Master > @jp-keros ou @pipiche38 
     - [ ] Modifier le paramètre de la GitHub-page pour qu'elle pointe sur la branche master > @pipiche38 

- [ ] Créer la stable6 > @pipiche38 

- [ ] Faire la communication
     - [ ] Sur Keybase > @pipiche38 
     - [ ] Sur le forum Anglais > @pipiche38 
     - [ ] Sur le forum Français > @pipiche38 

- [ ] Prévenir nos partenaires (mettre un message dans les sujets de discussions ouverts)
     - [ ] zigpy
     - [ ] electrolama
     - [ ] elelabs
     - [ ] Bellows

À compléter, Modifier et cocher ;-)

jp-keros, 2022-03-24:
Les points à communiquer :

- Prise en charge des Coordinateurs TI
- Procédure particulière pour le passage en V6
- Modification des liens vers le wiki avec un affichage sous forme de site web
- Possibilité d'intégrer le WebUI directement dans DomoticZ via le custom menu (voir étape 4 de la procédure d'installation)
- Mise en place de statistiques sur l'utilisation du plugin.
- Future ouverture aux coordinateurs bellows et conbee

pipiche38, 2022-03-25:
la création de stable6 se ferra au dernier moment, car sinon, on risque d'avoir du monde qui bascule dessus

pipiche38, 2022-03-31:
@hedda, @puddly for your information we are in the final lap to make the first release of the plugin including zigpy layers and mainly the zigpy-znp and bellows radio libraries.

Up to now it was on beta, and now we consider ready for stable version. this will be pushed during the week-end and announced on the different Domoticz forum.

For that release we will rely on the legacy radio api

thanks a lot for your support and looking forward to continue our working relationship

Hedda, 2022-04-01:
Nice! Like to suggest writing on GitHub and wikis in English first as primary published language and then translate to French after.

By writing in French first as primary and only later translating to English you risk alienating/excluding new non-French users/devs.

jp-keros, 2022-04-01:
That is all my fault.

Considering this issue, it is written in French because I needed somewhere to write quick notes.

Considering the wiki, at the end of 2020, Pipiche asked for a translation of the wiki in French. I did it and I also improved it. I was able to think about formatting because I was thinking in French. I could not have done the same work in English directly. 

But if you look at the solved point of the wiki https://github.com/zigbeefordomoticz/wiki/issues?q=is%3Aissue+is%3Aclosed, the English element are done at the same moments and there are all checked  - [x] 

I know, all the wiki issues are written in French. I consider them as quick notes for me, so I wrote them in French.

Pipiche asked me several times if can do it in English first : I can not, sorry.

pipiche38, 2022-04-01:
I did update the PP-Manager to use stable6 branch instead of stable5

pipiche38, 2022-04-02:
j'ai updaté egalement le Wiki Domoticz avec les nouveaux liens.

jp-keros, 2022-04-02:
Every thing is checked, this issue may now be closed  ;-)

Thanks for the work done

closed 2022-04-02 by jp-keros