zigbeefordomoticz/Domoticz-Zigbee #1120: [zigpy-integration] - changing channel seems not working with deconz

pipiche38, 2022-03-29:
I'm trying to change zigbee channel from 15 to 26, but it looks like the ConBee remains on channel 15 .

`2022-03-29 09:45:44,394 DEBUG   : [       MainThread] zdp_raw_nwk_update_request ffff 00000004 fe 00 01
2022-03-29 09:45:44,398 DEBUG   : [       MainThread] zdp_raw_nwk_update_request Payload: 0000000004fe01
2022-03-29 09:45:44,399 INFO    : [       MainThread] zdp_raw_nwk_update_request  - [00] ffff 0000000004fe01 Queue Length: 0
2022-03-29 09:45:44,400 INFO    : [       MainThread] sendData       - [0] RAW-COMMAND {'Function': 'zdp_raw_nwk_update_request', 'Profile': 0, 'Cluster': 56, 'TargetNwk': 65535, 'TargetEp': 0, 'SrcEp': 0, 'Sqn': 0, 'payload': '0000000004fe01', 'timestamp': 1648543544.393418, 'AddressMode': 2} 65535 Queue Length: 0
2022-03-29 09:45:44,402 INFO    : [       MainThread] sendData       - [0] RAW-COMMAND {'Function': 'zdp_raw_nwk_update_request', 'Profile': 0, 'Cluster': 56, 'TargetNwk': 0, 'TargetEp': 0, 'SrcEp': 0, 'Sqn': 0, 'payload': '0000000004fe01', 'timestamp': 1648543544.3959684, 'AddressMode': 2} 0 Queue Length: 1
2022-03-29 09:45:44,403 INFO    : [       MainThread] sendData       - [None] REQ-NWK-STATUS None None Queue Length: 2
2022-03-29 09:45:44,411 DEBUG   :Sending Zigbee broadcast with tsn 35 under 36 request id, data: b'0000000004fe01'
2022-03-29 09:45:44,413 DEBUG   :Command Command.aps_data_request (22, 36, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=65533 endpoint=0>, 0, 56, 0, b'\x00\x00\x00\x00\x04\xfe\x01', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-03-29 09:45:44,414 DEBUG   : [       ZigpyCom_6] got a command RAW-COMMAND
2022-03-29 09:45:44,414 DEBUG   :Send: 0x1285001d001600240002fdff00000038000007000000000004fe010200
2022-03-29 09:45:44,415 DEBUG   : [       ZigpyCom_6] RAW-COMMAND: {'Function' : zdp_raw_nwk_update_request,'Profile' : 0000,'Cluster' : 0038,'TargetNwk' : ffff,'TargetEp' : 00,'SrcEp' : 00,'Sqn' : 00,'payload' : 0000000004fe01,'timestamp' : 1648543544.393418,'AddressMode' : 02,}
2022-03-29 09:45:44,417 DEBUG   : [       ZigpyCom_6] ZigyTransport: process_raw_command ready to request Function: zdp_raw_nwk_update_request NwkId: ffff/0 Cluster: 0038 Seq: 23 Payload: 0000000004fe01 AddrMode: 02 EnableAck: True, Sqn: 0
2022-03-29 09:45:44,418 DEBUG   : [       ZigpyCom_6] process_raw_command  call broadcast destination: ffff
2022-03-29 09:45:44,423 DEBUG   :Frame received: 0x128500090002002224
2022-03-29 09:45:44,424 DEBUG   :APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 36]
2022-03-29 09:45:44,863 DEBUG   :Frame received: 0x1c86000c00050002ba66ffd3
2022-03-29 09:45:45,070 DEBUG   :Frame received: 0x0e87000700a600
2022-03-29 09:45:45,071 DEBUG   :Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-03-29 09:45:45,073 DEBUG   :Command Command.aps_data_confirm (0,)
2022-03-29 09:45:45,073 DEBUG   :Send: 0x04860007000000
2022-03-29 09:45:45,076 DEBUG   :Frame received: 0x04860013000c00222402fdff00000000000000
2022-03-29 09:45:45,077 DEBUG   :APS data confirm response for request with id 36: 00
2022-03-29 09:45:45,078 DEBUG   :Request id: 0x24 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xfffd endpoint=0>, status: 0x00
2022-03-29 09:45:45,082 DEBUG   :Sending Zigbee request with tsn 37 under 38 request id, data: b'0000000004fe01'
2022-03-29 09:45:45,083 DEBUG   : [       ZigpyCom_6] got a command RAW-COMMAND
2022-03-29 09:45:45,084 DEBUG   :Command Command.aps_data_request (22, 38, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x0000 endpoint=0>, 0, 56, 0, b'\x00\x00\x00\x00\x04\xfe\x01', <DeconzTransmitOptions.USE_APS_ACKS|USE_NWK_KEY_SECURITY: 6>, 0)
2022-03-29 09:45:45,084 DEBUG   : [       ZigpyCom_6] RAW-COMMAND: {'Function' : zdp_raw_nwk_update_request,'Profile' : 0000,'Cluster' : 0038,'TargetNwk' : 0000,'TargetEp' : 00,'SrcEp' : 00,'Sqn' : 00,'payload' : 0000000004fe01,'timestamp' : 1648543544.3959684,'AddressMode' : 02,}
2022-03-29 09:45:45,086 DEBUG   :Send: 0x1287001d001600260002000000000038000007000000000004fe010600
2022-03-29 09:45:45,087 DEBUG   : [       ZigpyCom_6] ZigyTransport: process_raw_command ready to request Function: zdp_raw_nwk_update_request NwkId: 0000/0 Cluster: 0038 Seq: 25 Payload: 0000000004fe01 AddrMode: 02 EnableAck: True, Sqn: 0
2022-03-29 09:45:45,089 DEBUG   : [       ZigpyCom_6] App - get_device ieee:None nwk:0x0000
2022-03-29 09:45:45,090 DEBUG   : [       ZigpyCom_6] App - get_device found device: <DeconzDevice model='ConBee II' manuf='dresden elektronik' nwk=0x0000 ieee=00:21:2e:ff:ff:05:fe:7f is_initialized=True>
2022-03-29 09:45:45,091 DEBUG   :Frame received: 0x128700090002002226
2022-03-29 09:45:45,092 DEBUG   : [       ZigpyCom_6] process_raw_command  call request destination: <DeconzDevice model='ConBee II' manuf='dresden elektronik' nwk=0x0000 ieee=00:21:2e:ff:ff:05:fe:7f is_initialized=True> Profile: 0 Cluster: 56 sEp: 0 dEp: 0 Seq: 37 Payload: b'\x00\x00\x00\x00\x04\xfe\x01'
2022-03-29 09:45:45,093 DEBUG   :APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 38]
2022-03-29 09:45:45,094 DEBUG   : [       ZigpyCom_6] got a command REQ-NWK-STATUS
2022-03-29 09:45:45,095 DEBUG   :Frame received: 0x0e88000700a600
2022-03-29 09:45:45,096 DEBUG   :Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-03-29 09:45:45,098 DEBUG   :Command Command.aps_data_confirm (0,)
2022-03-29 09:45:45,098 DEBUG   :Send: 0x04880007000000
2022-03-29 09:45:45,101 DEBUG   :Frame received: 0x04880013000c00222602000000000000000000
2022-03-29 09:45:45,102 DEBUG   :APS data confirm response for request with id 38: 00
2022-03-29 09:45:45,103 DEBUG   :Request id: 0x26 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x0000 endpoint=0>, status: 0x00
2022-03-29 09:45:45,105 DEBUG   : [       ZigpyCom_6] ZigyTransport: process_raw_command completed 37 NwkId: 0000 result: TXStatus.SUCCESS msg: message send success
2022-03-29 09:45:45,867 DEBUG   :Frame received: 0x1c89000c00050002ba66ffd5
2022-03-29 09:45:46,874 DEBUG   :Frame received: 0x1c8a000c00050002ba66ffd5
2022-03-29 09:45:47,881 DEBUG   :Frame received: 0x1c8b000c00050002ba66ffd4
2022-03-29 09:45:48,888 DEBUG   :Frame received: 0x1c8c000c00050002ba66ffd5
2022-03-29 09:45:49,892 DEBUG   :Frame received: 0x1c8d000c00050002ba66ffd5
2022-03-29 09:45:50,897 DEBUG   :Frame received: 0x1c8e000c00050002ba66ffd5
2022-03-29 09:45:51,905 DEBUG   :Frame received: 0x1c8f000c00050002ba66ffd5
2022-03-29 09:45:52,911 DEBUG   :Frame received: 0x1c90000c00050002ba66ffd5
2022-03-29 09:45:53,916 DEBUG   :Frame received: 0x1c91000c00050002ba66ffd4
2022-03-29 09:45:54,923 DEBUG   :Frame received: 0x1c92000c00050002ba66ffd4
2022-03-29 09:45:55,084 DEBUG   : [       ZigpyCom_6] build_plugin_8009_frame_content 0x0000 00:21:2e:ff:ff:05:fe:7f dc:ba:48:45:04:01:5e:10 0x047b 15
2022-03-29 09:45:55,087 INFO    : [ ZigpyForwarder_6] Zigbee Coordinator ieee: 00212effff05fe7f , short addr: 0000
2022-03-29 09:45:55,089 INFO    : [ ZigpyForwarder_6] Network state UP, PANID: 047b extPANID: 0xdcba484504015e10 Channel: 15
2022-03-29 09:45:55,929 DEBUG   :Frame received: 0x1c93000c00050002ba66ffd5`

Hedda, 2022-03-29:
> I'm trying to change zigbee channel from 15 to 26, but it looks like the ConBee remains on channel 15 .

Did you also try changing to Zigbee channel 20 to 23 as it could just be that only Zigbee channel 26 is not supported in firmware? 

I am sure you are already aware of this but for reference: 

I understand that especially Zigbee channels 11, 16, 24, 25 and 26 are not supported by many older Zigbee devices. And Zigbee channel 26 may also be restricted in TX power in some countries so been set to use lower power than all the others channels.

Zigbee channels 11, 15, 20, or 25 was initially the "primary ZLL channels" for Zigbee Light Link, and for backwards compatibility or whatever historic reasons some Zigbee devices will only pair on those four primary ZZL channels by default. 

Even though newer ZLL devices should scan and be able to operate on all channels (11 to 26) if no response is received on one of the primary ZLL channels, the general recommendation given to end-users will still usually be to preferably use Zigbee channels 15, 20, or 25 for best compatibility in order to avoid interoperability problems with older  Zigbee devices that are limited to only being compatible with pairing on those Zigbee channels. 

So at least end-users should especially consider not using Zigbee channels 11, 16, 24, 25, or 26 at all as otherwise, it could mean that they get problems pairing older devices as pairing on what constitutes as "secondary ZLL channels" are commonly only supported by relatively modern Zigbee hardware devices with newer Zigbee firmware.

PS: Best is even to only use channels 15, 20, or 25 as well as lessen the chance of Wi-Fi networks interfering too much with the Zigbee network).

pipiche38, 2022-03-29:
@hedda , 26 is used by Livolo switches for pairing, this is why I did the test on 26
But the issue is the same with other channel, I believe that is the same as it was for ZNP, the coordinator do not get the network change request and only broadcast it to the routers.

pipiche38, 2022-04-02:
https://github.com/zigpy/zigpy-deconz/issues/192

closed 2023-01-27 by github-actions[bot]