zigbeefordomoticz/Domoticz-Zigbee #1142: [zigpy-Errors] - ZNP happening on user

pipiche38, 2022-04-16:
```
2022-04-16 11:48:08,176 ERROR   :Task exception was never retrieved
future: <Task finished coro=<ControllerApplication.on_zdo_message() done, defined at /home/pi/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py:643> exception=ValueError('Data is too short to contain 2 bytes')>
Traceback (most recent call last):
  File "/home/pi/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 651, in on_zdo_message
    args, data = list_deserialize(data, types)
  File "/home/pi/domoticz/plugins/Domoticz-Zigbee/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/home/pi/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 142, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 2 bytes
```

CC: @puddly

pipiche38, 2022-04-18:
@puddly it looks like we have this error time to time especially during the startup phase. Is that something we should consider as a normal error ?

puddly, 2022-04-18:
Without a debug log it's hard to tell what the error is, only that it happens when parsing a ZDO command from some device.

pipiche38, 2022-04-18:
Hard to get more, as these are logs file that I'm collecting from users, next time If I get it on my test system I'll give you more

pipiche38, 2022-04-19:
@puddly, do hope that is helping

`
2022-04-19 22:09:43,962 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32817, SecurityUse=0, TSN=1, MacDst=0x0000, Data=b'\x00\x07\x00\x02\xA3\xC5\x33\x35\xD0\xD2\x3A\xE3\xA1\xC7\x86\xFE\xFF\x9F\xFD\x90\xDE\xF2\x15\x02\x01\x34\xA3\xC5\x33\x35\xD0\xD2\x3A\xE3\x27\xAA\x01\x1A\x00\xBC\x15\x00\x67\xE9\x12\x02\x01\xAA')
2022-04-19 22:09:43,987 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Lqi_rsp: [<Status.SUCCESS: 0>, Neighbors(entries=7, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=90:fd:9f:ff:fe:86:c7:a1, nwk=0xF2DE, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=52), Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:15:bc:00:1a:01:aa:27, nwk=0xE967, device_type=<DeviceType.EndDevice: 2>, rx_on_when_idle=<RxOnWhenIdle.Off: 0>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=170)])]
2022-04-19 22:09:43,992 DEBUG   :[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Lqi_rsp([<Status.SUCCESS: 0>, Neighbors(entries=7, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=90:fd:9f:ff:fe:86:c7:a1, nwk=0xF2DE, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=52), Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:15:bc:00:1a:01:aa:27, nwk=0xE967, device_type=<DeviceType.EndDevice: 2>, rx_on_when_idle=<RxOnWhenIdle.Off: 0>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=170)])])
2022-04-19 22:09:44,217 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:00:00:00:00:00:e2:1d, 0, 0]
2022-04-19 22:09:44,225 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x0000), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=0, TSN=2, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x02\x1D\xE2\x00\x00\x00\x00\x00\x00\x00\x00')
2022-04-19 22:09:44,255 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-04-19 22:09:44,262 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32768, SecurityUse=0, TSN=2, MacDst=0x0000, Data=b'\x81\x1D\xE2\x00\x00\x00\x00\x00\x00\xFF\xFF')
2022-04-19 22:09:44,279 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.NWK_addr_rsp: [<Status.DEVICE_NOT_FOUND: 129>, 00:00:00:00:00:00:e2:1d, 0xFFFF, None, None, []]
2022-04-19 22:09:44,284 DEBUG   :[0x0000:zdo] No handler for ZDO request:ZDOCmd.NWK_addr_rsp([<Status.DEVICE_NOT_FOUND: 129>, 00:00:00:00:00:00:e2:1d, 0xFFFF, None, None, []])
2022-04-19 22:09:44,526 ERROR   :Failed to parse message (b'0281000000000000e21d') on cluster 32768, because Data is too short to contain 2 bytes
2022-04-19 22:09:44,531 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x0000), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=32768, TSN=2, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x02\x81\x00\x00\x00\x00\x00\x00\xE2\x1D')
2022-04-19 22:09:44,553 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-04-19 22:09:44,559 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32768, SecurityUse=0, TSN=2, MacDst=0x0000, Data=b'\x81\x00\x00\x00\x00\x00\x00\xE2\x1D')
2022-04-19 22:09:44,567 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-98' coro=<ControllerApplication.on_zdo_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py:643> exception=ValueError('Data is too short to contain 2 bytes')>
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 651, in on_zdo_message
    args, data = list_deserialize(data, types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 142, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 2 bytes
2022-04-19 22:09:44,569 INFO    : [ ZigpyForwarder_4] Decode8040 - Reception of Node Descriptor for 1de2000000000000 with status 81

`

puddly, 2022-04-19:
It's a bad ZDO `NWK_addr_rsp` response:

```
[<Status.DEVICE_NOT_FOUND: 129>, 00:00:00:00:00:00:e2:1d, 0xFFFF, None, None, []]
                                                          ^^^^^^
                                                          missing
```

The IEEE address looks wrong too.

> If the NWK_addr_rsp command is received with a Status of SUCCESS, the remaining fields of the command contain the appropriate discovery information, according to the RequestType as specified in the original NWK_Addr_req command. **Otherwise, the Status field indicates the error and the NumAssocDev, StartIndex, and NWKAddrAssocDevList fields shall not be included.**

I'll add logging to ZNP to generate a more helpful error message.

pipiche38, 2022-04-20:
I'll try to work on the recent version, but this looks to be done by the zigpy lib itself. from plugin perspective we are not doing a Network Address Request.

pipiche38, 2022-05-02:
@puddly, Here is a zigpy traces, with a more recent version I believe. 
I didn't see any request from the plugin. Can it be that zigpy layer is performing a NWK_addr_req when receiving a Mngt_LQI_response ?  

```

2022-05-02 17:25:18,614 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1, SrcAddr=0x0BD8, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=131, SecurityUse=<Bool.false: 0>, TimeStamp=9379656, TSN=0, Data=b'\x08\x68\x0A\x20\x00\x20\x25', MacSrcAddr=0x0BD8, MsgResultRadius=29)
2022-05-02 17:25:18,646 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Lqi_req: [0]
2022-05-02 17:25:18,651 DEBUG   :[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Lqi_req([0])
2022-05-02 17:25:18,658 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x0000), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=49, TSN=53, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x00\x00')
2022-05-02 17:25:18,682 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-05-02 17:25:18,700 DEBUG   :Received command: ZDO.MgmtLqiRsp.Callback(Src=0x0000, Status=<Status.SUCCESS: 0>, Neighbors=Neighbors(entries=7, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:0d:6f:00:11:08:70:79, nwk=0x0BD8, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=130), Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:15:8d:00:02:b0:66:15, nwk=0x119A, device_type=<DeviceType.EndDevice: 2>, rx_on_when_idle=<RxOnWhenIdle.Off: 0>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=170)]))
2022-05-02 17:25:18,707 DEBUG   :Command was not handled
2022-05-02 17:25:18,715 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32817, SecurityUse=0, TSN=0, MacDst=0x0000, Data=b'\x00\x07\x00\x02\xA3\xC5\x33\x35\xD0\xD2\x3A\xE3\x79\x70\x08\x11\x00\x6F\x0D\x00\xD8\x0B\x15\x02\x01\x82\xA3\xC5\x33\x35\xD0\xD2\x3A\xE3\x15\x66\xB0\x02\x00\x8D\x15\x00\x9A\x11\x12\x02\x01\xAA')
2022-05-02 17:25:18,738 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Lqi_rsp: [<Status.SUCCESS: 0>, Neighbors(entries=7, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:0d:6f:00:11:08:70:79, nwk=0x0BD8, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=130), Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:15:8d:00:02:b0:66:15, nwk=0x119A, device_type=<DeviceType.EndDevice: 2>, rx_on_when_idle=<RxOnWhenIdle.Off: 0>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=170)])]
2022-05-02 17:25:18,744 DEBUG   :[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Lqi_rsp([<Status.SUCCESS: 0>, Neighbors(entries=7, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:0d:6f:00:11:08:70:79, nwk=0x0BD8, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=130), Neighbor(extended_pan_id=e3:3a:d2:d0:35:33:c5:a3, ieee=00:15:8d:00:02:b0:66:15, nwk=0x119A, device_type=<DeviceType.EndDevice: 2>, rx_on_when_idle=<RxOnWhenIdle.Off: 0>, relationship=<RelationShip.Child: 1>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=170)])])
2022-05-02 17:25:19,222 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:00:00:00:00:00:0b:d8, 0, 0]
2022-05-02 17:25:19,228 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x0000), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=0, TSN=1, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\xD8\x0B\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-02 17:25:19,253 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-05-02 17:25:19,257 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32768, SecurityUse=0, TSN=1, MacDst=0x0000, Data=b'\x81\xD8\x0B\x00\x00\x00\x00\x00\x00\xFF\xFF')
2022-05-02 17:25:19,267 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.NWK_addr_rsp: [<Status.DEVICE_NOT_FOUND: 129>, 00:00:00:00:00:00:0b:d8, 0xFFFF, None, None, []]
2022-05-02 17:25:19,271 DEBUG   :[0x0000:zdo] No handler for ZDO request:ZDOCmd.NWK_addr_rsp([<Status.DEVICE_NOT_FOUND: 129>, 00:00:00:00:00:00:0b:d8, 0xFFFF, None, None, []])
2022-05-02 17:25:19,772 ERROR   :Failed to parse message (b'01810000000000000bd8') on cluster 32768, because Data is too short to contain 2 bytes
2022-05-02 17:25:19,778 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x0000), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=32768, TSN=1, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x81\x00\x00\x00\x00\x00\x00\x0B\xD8')
[... cut ...]
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 651, in on_zdo_message
    args, data = list_deserialize(data, types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 142, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 2 bytes
```

puddly, 2022-05-02:
Hmm, looking more closely it seems like this is a loopback message, coming from the radio itself (`0x0000`).

Can you post the last minute of logs?

pipiche38, 2022-05-03:
@puddly hope it is enought.

```
2022-05-02 17:18:58,460 DEBUG   :Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-05-02 17:18:58,465 DEBUG   :Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0080] = TCLKDevEntry(txFrmCntr=0, rxFrmCntr=0, extAddr=00:00:00:00:00:00:00:00, keyAttributes=<KeyAttributes.DEFAULT_KEY: 255>, keyType=<KeyType.NONE: 0>, SeedShift_IcIndex=0)
2022-05-02 17:18:58,467 DEBUG   :Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=129)
2022-05-02 17:18:58,480 DEBUG   :Received command: SYS.NVLength.Rsp(Length=20)
2022-05-02 17:18:58,484 DEBUG   :Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=129, Offset=0, Length=20)
2022-05-02 17:18:58,499 DEBUG   :Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-05-02 17:18:58,504 DEBUG   :Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0081] = TCLKDevEntry(txFrmCntr=0, rxFrmCntr=0, extAddr=00:00:00:00:00:00:00:00, keyAttributes=<KeyAttributes.DEFAULT_KEY: 255>, keyType=<KeyType.NONE: 0>, SeedShift_IcIndex=0)
2022-05-02 17:18:58,507 DEBUG   :Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=130)
2022-05-02 17:18:58,520 DEBUG   :Received command: SYS.NVLength.Rsp(Length=20)
2022-05-02 17:18:58,523 DEBUG   :Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=130, Offset=0, Length=20)
2022-05-02 17:18:58,539 DEBUG   :Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-05-02 17:18:58,544 DEBUG   :Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0082] = TCLKDevEntry(txFrmCntr=0, rxFrmCntr=0, extAddr=00:00:00:00:00:00:00:00, keyAttributes=<KeyAttributes.DEFAULT_KEY: 255>, keyType=<KeyType.NONE: 0>, SeedShift_IcIndex=0)
2022-05-02 17:18:58,546 DEBUG   :Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=131)
2022-05-02 17:18:58,559 DEBUG   :Received command: SYS.NVLength.Rsp(Length=20)
2022-05-02 17:18:58,563 DEBUG   :Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=131, Offset=0, Length=20)
2022-05-02 17:18:58,578 DEBUG   :Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-05-02 17:18:58,583 DEBUG   :Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0083] = TCLKDevEntry(txFrmCntr=0, rxFrmCntr=0, extAddr=00:00:00:00:00:00:00:00, keyAttributes=<KeyAttributes.DEFAULT_KEY: 255>, keyType=<KeyType.NONE: 0>, SeedShift_IcIndex=0)
2022-05-02 17:18:58,586 DEBUG   :Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=132)
2022-05-02 17:18:58,600 DEBUG   :Received command: SYS.NVLength.Rsp(Length=20)
2022-05-02 17:18:58,603 DEBUG   :Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=132, Offset=0, Length=20)
2022-05-02 17:18:58,619 DEBUG   :Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-05-02 17:18:58,624 DEBUG   :Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0084] = TCLKDevEntry(txFrmCntr=0, rxFrmCntr=0, extAddr=00:00:00:00:00:00:00:00, keyAttributes=<KeyAttributes.DEFAULT_KEY: 255>, keyType=<KeyType.NONE: 0>, SeedShift_IcIndex=0)
2022-05-02 17:18:58,627 DEBUG   :Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=133)
2022-05-02 17:18:58,640 DEBUG   :Received command: SYS.NVLength.Rsp(Length=20)
2022-05-02 17:18:58,644 DEBUG   :Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=133, Offset=0, Length=20)
2022-05-02 17:18:58,659 DEBUG   :Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-05-02 17:18:58,664 DEBUG   :Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0085] = TCLKDevEntry(txFrmCntr=0, rxFrmCntr=0, extAddr=00:00:00:00:00:00:00:00, keyAttributes=<KeyAttributes.DEFAULT_KEY: 255>, keyType=<KeyType.NONE: 0>, SeedShift_IcIndex=0)
2022-05-02 17:18:58,667 DEBUG   :Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=134)
2022-05-02 17:18:58,680 DEBUG   :Received command: SYS.NVLength.Rsp(Length=20)
2022-05-02 17:18:58,683 DEBUG   :Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=134, Offset=0, Length=20)
2022-05-02 17:18:58,699 DEBUG   :Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-05-02 17:18:58,704 DEBUG   :Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0086] = TCLKDevEntry(txFrmCntr=0, rxFrmCntr=0, extAddr=00:00:00:00:00:00:00:00, keyAttributes=<KeyAttributes.DEFAULT_KEY: 255>, keyType=<KeyType.NONE: 0>, SeedShift_IcIndex=0)
[... cut ...]
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 651, in on_zdo_message
    args, data = list_deserialize(data, types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 142, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 2 bytes

```

pipiche38, 2022-05-10:
```
2022-05-10 11:10:56,114 ERROR   :Failed to parse message (b'038100000000000016dd') on cluster 32768, because Data is too short to contain 2 bytes
2022-05-10 11:10:56,131 INFO    : [ ZigpyForwarder_3] Decode8040 - Reception of Node Descriptor for dd16000000000000 with status 81
2022-05-10 11:10:56,150 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-95842' coro=<ControllerApplication.on_zdo_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:643> exception=ValueError('Data is too short to contain 2 bytes')>
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 651, in on_zdo_message
    args, data = list_deserialize(data, types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy/types/basic.py", line 142, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 2 bytes
2022-05-10 11:10:56,152 INFO    : [       ZigpyCom_3] ZigyTransport: process_raw_command  00:12:4b:00:25:6c:12:07 0 (<class 'int'>) 32768 (<class 'int'>)
2022-05-10 11:10:56,192 INFO    : [       ZigpyCom_3] transport_request: _limit_concurrency <Device model=None manuf=None nwk=0x16DD ieee=00:15:8d:00:02:c6:3a:f3 is_initialized=False> 1
2022-05-10 11:10:56,243 INFO    : [       ZigpyCom_3] ZigyTransport: process_raw_command  00:15:8d:00:02:c6:3a:f3 0 (<class 'int'>) 33 (<class 'int'>)
2022-05-10 11:10:56,528 ERROR   :Failed to parse message (b'048100000000000016dd') on cluster 32768, because Data is too short to contain 2 bytes
2022-05-10 11:10:56,530 INFO    : [       ZigpyCom_3] transport_request: _limit_concurrency <ZNPCoordinator model='CC1352/CC2652, Z-Stack 3.30+ (build 20220219)' manuf='Texas Instruments' nwk=0x0000 ieee=00:12:4b:00:25:6c:12:07 is_initialized=True> 4
2022-05-10 11:10:56,541 INFO    : [ ZigpyForwarder_3] Decode8040 - Reception of Node Descriptor for dd16000000000000 with status 81
2022-05-10 11:10:56,547 INFO    : [       ZigpyCom_3] ZigyTransport: process_raw_command  00:12:4b:00:25:6c:12:07 0 (<class 'int'>) 32768 (<class 'int'>)
2022-05-10 11:10:56,563 ERROR   :Task exception was never retrieved
future: <Task finished name='Task-95845' coro=<ControllerApplication.on_zdo_message() done, defined at /var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py:643> exception=ValueError('Data is too short to contain 2 bytes')>
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy_znp/zigbee/application.py", line 651, in on_zdo_message
    args, data = list_deserialize(data, types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigate/zigpy/types/basic.py", line 142, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 2 bytes
```

pipiche38, 2022-05-10:
@puddly . is the recent logs helping you https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/1142#issuecomment-1116101612

closed 2022-07-13 by pipiche38