zigbeefordomoticz/Domoticz-Zigbee #1154: add Zigate menu to Custom UI

khongpt, 2022-04-26:
**Is your feature request related to a problem? Please describe.**
Based on this feture https://zigbeefordomoticz.github.io/wiki/en-eng/Plugin_Integration.html
After using zigbee2mqtt plugin for a while, i came up with a way to copy zigate.html into custom UI on domoticz automatically when enable plugin.

**Describe the solution you'd like**
I created a file zigate.html and put in to the plugin folder. Add the following lines of code in plugin.py

# import copy2 in import area 
from shutil import copy2

# addition some lines of code in on Start func
    def onStart(self):
        ################################################################################
        Domoticz.Log("Installing plugin custom page")
        try:
            source_path = Parameters['HomeFolder']
            templates_path = Parameters['StartupFolder'] + 'www/templates'           
            
            copy2(source_path + '/Zigate.html', templates_path)
        except Exception as e:
            Domoticz.error('Error during installing plugin custom page')
            Domoticz.error(repr(e))
        ################################################################################

**content of zigate.html file**
#############################################
<iframe id="Zigate" style="width:100%;height:800px;overflow:scroll;"></iframe>
<script>
document.getElementById('Zigate').src = "http://" + location.hostname + ":9440/";
</script>
#############################################

pipiche38, 2022-04-26:
Integrated into PR #1155

closed 2022-04-26 by pipiche38

bierdok, 2024-03-06:
hello @pipiche38, the iframe src value should be a dedicated parameter in my opinion.

pipiche38, 2024-03-06:
> hello @pipiche38, the iframe src value should be a dedicated parameter in my opinion.

I'm not a specialist on that one. Feel free to provide a PR, this would be really appreciated