zigbeefordomoticz/Domoticz-Zigbee #1177: [zigpy-integration] - at startup Permit to join with Duration 0 is sent

pipiche38, 2022-05-11:
@puddly when starting the zigpy layer, it looks like several permit to join messages are sent. Are they from the firmware or from the zigpy layers ?

[image]

What is the purpose ?

This is very annoying as I have 7 shutters which when receiving a Permit to Join order make a move !

pipiche38, 2022-05-11:
This looks ZNP, as I don't see this behaviour on Elelabs with bellows

puddly, 2022-05-11:
I've not encountered a device like that before. It activates when joins are permitted?

ZNP disables joins because some devices (CC2531) permanently permits joins on startup unless you disable them.

pipiche38, 2022-05-11:
Basically these are the Profalux Shutter. And when you do a Permit to Join for more than 0 you get a little move Up and Down. Then when the Permit to Join expire or you set it for 0 (basically closing the network) you got a 2 times Up - Down

Shouldn't be the plugin responsibility rather than the zigpy library ? Or would it be possible to make it optional ?

pipiche38, 2022-05-13:
@puddly, can we have a znp config parameter, enable by default, but which can be then be disable by the plugin ?

puddly, 2022-05-13:
This is going to be moved out of individual radio libraries and into zigpy's `startup` method in the new radio API: https://github.com/puddly/zigpy/blob/8cadae87d8989ccd5ce6edec2184ff8ba2aa15d3/zigpy/application.py#L78

pipiche38, 2022-05-13:
Not sure I understand. DO you mean that this will be the same for ALL radios ?

puddly, 2022-05-13:
This is currently the case with the new API and has been for a while, ZNP was just doing it a second time. If you find this default behavior conflicting with your use case, override `startup`. It's only 16 lines of code and is a very high-level function, you may not even be using it directly.

pipiche38, 2022-05-16:
@puddly I had a look and did some test, I don't think this is 16 lines of code, or I missed something.
If I understood overriding the ZNP startup() this consist into taking the all _startup() from znp/application.py which is totally crazy in terms of manageability.

puddly, 2022-05-16:
I was referring to zigpy's `startup` in the new radio API.

pipiche38, 2022-05-19:
fix implemented 4528ab4cf08bc6860de9ae848a3cb43c769feadc

closed 2022-05-19 by pipiche38