zigbeefordomoticz/Domoticz-Zigbee #1178: [zigpy-integration] - zigpy is not filtering duplicate

pipiche38, 2022-05-11:
@puddly if a message is sent several time by a device, a handle_message() is call each time . Why is the zigpy low layer not filtering ? This is on a ZNP / SonOff 

[image]

```
2022-05-11 21:36:37,097 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=204, SecurityUse=<Bool.false: 0>, TimeStamp=5060689, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:37,250 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=233, SecurityUse=<Bool.false: 0>, TimeStamp=5070219, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:37,449 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=229, SecurityUse=<Bool.false: 0>, TimeStamp=5082642, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:37,653 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=229, SecurityUse=<Bool.false: 0>, TimeStamp=5095399, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:37,847 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5107517, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:38,048 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5120116, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:38,246 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=229, SecurityUse=<Bool.false: 0>, TimeStamp=5132495, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:38,447 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=229, SecurityUse=<Bool.false: 0>, TimeStamp=5145016, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:38,646 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5157494, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:38,844 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5169909, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:39,045 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5182381, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:39,242 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5194732, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:39,441 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5207193, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:39,641 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5219725, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
2022-05-11 21:36:39,840 DEBUG   :Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x836C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=225, SecurityUse=<Bool.false: 0>, TimeStamp=5232101, TSN=0, Data=b'\x08\x0F\x0A\x01\x00\x20\x45\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x836C, MsgResultRadius=29)
```

puddly, 2022-05-11:
I believe this was reported before https://github.com/zigpy/zigpy/issues/671

You likely need to send a default response.

pipiche38, 2022-05-11:
I wonder if this couldn't be done at the firmware level, this will prevent overhead on the library or the plugin side

pipiche38, 2022-05-12:
implement a ZCL filtering mecanism

closed 2022-05-12 by pipiche38