zigbeefordomoticz/Domoticz-Zigbee #1199: Tuya TS0601 coil fan thermostat (not TRV) support

mitkodotcom, 2022-05-27:


pipiche38, 2022-05-27:
Thanks for the issue. I have already responded to the question on the forum.

let us know which version of the plugin you are using

pipiche38, 2022-05-27:
You can also have a look here to the needed information
https://zigbeefordomoticz.github.io/wiki/en-eng/Problem_Dealing-with-none-certified-device.html

Otherwise:
from stable6 ( https://zigbeefordomoticz.github.io/wiki/en-eng/Plugin_Version-6.html )
switch to beta6

git pull
git checkout beta6
git pull

remove all widgets already created for this device

restart the plugin

redo the pairing

mitkodotcom, 2022-05-27:
Meanwhile I have found a workaround. If I add

"_TZE200_dzuqwsyg",

to

TUYA_THERMOSTAT_MANUFACTURER = (

in Modules/tuya.py

the device is now recognized as a thermostat and it is working somehow: the temperature, the setpoint and the on/off switch are OK. The mode selector switch is working, however Auto turns Heating mode, Manual turns Cooling mode. There is no control of the fan speed. I can try to add a more suitable widgets, but I'm not sure where - in thermostats.py or tuyaTRV.py as the fan coil thermostat is something in between thermostat and trv (basically it is a trv with a 3-stage fan control)

pipiche38, 2022-05-28:
Please do switch to beta6

and then provides me

export.json as documented here ( https://zigbeefordomoticz.github.io/wiki/en-eng/Problem_Dealing-with-none-certified-device.html )

enable debugTuya
and provide me the log produced when using the Fan command (from the Thermostat).

mitkodotcom, 2022-05-28:
Here is the log of the fan control, switching to level 3, 2, 1 and Auto, in this order.

2022-05-28 09:26:51,908 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-thermostat Nwkid: 8f11/01 dp: 1c dt: 04 data: 02
2022-05-28 09:26:51,909 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 02
2022-05-28 09:26:51,909 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 02 UNKNOW dp for Manuf: _TZE200_dzuqwsyg, Model: TS0601-thermostat
2022-05-28 09:26:54,608 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-thermostat Nwkid: 8f11/01 dp: 1c dt: 04 data: 01
2022-05-28 09:26:54,609 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 01
2022-05-28 09:26:54,610 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 01 UNKNOW dp for Manuf: _TZE200_dzuqwsyg, Model: TS0601-thermostat
2022-05-28 09:26:55,434 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-thermostat Nwkid: 8f11/01 dp: 1c dt: 04 data: 00
2022-05-28 09:26:55,435 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 00
2022-05-28 09:26:55,436 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 00 UNKNOW dp for Manuf: _TZE200_dzuqwsyg, Model: TS0601-thermostat
2022-05-28 09:26:56,103 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-thermostat Nwkid: 8f11/01 dp: 1c dt: 04 data: 03
2022-05-28 09:26:56,104 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 03
2022-05-28 09:26:56,105 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 1c datatype: 4 data: 03 UNKNOW dp for Manuf: _TZE200_dzuqwsyg, Model: TS0601-thermostat

Mode switching to Cooling, Heatng and Fan only, in this order:

2022-05-28 09:28:39,074 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-thermostat Nwkid: 8f11/01 dp: 02 dt: 04 data: 00
2022-05-28 09:28:39,076 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 02 datatype: 4 data: 00
2022-05-28 09:28:39,078 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: 8f11/01 Dp: 2 DataType: 4 ManualMode: 00
2022-05-28 09:28:39,079 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: 8f11/01 Manual Mode Off
2022-05-28 09:28:40,548 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-thermostat Nwkid: 8f11/01 dp: 02 dt: 04 data: 01
2022-05-28 09:28:40,549 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 02 datatype: 4 data: 01
2022-05-28 09:28:40,550 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: 8f11/01 Dp: 2 DataType: 4 ManualMode: 01
2022-05-28 09:28:42,729 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-thermostat Nwkid: 8f11/01 dp: 02 dt: 04 data: 02
2022-05-28 09:28:42,730 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 8f11 dp: 02 datatype: 4 data: 02
2022-05-28 09:28:42,731 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: 8f11/01 Dp: 2 DataType: 4 ManualMode: 02

[export.json.gz](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/8790624/export.json.gz)

pipiche38, 2022-05-29:
Thanks for the info provided.

Can you kindly switch to pipiche38/device/TS0601-Coil-Fan

`git pull; git checkout pipiche38/device/TS0601-Coil-Fan`

then you need to remove all widgets which have created in Domoticz

restart the plugin to take the new version

redo the pairing

if the pairing works you should have now a Fan Control widget created (there is a Off mode, but as the Tuya do not provide this possibility for now it won't work).

Let me know how it does ?

mitkodotcom, 2022-05-29:
There's no difference. Only one selector switch is created - ThermoMode (Off, Auto, Manual)

2022-05-29 20:39:31.035  Status: Zigate: [None] NEW OBJECT: b954 None
2022-05-29 20:39:31.157  Status: Zigate: [-] NEW OBJECT: b954 Active Endpoint Response Ep: 01 LQI: 116
2022-05-29 20:39:31.160  Status: Zigate: [-] NEW OBJECT: b954 Request Simple Descriptor for Ep: 01
2022-05-29 20:39:31.373  Status: Zigate: [-] NEW OBJECT: b954 Simple Descriptor Response EP: 0x01 LQI: 115
2022-05-29 20:39:31.374  Status: Zigate: [-]    NEW OBJECT: b954 ProfileID 0104
2022-05-29 20:39:31.375  Status: Zigate: [-]    NEW OBJECT: b954 ZDeviceID 0051
2022-05-29 20:39:31.376  Status: Zigate: [-]    NEW OBJECT: b954 Application Version 0001
2022-05-29 20:39:31.377  Status: Zigate: [-]    NEW OBJECT: b954 Cluster IN Count: 04
2022-05-29 20:39:31.378  Status: Zigate: [-]       NEW OBJECT: b954 Cluster In 1: 0000 (Basic)
2022-05-29 20:39:31.379  Status: Zigate: [-]       NEW OBJECT: b954 Cluster In 2: 0004 (Groups)
2022-05-29 20:39:31.381  Status: Zigate: [-]       NEW OBJECT: b954 Cluster In 3: 0005 (Scenes)
2022-05-29 20:39:31.382  Status: Zigate: [-]       NEW OBJECT: b954 Cluster In 4: ef00
2022-05-29 20:39:31.383  Status: Zigate: [-]    NEW OBJECT: b954 Cluster OUT Count: 02
2022-05-29 20:39:31.384  Status: Zigate: [-]       NEW OBJECT: b954 Cluster Out 1: 0019 (Over-the-Air Upgrade)
2022-05-29 20:39:31.385  Status: Zigate: [-]       NEW OBJECT: b954 Cluster Out 2: 000a (Time)
2022-05-29 20:39:37.941  Status: Zigate: [0] NEW OBJECT: b954 Model Name: TS0601-thermostat
2022-05-29 20:39:37.942  Status: Zigate: [0] NEW OBJECT: b954 Manufacturer Name: _TZE200_dzuqwsyg
2022-05-29 20:39:37.942  Status: Zigate: [None] NEW OBJECT: b954 Request Node Descriptor

mitkodotcom, 2022-05-29:
I don't believe Conf/Certified/Tuya/TS0601-_TZE200_dzuqwsyg.json is loaded. I have changed ThermoMode_2 to ACMode in this file but it is still detected as ThermoMode_2 (it shoud actually have it's own ACMode as there is no Dry mode, only Cooling, Heating and Fan).

pipiche38, 2022-05-29:
that is my fear. I'll investigate what is happening

mitkodotcom, 2022-05-29:
For some reason it is recognized as "-thermostat" even if I move the manufacturer name to TUYA_eTRV3_MANUFACTURER, for example. However, Conf/Certified/Tuya/TS0601-thermostat.json is not loaded either, as I have edited the file and changed ThermoMode_2 to ThermoMode_5, but the device is still recognized as TS0601-thermostat, ThermoMode_2

I have tried to reorder elifs in Modules/readClusters.py putting modelName += "-thermostat" at the end, but it is still recognized as TS0601-thermostat instead of TS0601-eTRV3. There may be some hardcoded defaults. I'll keep searching tomorrow.

pipiche38, 2022-05-29:
The only possibility that I can see is that we get the Manufacturer Name somehow late, so we might have to hardcode

pipiche38, 2022-05-29:
if yoou can redo a pairing with

debugPairing
debugCluster

that would be great, I'll fixed it tomorrow

mitkodotcom, 2022-05-29:
Sorry, it was my mistake. I was on the wrong branch (right branch, wrong folder). 

Now the plugin detects the device as it should and a Fan selector widget is added, however no commands are sent or received. Tomorrow I will continue with the debugging as it is too late here.

022-05-29 23:36:25,478 INFO    : [ZiGateForwarder_3] --> Adding device c4d6 in self.DevicesInPairingMode
2022-05-29 23:36:25,479 INFO    : [ZiGateForwarder_3] --> ['c4d6']
2022-05-29 23:36:25,480 INFO    : [ZiGateForwarder_3] [None] NEW OBJECT: c4d6 None
2022-05-29 23:36:25,638 INFO    : [ZiGateForwarder_3] [-] NEW OBJECT: c4d6 Active Endpoint Response Ep: 01 LQI: 119
2022-05-29 23:36:25,640 INFO    : [ZiGateForwarder_3] Request Basic  Manufacturer via Read Attribute request: 0004
2022-05-29 23:36:25,641 INFO    : [ZiGateForwarder_3] EP: {'01': {}}
2022-05-29 23:36:25,642 INFO    : [ZiGateForwarder_3] [-] NEW OBJECT: c4d6 Request Simple Descriptor for Ep: 01
2022-05-29 23:36:25,908 INFO    : [ZiGateForwarder_3] [-] NEW OBJECT: c4d6 Simple Descriptor Response EP: 0x01 LQI: 120
2022-05-29 23:36:25,909 INFO    : [ZiGateForwarder_3] [-]    NEW OBJECT: c4d6 ProfileID 0104
2022-05-29 23:36:25,910 INFO    : [ZiGateForwarder_3] [-]    NEW OBJECT: c4d6 ZDeviceID 0051
2022-05-29 23:36:25,911 INFO    : [ZiGateForwarder_3] [-]    NEW OBJECT: c4d6 Application Version 0001
2022-05-29 23:36:25,912 INFO    : [ZiGateForwarder_3] [-]    NEW OBJECT: c4d6 Cluster IN Count: 04
2022-05-29 23:36:25,913 INFO    : [ZiGateForwarder_3] [-]       NEW OBJECT: c4d6 Cluster In 1: 0000 (Basic)
2022-05-29 23:36:25,914 INFO    : [ZiGateForwarder_3] [-]       NEW OBJECT: c4d6 Cluster In 2: 0004 (Groups)
2022-05-29 23:36:25,916 INFO    : [ZiGateForwarder_3] [-]       NEW OBJECT: c4d6 Cluster In 3: 0005 (Scenes)
2022-05-29 23:36:25,917 INFO    : [ZiGateForwarder_3] [-]       NEW OBJECT: c4d6 Cluster In 4: ef00
2022-05-29 23:36:25,928 INFO    : [ZiGateForwarder_3] [-]    NEW OBJECT: c4d6 Cluster OUT Count: 02
2022-05-29 23:36:25,928 INFO    : [ZiGateForwarder_3] [-]       NEW OBJECT: c4d6 Cluster Out 1: 0019 (Over-the-Air Upgrade)
2022-05-29 23:36:25,928 INFO    : [ZiGateForwarder_3] [-]       NEW OBJECT: c4d6 Cluster Out 2: 000a (Time)
2022-05-29 23:36:26,912 INFO    : [       MainThread] [0] NEW OBJECT: c4d6 Model Name: TS0601-_TZE200_dzuqwsyg
2022-05-29 23:36:26,913 INFO    : [       MainThread] [0] NEW OBJECT: c4d6 Manufacturer Name: _TZE200_dzuqwsyg
2022-05-29 23:36:26,914 INFO    : [       MainThread] [None] NEW OBJECT: c4d6 Request Node Descriptor
2022-05-29 23:36:26,959 INFO    : [       MainThread] WARNING - send_zigatecmd : [32] 0100 02c4d60101000000000000010005 c4d6 ZigateQueue: 16
2022-05-29 23:36:26,960 INFO    : [       MainThread] WARNING - send_zigatecmd : [33] 0100 02c4d60101000000000000010007 c4d6 ZigateQueue: 17
2022-05-29 23:36:26,962 INFO    : [       MainThread] WARNING - send_zigatecmd : [34] 0100 02c4d6010100000000000001fffe c4d6 ZigateQueue: 18
2022-05-29 23:36:26,996 INFO    : [       MainThread] WARNING - send_zigatecmd : [35] 0530 02c4d60000003300000200020000 None ZigateQueue: 19
2022-05-29 23:36:31,933 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 01 data type: 1 data: 00
2022-05-29 23:36:31,993 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 02 data type: 4 data: 01
2022-05-29 23:36:32,044 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 04 data type: 1 data: 01
2022-05-29 23:36:32,054 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 10 data type: 2 data: 00000019
2022-05-29 23:36:32,149 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 18 data type: 2 data: 0000010e
2022-05-29 23:36:32,162 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 13 data type: 2 data: 00000023
2022-05-29 23:36:32,175 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 1a data type: 2 data: 00000005
2022-05-29 23:36:32,201 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 1b data type: 2 data: 00000000

mitkodotcom, 2022-05-30:
This is the log when trying to send commands to the device:

2022-05-30 09:03:57,021 DEBUG   : [       MainThread] tuya_trv_onoff - c4d6 Switch: 0
2022-05-30 09:03:57,021 DEBUG   : [       MainThread] tuya_trv_switch_onoff - c4d6 Switch: 0
2022-05-30 09:03:57,021 DEBUG   : [       MainThread] get_datapoint_command - c4d6 Switch not found in eTRV_MATRIX[ TS0601-eTRV1 ]['ToDevice']
2022-05-30 09:03:57,022 DEBUG   : [       MainThread] tuya_trv_switch_onoff - c4d6 dp for Switch: None
2022-05-30 09:03:59,217 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 18 dt: 02 data: 00000109
2022-05-30 09:03:59,218 INFO    : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg UNMANAGED Nwkid: c4d6/01 dp: 18 data type: 2 data: 00000109
2022-05-30 09:04:06,334 DEBUG   : [       MainThread] tuya_trv_onoff - c4d6 Switch: 1
2022-05-30 09:04:06,335 DEBUG   : [       MainThread] tuya_trv_switch_onoff - c4d6 Switch: 1
2022-05-30 09:04:06,336 DEBUG   : [       MainThread] get_datapoint_command - c4d6 Switch not found in eTRV_MATRIX[ TS0601-eTRV1 ]['ToDevice']
2022-05-30 09:04:06,337 DEBUG   : [       MainThread] tuya_trv_switch_onoff - c4d6 dp for Switch: None
2022-05-30 09:04:29,010 DEBUG   : [       MainThread] tuya_fan_speed - c4d6 Speed: 0
2022-05-30 09:04:29,011 DEBUG   : [       MainThread] get_datapoint_command - c4d6 FanSpeed not found in eTRV_MATRIX[ TS0601-eTRV1 ]['ToDevice']
2022-05-30 09:04:29,012 DEBUG   : [       MainThread] tuya_trv_switch_onoff - c4d6 dp for FanSpeed: None
2022-05-30 09:04:42,724 DEBUG   : [       MainThread] tuya_trv_mode - c4d6 tuya_trv_mode: 20
2022-05-30 09:04:42,726 DEBUG   : [       MainThread] tuya_trv_switch_mode - c4d6 Switch Mode: 14
2022-05-30 09:04:42,727 DEBUG   : [       MainThread] tuya_trv_switch_mode - c4d6 dp for TrvMode: 4
2022-05-30 09:04:42,727 DEBUG   : [       MainThread] tuya_trv_switch_mode - c4d6 Action: 0404 Data: 02 
2022-05-30 09:04:42,729 DEBUG   : [       MainThread] tuya_cmd - c4d6/01 cmd: 00 payload: 111b0000040404000102
2022-05-30 09:04:51,469 DEBUG   : [       MainThread] tuya_trv_mode - c4d6 tuya_trv_mode: 10
2022-05-30 09:04:51,472 DEBUG   : [       MainThread] tuya_trv_switch_mode - c4d6 Switch Mode: a
2022-05-30 09:04:51,472 DEBUG   : [       MainThread] tuya_trv_switch_mode - c4d6 dp for TrvMode: 4
2022-05-30 09:04:51,472 DEBUG   : [       MainThread] tuya_trv_switch_mode - c4d6 Action: 0404 Data: 01 
2022-05-30 09:04:51,472 DEBUG   : [       MainThread] tuya_cmd - c4d6/01 cmd: 00 payload: 111c0000050404000101

mitkodotcom, 2022-05-30:
adding     "TS0601-_TZE200_dzuqwsyg": "TS0601-thermostat",  to eTRV_MODELS made the switches work from domoticz to device, although the setpoint does not work - nothing is logged when changing it in domoticz. From device to domoticz is still UNMANAGED

May be eTRV_MODELS should be populated from Conf/Certified/*

mitkodotcom, 2022-05-30:
adding the model to     elif _ModelName in ("TS0601-thermostat","TS0601-_TZE200_dzuqwsyg"): in tuya_response made the incoming direction working somehow (no fan control).

Before continuing to hardcode the model here and there, may I suggest to replace all those elifs with something like elif _ReceiveType == 1, elif _ReceiveType == 2, etc. and populate _ReceiveType the same way as eTRV_MODEL - by manufacturer, by model type, by Conf/Certified/, etc. The idea is to have one big matrix with data for each model and whenever a new model is added you will have to edit only the matrix / config file, and not a bunch of ifs (unless, of course, a new type has to be added, as in this case). You already have something similar for the EP->Type parameter.

pipiche38, 2022-05-30:
Can you add debugCommand so we can track from the time you do the Domoticz Widget command ?

mitkodotcom, 2022-05-30:
2022-05-30 10:13:15,054 DEBUG   : [       MainThread] mgtCommand (c4d6) Devices[4].Name: Zigate - TS0601-_TZE200_dzuqwsyg_ThermoSetpoint-8cf681fffef8ceb0-01 Command: Set Level Level: 21.0 Color: 
2022-05-30 10:13:15,055 DEBUG   : [       MainThread] --------->1   ClusterType founds: [('01', '39', 'ThermoSetpoint')] for Unit: 4
2022-05-30 10:13:15,056 DEBUG   : [       MainThread] --------->2   EPOut: 01 DeviceType: ThermoSetpoint WidgetID: 39
2022-05-30 10:13:15,057 DEBUG   : [       MainThread] --------->3   forceUpdateDev: False
2022-05-30 10:13:15,061 DEBUG   : [       MainThread] --------->4   Ready to action
2022-05-30 10:13:15,062 DEBUG   : [       MainThread] --------->5   profalux: False
2022-05-30 10:13:15,062 DEBUG   : [       MainThread] --------->6   Model Name: TS0601-_TZE200_dzuqwsyg
2022-05-30 10:13:15,063 DEBUG   : [       MainThread] --------->7   Health: Live
2022-05-30 10:13:15,064 DEBUG   : [       MainThread] mgtCommand : Set Level for Device: c4d6 EPout: 01 Unit: 4 DeviceType: ThermoSetpoint Level: 21.0
2022-05-30 10:13:15,067 DEBUG   : [       MainThread] mgtCommand : Set Level for Device: c4d6 EPout: 01 Unit: 4 DeviceType: ThermoSetpoint Level: 21.0

pipiche38, 2022-05-30:
> adding the model to elif _ModelName in ("TS0601-thermostat","TS0601-_TZE200_dzuqwsyg"): in tuya_response made the incoming direction working somehow (no fan control).
>
> Before continuing to hardcode [...]

This looks interesting. I know that the current way is a very quick and dirty. Mainly because we were not expecting Tuya to work like that when started integrating the first Tuya equipement

What I have in mind , is to rely more on an information which is in the Config file itself, so there is no code update in that case. (except if there is new features ....)

pipiche38, 2022-05-30:
> 2022-05-30 10:13:15,054 DEBUG : [ MainThread] mgtCommand (c4d6) Devices[4].Name: Zigate - TS0601-_TZE200_dzuqwsyg_ThermoSetpoint-8cf681fffef8ceb0-01 Command: Set Level Level: 21.0 Color: 2022-05-30 [...]

Do you mean that when doing a Fan control, we get to a Setpoint command ?

mitkodotcom, 2022-05-30:
> 
> Do you mean that when doing a Fan control, we get to a Setpoint command ?

No. The Setpoint command does not work, even after adding "TS0601-_TZE200_dzuqwsyg": "TS0601-thermostat", to eTRV_MODELS. 

The fan control does not work in the opposite direction (from device to domoticz). It works from domoticz to device. The other selector switch ThermoMode_2 (receive_manual_mode) also does not work from device to domoticz (there should be another ThermoMode_6 ot ACmode_2 for this device anyway). Receive_setpoint, receive_onoff and receive_temperature are OK. 

2022-05-30 10:56:02,412 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 18 dt: 02 data: 0000011d
2022-05-30 10:56:02,413 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 18 datatype: 2 data: 0000011d
2022-05-30 10:56:02,413 DEBUG   : [ZiGateForwarder_3] receive_temperature - Nwkid: c4d6/01 Temperature: 285
2022-05-30 10:56:02,713 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 1c dt: 04 data: 00
2022-05-30 10:56:02,714 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 1c datatype: 4 data: 00
2022-05-30 10:56:02,714 DEBUG   : [ZiGateForwarder_3] receive_fan - Nwkid: c4d6/01 : 0
2022-05-30 10:56:03,980 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 1c dt: 04 data: 03
2022-05-30 10:56:03,981 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 1c datatype: 4 data: 03
2022-05-30 10:56:03,982 DEBUG   : [ZiGateForwarder_3] receive_fan - Nwkid: c4d6/01 : 3
2022-05-30 10:56:05,507 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 1c dt: 04 data: 02
2022-05-30 10:56:05,507 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 1c datatype: 4 data: 02
2022-05-30 10:56:05,509 DEBUG   : [ZiGateForwarder_3] receive_fan - Nwkid: c4d6/01 : 2
2022-05-30 10:56:06,438 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 1c dt: 04 data: 01
2022-05-30 10:56:06,439 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 1c datatype: 4 data: 01
2022-05-30 10:56:06,440 DEBUG   : [ZiGateForwarder_3] receive_fan - Nwkid: c4d6/01 : 1
2022-05-30 10:56:07,955 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 02 dt: 04 data: 02
2022-05-30 10:56:07,956 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 02 datatype: 4 data: 02
2022-05-30 10:56:07,957 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: c4d6/01 Dp: 2 DataType: 4 ManualMode: 02
2022-05-30 10:56:08,915 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 02 dt: 04 data: 00
2022-05-30 10:56:08,916 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 02 datatype: 4 data: 00
2022-05-30 10:56:08,917 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: c4d6/01 Dp: 2 DataType: 4 ManualMode: 00
2022-05-30 10:56:08,918 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: c4d6/01 Manual Mode Off
2022-05-30 10:56:09,624 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 02 dt: 04 data: 01
2022-05-30 10:56:09,625 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 02 datatype: 4 data: 01
2022-05-30 10:56:09,626 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: c4d6/01 Dp: 2 DataType: 4 ManualMode: 01
2022-05-30 10:56:10,549 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 02 dt: 04 data: 02
2022-05-30 10:56:10,549 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 02 datatype: 4 data: 02
2022-05-30 10:56:10,549 DEBUG   : [ZiGateForwarder_3] receive_manual_mode - Nwkid: c4d6/01 Dp: 2 DataType: 4 ManualMode: 02
2022-05-30 10:56:13,373 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: c4d6/01 dp: 10 dt: 02 data: 00000019
2022-05-30 10:56:13,374 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: c4d6 dp: 10 datatype: 2 data: 00000019
2022-05-30 10:56:13,375 DEBUG   : [ZiGateForwarder_3] receive_setpoint - Nwkid: c4d6/01 Setpoint: 25 for model taget: TS0601-thermostat
2022-05-30 10:56:13,376 DEBUG   : [ZiGateForwarder_3] receive_setpoint - After Nwkid: c4d6/01 Setpoint: 25 for model taget: TS0601-thermostat

mitkodotcom, 2022-05-30:
Some more debug for set setpoint command:

2022-05-30 13:22:21,689 DEBUG   : [       MainThread] mgtCommand (c4d6) Devices[4].Name: Zigate - TS0601-_TZE200_dzuqwsyg_ThermoSetpoint-8cf681fffef8ceb0-01 Command: Set Level Level: 22.0 Color: 
2022-05-30 13:22:21,690 DEBUG   : [       MainThread] --------->1   ClusterType founds: [('01', '39', 'ThermoSetpoint')] for Unit: 4
2022-05-30 13:22:21,691 DEBUG   : [       MainThread] --------->2   EPOut: 01 DeviceType: ThermoSetpoint WidgetID: 39
2022-05-30 13:22:21,694 DEBUG   : [       MainThread] --------->3   forceUpdateDev: False
2022-05-30 13:22:21,695 DEBUG   : [       MainThread] --------->4   Ready to action
2022-05-30 13:22:21,697 DEBUG   : [       MainThread] --------->5   profalux: False
2022-05-30 13:22:21,707 DEBUG   : [       MainThread] --------->6   Model Name: TS0601-_TZE200_dzuqwsyg
2022-05-30 13:22:21,708 DEBUG   : [       MainThread] --------->7   Health: Live
2022-05-30 13:22:21,708 DEBUG   : [       MainThread] mgtCommand : Set Level for Device: c4d6 EPout: 01 Unit: 4 DeviceType: ThermoSetpoint Level: 22.0
2022-05-30 13:22:21,708 DEBUG   : [       MainThread] mgtCommand : Set Level for Device: c4d6 EPout: 01 Unit: 4 DeviceType: ThermoSetpoint Level: 22.0
2022-05-30 13:22:21,709 DEBUG   : [       MainThread] thermostat_Setpoint - for c4d6 with value 2200
2022-05-30 13:22:21,709 DEBUG   : [       MainThread] thermostat_Setpoint - standard for c4d6 with value 2200
2022-05-30 13:22:21,709 DEBUG   : [       MainThread] setpoint: 2200
2022-05-30 13:22:21,709 DEBUG   : [       MainThread] setpoint: 2200
2022-05-30 13:22:21,710 DEBUG   : [       MainThread] ReadAttributeRequest_0201 - Key: c4d6 
2022-05-30 13:22:21,710 DEBUG   : [       MainThread] ---- retreive_ListOfAttributesByCluster: List of Attributes for cluster 0201 : 0x0000 0x0008 0x0010 0x0011 0x0012 0x0014 0x0015 0x0016 0x001b 0x001c 0x001f 0xfd00
2022-05-30 13:22:21,710 DEBUG   : [       MainThread] Request Thermostat  via Read Attribute request c4d6/01 0x0000 0x0008 0x0010 0x0011 0x0012 0x0014 0x0015 0x0016 0x001b 0x001c 0x001f 0xfd00

pipiche38, 2022-05-30:
I beleive Fan Control should be ok on the 2 ways with the latest version.

for the rest, I will work on that later this week

mitkodotcom, 2022-05-30:
BTW, this device supports (and needs) temperature calibration. Fortunately, it can be calibrated with tuya's original gateway as it remembers the calibration settings when moved to a new coordinator. However, is it possible to calubrate it via this plugin? I saw there is Param->Calibration in config file, but it does not seem to be used.

pipiche38, 2022-06-01:
Ok. i have updated the eTRV_MODELS sorry for missing this one, did further update.
Can I ask you to redo the full pairing .

Then do what ever test you can. Would be really great if I could have the corresponding tuya log when something doesn't behave as you expect.

And finally if you could drop me the export.json that would be great

mitkodotcom, 2022-06-03:
> @mitkodotcom did you have a chance to test ?

Nope. I will be able to spend some time on the topic during the weekend.

mitkodotcom, 2022-06-05:
It works now in both directions, but this device must have different mode selector switch. Currently the mode widget has two options: Auto (level 10) and Manual (level 20).  From domoticz to device, Auto (level 10) turns on heating, Manual (level 20) turns on cooling. From device to domoticz turning to cooling sets Level 20 (Manual), but turning to heating does not set Level 10 (Auto).

It should have it's own mode (ThermoMode_6 or ACmode_2) with three options: Cooling, Heating and Fan only. You can see the corresponding logs here: https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/1199#issuecomment-1140186784

There is another problem with this mode selector widget (ThermoMode_2), but as long as it is not suitable for this device anyway, I don't know if we should consider it a problem before creating a new mode widget first. If you turn off the device, the selector switch goes to off state. So far so good. When you turn the device on, the widget stays off - it does not turn to the corresponding level / mode. 

Attached is the exports.jso
[export.json.gz](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/8840173/export.json.gz)
n

mitkodotcom, 2022-06-06:
Temperature calibration from config file and from Management / Device management / Parameters also works, but tuya_set_calibration_if_needed()  is only executed when changing the setpoint. It is not too big of a problem as long as you know that you have to update setpoint after editing the parameters.

Suggestion: you may compare the data returned by receive_calibration to the parameter Calibration from config file / device management and call tuya_set_calibration_if_needed() if it is different.

pipiche38, 2022-06-06:
@mitkodotcom  pushed an update which should also make the Thermo working.
Unfortunately you'll have to redo the pairing from scratch in order to get the right widgets created

so update on the same branch
remove all widgets
restart
do the pairing
 and let me know

pipiche38, 2022-06-06:
you can try with the latest

pipiche38, 2022-06-07:
So please provides a PR.

mitkodotcom, 2022-06-07:
If you accept my suggestion to use Type instead of model name, you just add

import re

and replace

    if model_target in ["TS0601-_TZE200_dzuqwsyg", ]:

to

    if re.search('/ThermoMode_6/',self.ListOfDevices[NwkId]["Ep"][srcEp]["Type"]):
    
in receive_manual_mode, receive_onoff and receive_preset

You also have to change the order of IFs in receive_onoff and put if..ThermoMode_6 before elif model_target in ["TS0601-thermostat", "TS0601-eTRV3", ]: because both IFs will match and only the first one will execute. 

Tested and works. If you need a PR I'll do it but I think there's no need to fork the project to change just one line. 

There's no need to change it anywhere else, as the model name is missing in receive_* functions only. However, in the future you may wish to use the same approach in domoticz-to-device direction, as this will reduce the number of IFs. 

I don't know if some sanity checks should be added - for example if ["Type"] does not exist (is it possible?).

pipiche38, 2022-06-07:
> If you accept my suggestion to use Type instead of model name, you just add
>
> import re
>
> and replace
>
> if model_target in ["TS0601-_TZE200_dzuqwsyg", ]:
> to
>
> if [...]
Better that you propose a PR , so you can test it as well.

Thanks

pipiche38, 2022-06-09:
Just pushed an update

mitkodotcom, 2022-06-09:
Two small details about this device:

1. Do you know what that means:

        2022-06-09 22:44:16,913 DEBUG   : [       MainThread] get_datapoint_command - 4bc8 BRTMode not found in eTRV_MATRIX[ TS0601-thermostat-Coil ]['ToDevice']
        
Here is the context:

2022-06-09 20:43:08,297 DEBUG   : [       MainThread] thermostat_Calibration - 2 complement form of Calibration offset on 4bc8 off 4294967294
2022-06-09 20:43:08,298 DEBUG   : [       MainThread] tuya_setpoint - 4bc8 setpoint: 2300
2022-06-09 20:43:08,298 DEBUG   : [       MainThread] tuya_setpoint - 4bc8 dp 16 for SetPoint: 2300
2022-06-09 20:43:08,299 DEBUG   : [       MainThread] tuya_trv_brt100_set_mode - 4bc8 mode: 1
2022-06-09 20:43:08,300 DEBUG   : [       MainThread] get_datapoint_command - 4bc8 BRTMode not found in eTRV_MATRIX[ TS0601-thermostat-Coil ]['ToDevice']
2022-06-09 20:43:08,301 DEBUG   : [       MainThread] tuya_trv_brt100_set_mode - 4bc8 dp for mode: None
2022-06-09 20:43:08,307 DEBUG   : [       MainThread] tuya_setpoint - 4bc8 dp 16 to Tuya Format SetPoint: 23 for model TS0601-thermostat-Coil with data 00000017
2022-06-09 20:43:08,308 DEBUG   : [       MainThread] tuya_cmd - 4bc8/01 cmd: 00 payload: 110d00004b1002000400000017
2022-06-09 20:43:08,308 DEBUG   : [       MainThread] tuya_setpoint - 4bc8 dp 16 cmd sent to Tuya Format SetPoint: 23
2022-06-09 20:43:08,494 DEBUG   : [ZiGateForwarder_3] tuya_response - Model: TS0601-_TZE200_dzuqwsyg Nwkid: 4bc8/01 dp: 10 dt: 02 data: 00000017
2022-06-09 20:43:08,495 DEBUG   : [ZiGateForwarder_3] tuya_eTRV_response - Nwkid: 4bc8 dp: 10 datatype: 2 data: 00000017
2022-06-09 20:43:08,496 DEBUG   : [ZiGateForwarder_3] receive_setpoint - Nwkid: 4bc8/01 Setpoint: 23 for model taget: TS0601-thermostat-Coil
2022-06-09 20:43:08,497 DEBUG   : [ZiGateForwarder_3] receive_setpoint - After Nwkid: 4bc8/01 Setpoint: 2.3 for model taget: TS0601-thermostat-Coil
     
It appears after setting the setpoint from domoticz and not when changing it from the device

2. How to debug if this device sends valve status? It actually has valve control and valve status indicator, but I'm not sure if it reports the status. There is nothing logged if debugTuya and debugInput are both checked. The same applies for fan status: when fan is set to Auto, the device's display shows the actual status of the fan (low, medium, high), but I'm not sure if it is reporting it to the gateway.

mitkodotcom, 2022-06-10:
Please add "TS0601-thermostat-Coil" to "if model_target" in receive_setpoint() as by default the setpoint is now divided by 10. I'll check the rest later.

pipiche38, 2022-06-13:
Did you have any chance to test over the week-end ?

pipiche38, 2022-06-16:
@mitkodotcom did you have a chance to test the latest update ?

mitkodotcom, 2022-06-26:
I just tested it.

In receive_setpoint() the model_target should be "TS0601-thermostat-Coil" instead of ""TS0601-_TZE200_dzuqwsyg"

pipiche38, 2022-06-26:
I don't see why, as the stack call is

receive_setpoint(self, Devices, model_target, NwkId, srcEp, ClusterID, dstNWKID, dstEP, dp, datatype, data)
tuya_eTRV_response(self, Devices, _ModelName, NwkId, srcEp, ClusterID, dstNWKID, dstEP, dp, datatype, data)
tuya_response(self, Devices, _ModelName, NwkId, srcEp, ClusterID, dstNWKID, dstEP, dp, datatype, data)
_ModelName = self.ListOfDevices[NwkId]["Model"]
tuyaReadRawAPS(self, Devices, NwkId, srcEp, ClusterID, dstNWKID, dstEP, MsgPayload)

So from that aspect it makes sense to have TS0601-_TZE200_dzuqwsyg

if it is not then we are not coming to receive_setpoint() by the same path, and that I don't understand

mitkodotcom, 2022-06-26:
We have already discussed this here: https://github.com/zigbeefordomoticz/Domoticz-Zigbee/pull/1211

It is because of the way the array eTRV_MODELS is used. 

eTRV_MODELS = {
    # Thermostat
    "TS0601-thermostat": "TS0601-thermostat",
    "TS0601-_TZE200_dzuqwsyg": "TS0601-thermostat-Coil",

pipiche38, 2022-06-26:
update pushed

mitkodotcom, 2022-06-26:
It's OK now. All widgets are working properly in both directions.

pipiche38, 2022-06-27:
so if all good, I'll merge de PR soon. Thanks for your support